Hypfer/Valetudo #589: Zone names not found

Schattenruf, 2020-08-25:
Hi,

I don't know if it is a bug or just a misunderstanding from me.
I added different zones:

[image]

And edited them where they are.
If I select one and want my robby to clean there it just says:
[image]

If I delete all these zones and just have one there it also states this error so I think it does not depend on how many I have created...

How can I fix this?
My robby is a Roborock S50 vith Firmware 1886 and Valetudo 0.6.1

Thanks :)

rumpeltux, 2020-08-26:
How did you edit their location? What’s the contents of your `valetudo.conf`?

Schattenruf, 2020-08-26:
Well nothing special. I added one simple room and clicked on the "+" and saved. I could always open it up and it has been saved correctly.
But the starting did not work.

Hypfer, 2021-01-09:
Should be fixed now 🤔

closed 2021-01-09 by Hypfer