Hypfer/Valetudo #591: Incorrect status and no maps on viomi v7

alceasan, 2020-08-28:
## Describe the bug
After a clean installation, Valetudo starts and commands work (start, return to home, ...), including MQTT integration, but after starting a cleaning session the status appears always as "docked". The area is being updated while the cleaning is up, and also battery indicator. Also, no map data...
I can start the cleaning from HomeAssistant through MQTT integration, but the problems persists.
## To Reproduce
Install valetudo following the steps on https://valetudo.cloud/pages/installation/viomi.html

### Screenshots

[image]
[image]

### Vacuum Model
Viomi v7

### Valetudo Version
0.6.1

### Firmware Version
3.5.3_0047

## Additional context
Not sure if related, but in my vaccum box the model is labeled as STYJ02YM (viomi v8), but after setting up on Mi Home App (on Chinese server, I couldn't do it on Europe servers) and rooting it, the model it apears on /etc/miio/device.conf is viomi.vacuum.v7

Hypfer, 2020-08-28:
Possibly a duplicate of  #550

alceasan, 2020-09-01:
Can I provide any relevant logs?

rumpeltux, 2020-09-04:
1. Make sure /etc/rc.d/S51valetudo has `dest=127.0.0.1`.
2. Same file: the line `iptables  -t nat -A OUTPUT -p tcp --dport 80   -d $host -j DNAT --to-destination $dest:8080`
should change to `$dest:80` (on the robot port is 80 not 8080 as in the development setup).
3. Also check that `/mnt/data/valetudo/config.json` has `map_upload_host` set to `http://127.0.0.1`

Once done, please reboot, start a clean and check again.

alceasan, 2020-09-04:
I just tried, but it didnn't work. Status continue to show as docked while it's cleaning.

[image]

alceasan, 2020-09-15:
Today I changed the deposit from the water tank and mop to the standard waste box and it suddenly started working.

Now the main interface shows correctly the status and map data:
[image]

[image]

I still haven't tried to go back to the water tank mod, tomorrow I'll try and report to see if it's related.

koper89, 2020-11-07:
I have same issue no map data, but I don't seem to have config.json (I installed through this rooting script + upgraded valetudo to newest version 0.6.1), in my case status and map data are not showing correctly (NO_MAP_DATA) and in info about device there's no firmware version, this is 3.5.3_0047 (but one which came in built so this is one without english version option for some reason).
How to add config.json where can I find it as you mentioned above that it should be in /mnt/data/valetudo/config.json but I don't have data folder in mnt valetudo on viomi.v7 is installed to /mnt/UDISK/valetudo

schannall, 2020-11-07:
https://valetudo.cloud/pages/troubleshooting.html

-> Check /etc/rc.d/S51valetudo if you have that spoofed IP there.

koper89, 2020-11-08:
> https://valetudo.cloud/pages/troubleshooting.html
> 
> -> Check /etc/rc.d/S51valetudo if you have that spoofed IP there.

Yes I seem to have it, here's content:
`#!/bin/sh
iptables         -F OUTPUT
iptables  -t nat -F OUTPUT
# for local development enter your local development host here
# and change port to 8080
dest=127.0.0.1
port=80
for host in 110.43.0.83 110.43.0.85; do
  iptables  -t nat -A OUTPUT -p tcp --dport 80   -d $host -j DNAT --to-destination $dest:$port
  iptables  -t nat -A OUTPUT -p udp --dport 8053 -d $host -j DNAT --to-destination $dest:8053
  iptables         -A OUTPUT                     -d $host/32  -j REJECT
done
`

koper89, 2020-11-08:
Resolved by changing spoofedIp  /mnt/UDISK/valetudo/config/valetudo.json to correct one. Thanks @schannall  (Maybe worth adding in docs that config file for viomi is in  /mnt/UDISK/valetudo/config/valetudo.json

rumpeltux, 2021-01-31:
This issue should be fixed in current HEAD (or rather is obsolete now, since we stopped faking the ip and instead patch miio_client).
Much of the new UI isn’t yet functional though, so looking forward to @Depau’s implementation of the missing capabilities.

Note that [installation instructions](https://valetudo.cloud/pages/installation/viomi.html#robot-setup) have [changed](https://github.com/Hypfer/Valetudo/commit/a38987affba30f37e0cd09a67627a51adf695679#diff-4005931a921cf15818e21388070f3da9fb515b88b9f2296f35012870bab3c3cd) so when redeploying a new binary you’ll need some adjustments.

closed 2021-01-31 by rumpeltux