Hypfer/Valetudo #592: Disappearing timers

Dinth, 2020-08-30:
## Describe the bug

Since reflashing Valetudo to 0.6.1 on my Xiaomi S50, timers stopped being persistent. Whenever i set up a timer, it is there for one day, sometimes two and after that, it just disappears from my settings. All my other settings, etc are preserved and it's not a bug with firmware resetting which was pestering me before. 

## To Reproduce

1. Create a new timer
2. Wait one or two days

### Vacuum Model

Xiaomi S50

### Valetudo Version

0.6.1

### Firmware Version
2020

## Expected behavior
Timers should be persistent as all other options

rimrul, 2020-08-31:
Firmware 2020 seems to be weird about timers. I had to enable persistent data, it wouldn't allow me to create timers otherwise. My timers don't disappear, though.

Hypfer, 2020-08-31:
It seems like roborock is trying to transition to server-based timers for some reason.
Maybe they want to complete eliminate local persistence?

In any case, I'm planning to completely ditch native timers and instead implement them in Valetudo which should solve this issue

Hypfer, 2021-01-09:
Timers have been removed in the latest version

closed 2021-01-09 by Hypfer

jansohn, 2021-01-09:
Any alternatives provided yet by Valetudo?

Hypfer, 2021-01-09:
No. I'd suggest using something like home assistant to do the scheduling.

This way, it can also act differently depending on whether or not someone is at home etc

conrad784, 2021-02-14:
> No. I'd suggest using something like home assistant to do the scheduling.
> 
> This way, it can also act differently depending on whether or not someone is at home etc

I did not find this in the docs and in a lot of forums I saw the edit timers interface. Do you mind adding this change to the FAQ for people searching this feature?