Hypfer/Valetudo #594: Pause doesn't work (STYTJ02YM)

Kaan88, 2020-09-05:
## Describe the bug

When I press "Pause" I get no audio confirmation, device stops where it was. When I press "Pause" again it doesn't continue and If I press "Start" device restarts the cleaning instead of continuing.

## To Reproduce

1.Press start
2.Pause
3.Doesn't resume

### Vacuum Model

STYTJ02YM
### Valetudo Version

 0.6.1 
### Firmware Version

 3.5.3_0017

## Expected behavior

It should resume cleaning after I press "Pause" again.

Logs

```
[2020-09-05T09:28:50.625Z] [INFO] Loading configuration file: local/config.json
[2020-09-05T09:28:50.631Z] [INFO] Starting Valetudo for Vacuum Model viomi.vacuum.v8
[2020-09-05T09:28:50.631Z] [INFO] DeviceId 328986625
[2020-09-05T09:28:50.631Z] [INFO] IP 192.168.1.7
[2020-09-05T09:28:50.631Z] [INFO] CloudSecret h9e9Bi9fnoPJJvkb
[2020-09-05T09:28:50.631Z] [INFO] LocalSecret 6fryEm0ZYM750PlB
[2020-09-05T09:28:50.631Z] [INFO] JS Runtime Version v14.4.0
[2020-09-05T09:28:50.631Z] [INFO] Max Heap Size: 144 MiB
[2020-09-05T09:28:50.632Z] [INFO] Node Flags: --max-old-space-size=96
[2020-09-05T09:28:50.645Z] [DEBUG] >>> local: HandshakePacket()
[2020-09-05T09:28:50.658Z] [INFO] Dummycloud is spoofing 110.43.0.83:8053 on 192.168.1.2:8053
[2020-09-05T09:28:50.661Z] [INFO] Webserver running on port 80
[2020-09-05T09:28:50.696Z] [DEBUG] <<< local* {"stamp":1599298130}
[2020-09-05T09:28:50.696Z] [DEBUG] >>> local* {"stamp":1599298130}
[2020-09-05T09:28:50.697Z] [DEBUG] <<= local: handshake complete
[2020-09-05T09:28:50.698Z] [DEBUG] >>> local: {"method":"set_language","params":[2],"id":2}
[2020-09-05T09:28:50.700Z] [DEBUG] <<< local* {"stamp":1599298130}
[2020-09-05T09:28:50.700Z] [DEBUG] local: Discarding pong
[2020-09-05T09:28:51.201Z] [DEBUG] local request 1 get_prop timed out
[2020-09-05T09:28:51.202Z] [DEBUG] >>> local: HandshakePacket()
[2020-09-05T09:28:51.203Z] [DEBUG] local request 2 set_language timed out
[2020-09-05T09:28:51.206Z] [DEBUG] <<< local* {"stamp":1599298130}
[2020-09-05T09:28:51.206Z] [DEBUG] local: Discarding pong
[2020-09-05T09:28:51.509Z] [DEBUG] >>> local: HandshakePacket()
[2020-09-05T09:28:51.515Z] [DEBUG] <<< local* {"stamp":1599298131}
[2020-09-05T09:28:51.515Z] [DEBUG] >>> local* {"stamp":1599298131}
[2020-09-05T09:28:51.515Z] [DEBUG] <<= local: handshake complete
[2020-09-05T09:28:51.516Z] [DEBUG] >>> local: {"method":"set_language","params":[2],"id":2004}
[2020-09-05T09:28:51.521Z] [DEBUG] <<< local* {"stamp":1599298131}
[2020-09-05T09:28:51.521Z] [DEBUG] local: Discarding pong
[2020-09-05T09:28:52.021Z] [DEBUG] local request 2003 get_prop timed out
[2020-09-05T09:28:52.022Z] [DEBUG] >>> local: HandshakePacket()
[2020-09-05T09:28:52.022Z] [DEBUG] local request 2004 set_language timed out
[2020-09-05T09:28:52.025Z] [DEBUG] <<< local* {"stamp":1599298131}
[2020-09-05T09:28:52.025Z] [DEBUG] local: Discarding pong
[2020-09-05T09:28:52.329Z] [DEBUG] >>> local: HandshakePacket()
[2020-09-05T09:28:52.332Z] [DEBUG] <<< local* {"stamp":1599298132}
[2020-09-05T09:28:52.332Z] [DEBUG] >>> local* {"stamp":1599298132}
[2020-09-05T09:28:52.333Z] [DEBUG] <<= local: handshake complete
[2020-09-05T09:28:52.333Z] [DEBUG] >>> local: {"method":"set_language","params":[2],"id":4006}
[2020-09-05T09:28:52.335Z] [DEBUG] <<< local* {"stamp":1599298132}
[2020-09-05T09:28:52.336Z] [DEBUG] local: Discarding pong
[2020-09-05T09:28:52.840Z] [DEBUG] local request 4005 get_prop timed out
[... 107 lines cut ...]
[2020-09-05T09:31:39.626Z] [DEBUG] <<< cloud: {"partner_id":"","params":{"hw_ver":"Linux","fw_ver":"3.5.3_0017","ap":{"ssid":"Keskuk","bssid":"24:58:6e:ae:1d:a0"},"netif":{"localIp":"192.168.1.7","mask":"255.255.255.0","gw":"192.168.1.1"},"model":"viomi.vacuum.v8","mac":"64:90:C1:68:D2:14","token":"36667279456d305a594d373530506c42","life":934},"id":1686093636,"method":"_otc.info"}
[2020-09-05T09:31:39.626Z] [DEBUG] >>> cloud: {"id":1686093636,"result":{"otc_list":[{"ip":"110.43.0.83","port":8053}],"otc_test":{"list":[{"ip":"110.43.0.83","port":8053}],"interval":1800,"firsttest":1193}}}
[2020-09-05T09:31:44.099Z] [DEBUG] <<< cloud: {"partner_id":"","params":{"hw_ver":"Linux","fw_ver":"3.5.3_0017","ap":{"ssid":"Keskuk","bssid":"24:58:6e:ae:1d:a0"},"netif":{"localIp":"192.168.1.7","mask":"255.255.255.0","gw":"192.168.1.1"},"model":"viomi.vacuum.v8","mac":"64:90:C1:68:D2:14","token":"36667279456d305a594d373530506c42","life":934},"id":1686093636,"method":"_otc.info"}
[2020-09-05T09:31:44.099Z] [DEBUG] >>> cloud: {"id":1686093636,"result":{"otc_list":[{"ip":"110.43.0.83","port":8053}],"otc_test":{"list":[{"ip":"110.43.0.83","port":8053}],"interval":1800,"firsttest":1193}}}
[2020-09-05T09:31:59.585Z] [DEBUG] <<< cloud* {"stamp":1599298319}
[2020-09-05T09:31:59.585Z] [DEBUG] >>> cloud* {"stamp":1599298319}
[2020-09-05T09:31:59.633Z] [DEBUG] <<< cloud: {"partner_id":"","params":{"hw_ver":"Linux","fw_ver":"3.5.3_0017","ap":{"ssid":"Keskuk","bssid":"24:58:6e:ae:1d:a0"},"netif":{"localIp":"192.168.1.7","mask":"255.255.255.0","gw":"192.168.1.1"},"model":"viomi.vacuum.v8","mac":"64:90:C1:68:D2:14","token":"36667279456d305a594d373530506c42","life":954},"id":1184778613,"method":"_otc.info"}
[2020-09-05T09:31:59.633Z] [DEBUG] >>> cloud: {"id":1184778613,"result":{"otc_list":[{"ip":"110.43.0.83","port":8053}],"otc_test":{"list":[{"ip":"110.43.0.83","port":8053}],"interval":1800,"firsttest":1193}}}
[2020-09-05T09:32:05.393Z] [DEBUG] <<< cloud: {"partner_id":"","params":{"hw_ver":"Linux","fw_ver":"3.5.3_0017","ap":{"ssid":"Keskuk","bssid":"24:58:6e:ae:1d:a0"},"netif":{"localIp":"192.168.1.7","mask":"255.255.255.0","gw":"192.168.1.1"},"model":"viomi.vacuum.v8","mac":"64:90:C1:68:D2:14","token":"36667279456d305a594d373530506c42","life":954},"id":1184778613,"method":"_otc.info"}
[2020-09-05T09:32:05.393Z] [DEBUG] >>> cloud: {"id":1184778613,"result":{"otc_list":[{"ip":"110.43.0.83","port":8053}],"otc_test":{"list":[{"ip":"110.43.0.83","port":8053}],"interval":1800,"firsttest":1193}}}
[2020-09-05T09:32:19.581Z] [DEBUG] <<< cloud* {"stamp":1599298339}
[2020-09-05T09:32:19.582Z] [DEBUG] cloud: Discarding pong
[2020-09-05T09:32:25.896Z] [INFO] Valetudo shutdown in progress...
[2020-09-05T09:32:25.896Z] [DEBUG] Webserver shutdown in progress...
[2020-09-05T09:32:30.286Z] [DEBUG] Webserver shutdown done
[2020-09-05T09:32:30.286Z] [DEBUG] MiioVacuum shutdown in progress...
[2020-09-05T09:32:30.287Z] [DEBUG] cloud socket shutdown in progress...
[2020-09-05T09:32:30.289Z] [DEBUG] cloud socket shutdown done
[2020-09-05T09:32:30.289Z] [DEBUG] local socket shutdown in progress...
[2020-09-05T09:32:30.289Z] [DEBUG] local socket shutdown done
[2020-09-05T09:32:30.290Z] [DEBUG] MiioVacuum shutdown done.
[2020-09-05T09:32:30.290Z] [INFO] Valetudo shutdown done
[2020-09-05T09:32:30.290Z] [INFO] exiting with code 0...

```

rumpeltux, 2020-09-18:
You seem to be running a pretty old firmware of the robot. Can you update to anything [between 3.5.3_0044 and 3.5.3_0047](https://itooktheredpill.irgendwo.org/2020/viomi-firmware-update-analysis/) try again?

rumpeltux, 2020-09-18:
Log looks good 
```
# start zone clean, room 0
[2020-09-05T09:29:10.754Z] [DEBUG] >>> cloud: {"method":"set_mode_withroom","params":[0,1,0],"id":5}
# pause
[2020-09-05T09:29:35.979Z] [DEBUG] >>> cloud: {"method":"set_mode","params":[0,2],"id":8}
# command was not acked in time, retry
[2020-09-05T09:29:42.256Z] [DEBUG] >>> cloud: {"method":"set_mode","params":[0,2],"id":10}
# command was not acked in time, retry
[2020-09-05T09:29:47.334Z] [DEBUG] >>> cloud: {"method":"set_mode","params":[0,2],"id":12}
# continue (start zone clean again)
[2020-09-05T09:29:49.227Z] [DEBUG] >>> cloud: {"method":"set_mode_withroom","params":[0,1,0],"id":14}
# stop
[2020-09-05T09:29:55.986Z] [DEBUG] >>> cloud: {"method":"set_mode","params":[0,0],"id":19}
# command was not acked in time, retry
[2020-09-05T09:29:58.615Z] [DEBUG] >>> cloud: {"method":"set_mode","params":[0,0],"id":22}
```

Just to confirm, for this you
1. started zone clean
2. pressed pause
3. pressed continue
4. pressed stop
right?

Maybe the retries cause the robot to lose state about the current operation that can be continued. Anyways, this stuff is also very fragile when using the physical button, so not sure we can do anything about that.

Cleaning by coords seems to be broken in recent Valetudo versions, as a quick fix, you can revert to 0.5.3.

Kaan88, 2020-09-18:
Hi, mine is the global version so that's actually the latest firmware, but it is missing the latest features that the chinese version has.

STYJ02YM - chinese version
STY**T**J02YM - global

So do you think the issues could be related older firmware? Any way I can be more helpful for better support?

About the pause issue, when I press Pause continue doesn't show up. There is just Pause, Home, Stop.

And I figured 0.5.3 might work better and tried it already but it doesn't work with viomi.vacuum.v8, seems to only work with v7.

codingstube, 2020-09-19:
It seems I have the same problem with the v8 model.

Kaan88, 2020-09-22:
Pause works fine with python-miio by the way. I just send pause and then start command, same isn't true for valetudo.

mikoladz, 2020-12-08:
For me room cleaning works, but only if I choose one room at a time. If I choose more than one room it throws some errors and cleaning does not start.

Hypfer, 2021-01-11:
Since pretty much the whole codebase changed recently, please re-check if this is still an issue in [the latest beta](https://github.com/Hypfer/Valetudo/releases/tag/2021.01.0b0) and if yes open a new issue.

Thank you!

closed 2021-01-11 by Hypfer