Hypfer/Valetudo #598: Expired certificate on MQTT server makes valetudo constantly restart.

Pocke01, 2020-09-13:
## Describe the bug

If SSL/TLS is enabled and the certificate on the MQTT server expires it causes valetudo to restart. 
Makes it impossible to connect to the web ui. 

## To Reproduce

1. Install expired certificate on MQTT server.  
2. Restart valetudo. 
3. check upstart/valetudo.log  to see that valetudo restarts as soon as the mqtt process reports "expired certificate on mqtt server"

### Screenshots

### Vacuum Model

Gen2/S5

### Valetudo Version

0.6.1

### Firmware Version

2008

### User-Agent

## Expected behavior

A error msg som

## Additional context

Hypfer, 2020-09-13:
Could you share the exact logfile contents?

Pocke01, 2020-09-13:
This is from valetudo.log. Anything more needed?

```[2020-09-13T10:57:11.395Z] [INFO] Loading configuration file: /mnt/data/valetudo/config.json
[2020-09-13T10:57:11.420Z] [INFO] Starting Valetudo for Vacuum Model roborock.vacuum.s5
[2020-09-13T10:57:11.421Z] [INFO] DeviceId 261700668
[2020-09-13T10:57:11.421Z] [INFO] IP 127.0.0.1
[2020-09-13T10:57:11.422Z] [INFO] CloudSecret ---
[2020-09-13T10:57:11.424Z] [INFO] LocalSecret ---
[2020-09-13T10:57:11.425Z] [INFO] JS Runtime Version v14.4.0
[2020-09-13T10:57:11.427Z] [INFO] Max Heap Size: 33.5 MiB
[2020-09-13T10:57:12.062Z] [INFO] Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053
[2020-09-13T10:57:12.066Z] [INFO] Webserver running on port 80
[2020-09-13T10:57:12.127Z] [ERROR] MQTT error: Error: certificate has expired
[2020-09-13T10:57:20.941Z] [INFO] Loading configuration file: /mnt/data/valetudo/config.json
[2020-09-13T10:57:20.966Z] [INFO] Starting Valetudo for Vacuum Model roborock.vacuum.s5
[2020-09-13T10:57:20.967Z] [INFO] DeviceId 261700668
[2020-09-13T10:57:20.968Z] [INFO] IP 127.0.0.1
[2020-09-13T10:57:20.968Z] [INFO] CloudSecret ---
[2020-09-13T10:57:20.971Z] [INFO] LocalSecret ---
[2020-09-13T10:57:20.971Z] [INFO] JS Runtime Version v14.4.0
[2020-09-13T10:57:20.973Z] [INFO] Max Heap Size: 33.5 MiB
[2020-09-13T10:57:21.622Z] [INFO] Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053
[2020-09-13T10:57:21.626Z] [INFO] Webserver running on port 80
[2020-09-13T10:57:26.346Z] [ERROR] MQTT error: Error: certificate has expired
[2020-09-13T10:57:35.134Z] [INFO] Loading configuration file: /mnt/data/valetudo/config.json
[2020-09-13T10:57:35.159Z] [INFO] Starting Valetudo for Vacuum Model roborock.vacuum.s5
[2020-09-13T10:57:35.160Z] [INFO] DeviceId 261700668
[2020-09-13T10:57:35.161Z] [INFO] IP 127.0.0.1
[2020-09-13T10:57:35.161Z] [INFO] CloudSecret ---
[2020-09-13T10:57:35.164Z] [INFO] LocalSecret ---
[2020-09-13T10:57:35.165Z] [INFO] JS Runtime Version v14.4.0
[2020-09-13T10:57:35.166Z] [INFO] Max Heap Size: 33.5 MiB
[2020-09-13T10:57:35.794Z] [INFO] Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053
[2020-09-13T10:57:35.799Z] [INFO] Webserver running on port 80
[2020-09-13T10:57:35.918Z] [ERROR] MQTT error: Error: certificate has expired
[2020-09-13T10:57:44.766Z] [INFO] Loading configuration file: /mnt/data/valetudo/config.json```

Hypfer, 2020-09-13:
Thank you.

It shouldn't restart when this happens. There's no code that would cause that 🤔
I assume it segfaults?

What is the output if you stop the service and run `/usr/local/bin/valetudo` manually?

Pocke01, 2020-09-13:
This happens

[2020-09-13T11:13:05.890Z] [ERROR] MQTT error: Error: certificate has expired
Segmentation fault (core dumped)

Hypfer, 2020-09-13:
Okay so for whatever reason, the getaddrinfo syscall of the packaged nodejs runtime fails and causes a segfault of the whole application.

This is why there is this hack:
https://github.com/Hypfer/Valetudo/blob/8a151ba7d7c523395f19bb8296c51813000f96a9/lib/DnsHack.js

which overrides the call to `dns.lookup`, checks if it is MQTT related (since I don't want to break other functionality) and transparently remaps it to `dns.resolve4/6` which doesn't use this syscall and therefore doesn't crash.

Apparently, the detection logic is flawed and doesn't detect this.

Since there's some work required to create a test setup, the easiest way would be if you could run the attached valetudo binary, which is the current master + added DNS logging and report back with its output.
[valetudo_dns_logs.zip](https://github.com/Hypfer/Valetudo/files/5213993/valetudo_dns_logs.zip)

The log will probably rather large so you might want to use this html tag to create a collapsible spoiler
https://www.w3schools.com/tags/tag_summary.asp

Pocke01, 2020-09-13:
The only error thats repeating is this.
The ip is the Vacuums ip. 

<details>
  <summary>Log</summary>

```
[2020-09-13T12:17:25.628Z] [INFO] dns.lookup call {
  hostname: '192.168.20.210',
  options: 4,
  stack: 'Error\n' +
    '    at lookupResolveHack (/snapshot/Valetudo_Github/lib/DnsHack.js:26:16)\n' +
    '    at UDP.lookup4 (internal/dgram.js:18:10)\n' +
    '    at Socket.send (dgram.js:635:18)\n' +
    '    at /snapshot/Valetudo_Github/lib/miio/MiioSocket.js:178:25\n' +
    '    at new Promise (<anonymous>)\n' +
    '    at MiioSocket.sendMessage (/snapshot/Valetudo_Github/lib/miio/MiioSocket.js:153:16)\n' +
    '    at Dummycloud.handleMessage (/snapshot/Valetudo_Github/lib/miio/Dummycloud.js:46:33)\n' +
    '    at Socket.<anonymous> (/snapshot/Valetudo_Github/lib/miio/MiioSocket.js:128:26)\n' +
    '    at Socket.emit (events.js:315:20)\n' +
    '    at UDP.onMessage [as onmessage] (dgram.js:908:8)'
}
[2020-09-13T12:17:25.972Z] [INFO] dns.lookup call {
  hostname: 'xxxxx.xxxxx.cc',
  options: { family: undefined, hints: 32 },
  stack: 'Error\n' +
    '    at lookupResolveHack (/snapshot/Valetudo_Github/lib/DnsHack.js:26:16)\n' +
    '    at net.js:1039:5\n' +
    '    at defaultTriggerAsyncIdScope (internal/async_hooks.js:378:12)\n' +
    '    at lookupAndConnect (net.js:1038:3)\n' +
    '    at TLSSocket.Socket.connect (net.js:974:5)\n' +
    '    at Object.connect (_tls_wrap.js:1608:13)\n' +
    '    at Object.buildBuilder (/snapshot/Valetudo_Github/node_modules/mqtt/lib/connect/tls.js:17:20)\n' +
    '    at MqttClient.wrapper [as streamBuilder] (/snapshot/Valetudo_Github/node_modules/mqtt/lib/connect/index.js:152:36)\n' +
    '    at MqttClient._setupStream (/snapshot/Valetudo_Github/node_modules/mqtt/lib/client.js:297:22)\n' +
    '    at MqttClient._reconnect (/snapshot/Valetudo_Github/node_modules/mqtt/lib/client.js:931:8)'
}
[2020-09-13T12:17:26.018Z] [ERROR] MQTT error: Error: certificate has expired
[2020-09-13T12:17:27.035Z] [INFO] dns.lookup call {
  hostname: 'xxxx.xxxx.cc',
  options: { family: undefined, hints: 32 },
  stack: 'Error\n' +
    '    at lookupResolveHack (/snapshot/Valetudo_Github/lib/DnsHack.js:26:16)\n' +
    '    at net.js:1039:5\n' +
    '    at defaultTriggerAsyncIdScope (internal/async_hooks.js:378:12)\n' +
    '    at lookupAndConnect (net.js:1038:3)\n' +
    '    at TLSSocket.Socket.connect (net.js:974:5)\n' +
    '    at Object.connect (_tls_wrap.js:1608:13)\n' +
    '    at Object.buildBuilder (/snapshot/Valetudo_Github/node_modules/mqtt/lib/connect/tls.js:17:20)\n' +
    '    at MqttClient.wrapper [as streamBuilder] (/snapshot/Valetudo_Github/node_modules/mqtt/lib/connect/index.js:152:36)\n' +
    '    at MqttClient._setupStream (/snapshot/Valetudo_Github/node_modules/mqtt/lib/client.js:297:22)\n' +
    '    at MqttClient._reconnect (/snapshot/Valetudo_Github/node_modules/mqtt/lib/client.js:931:8)' 
}
```

</details>

Hypfer, 2020-09-13:
After the last dns.lookup logline the process just segfaults?
That's very strange, because the code checks for `lookupAndConnect` being part of the stacktrace which it is.

I guess there's no way around replicating that setup and attaching a debugger then :(

Pocke01, 2020-09-13:
Yes it segfaults.   
As soon as i change the certificate on the mqtt server to a valid cert its starts to work again.

Hypfer, 2021-01-11:
Since pretty much the whole codebase changed recently, please re-check if this is still an issue in [the latest beta](https://github.com/Hypfer/Valetudo/releases/tag/2021.01.0b0) and if yes open a new issue.

Thank you!

closed 2021-01-11 by Hypfer