Hypfer/Valetudo #599: Keeps Reverting to Medium Power Mode When Using Timers

shepler007, 2020-09-17:
## Describe the bug

Keeps Reverting to Medium Power Mode When Using Timers.  I have a 10 am timer setup to vacuum my home and ever since I installed version 0.6.1 when it starts I have to go into the UI and change it to max.

## To Reproduce

### Screenshots

### Vacuum Model
Gen2/S5

### Valetudo Version
e.g. 0.6.1

### Firmware Version
002008

### User-Agent

## Expected behavior

## Additional context

Hypfer, 2020-09-17:
You're completely right

The medium fan speed has been hardcoded when basic timer support for newer firmwares was added, since it's apparently not possible anymore to have a timer without a fanspeed.

https://github.com/Hypfer/Valetudo/blob/8a151ba7d7c523395f19bb8296c51813000f96a9/lib/devices/RoborockS5.js#L52-L55

This will also be addressed when the timers are reworked as mentioned in #592

Hypfer, 2021-01-09:
Timers have been removed in the latest version

closed 2021-01-09 by Hypfer