Hypfer/Valetudo #603: Roborock S5/S6 Support Now that Root is Available?

kars85, 2020-09-21:
Pretty cut and dry question, but just curious if I go through the disassembly of my S5 & S6's to root, if I'll be able to load valetudo on without any fuss? I noticed it wasn't explicitly noted in the project wiki, but root is only a couple months old on these robots.

Thanks in advance!

rimrul, 2020-09-22:
Root on the S5 isn't new and doesn't require disassembly. Do you mean the S5 max?

kars85, 2020-09-22:
> 
> 
> Root on the S5 isn't new and doesn't require disassembly. Do you mean the S5 max?

My mistake - I haven't been following the S5 root as closely (despite owning one). Was more interested in the S6 (one one of those for my main level, too).

stecky, 2020-09-23:
Are you saying that root is now possible on the S5 Max or just the S6?

ni-ko-b, 2020-09-28:
Other than issue #606 I had no problems at all with my S6.

closed 2020-10-04 by Hypfer

nununo, 2020-10-05:
Hi, can you please confirm that S5 Max is now also compatible with Valetudo? If so, wouldn't it make sense to update the website? Because it still clearly states that S5 Max and S6 are not compatible. Thanks!

I'm deciding which model to buy and I'm sticking to models compatible with Valetudo 😄

Hypfer, 2020-10-05:
Maybe the fact that the website isn't updated with support for the S5 Max and this issue being closed might give away something?

nununo, 2020-10-05:
@Hypfer fair point 😄  

But still, @ni-ko-b seems to have it working on S6 almost perfectly.

Ok, let me then ask it differently: considering what I've read in this issue, what are the expectations to have it working on S5 Max and S6 in a near future?

Thanks

stecky, 2020-10-05:
> Maybe the fact that the website isn't updated with support for the S5 Max and this issue being closed might give away something?

It was a fair question. You closed the issue, but didn't comment on your reason for doing so. Even a simple "won't fix" or "duplicate issue" would have been helpful

Hypfer, 2020-10-05:
Yeah well you see not answering at all is the only was I can answer without being unfriendly regarding issues like these.

As stated multiple times, this issue tracker is _not_ for support requests. Please direct them elsewhere where other people can handle them, because otherwise this issue tracker becomes unusable.

There's a reason why there are exactly two issue templates of which only one is allowed to be here
If none if those one issue templates matches, then it doesn't belong here.

It would be insanely helpful if people would stop annoying the fuck out of me.
That's all I'm asking for in exchange for this software project. _Don't_

Frankly, I feel like I'm constantly being punished for having this as an open source project instead of simply building it for myself and keeping it on my local server without read access for anyone.

There's just no end to this. I can't just leave all the support requests open because they'd pile up, but I also cannot answer them again and again each time saying exactly the same things.

What should I do? Delete everything?
That seems like the only option to get people to stop.

stecky, 2020-10-05:
fair enough... Perhaps a simple statement like `support issues should be directed to http://xyz.com. Closing this issue as it is out of scope` could help ward off some of these questions. I for one found this through a google search for 'S5 Max root'. I had never even heard of Valetudo prior to that and was only interested in finding information about rooting my S5 max. So I would have no reason to know that support requests don't belong here. Just my two cents. :) Thanks for making this available. maybe I'll get to try it out some day if I can ever root my vac. Cheers

Hypfer, 2020-10-05:
```
Hi there,

your issue has been closed, because it is either a support request or something else that would usually be handled by 1st level support.

Since this is not a commercial project, but rather just something I've built for myself and then shared with the world for free, I really can't provide said support.

Therefore, please direct this request somewhere else, where other users might be able to help you.
The Telegram group would be one option but certainly isn't the only one.

Thank you for understanding as well as your interest in the project.
```

This should be a suffient response I hope.

So anyways to at least handle this topic:

S6 works with the latest build, however there are still some issues remaining such as the initial provisioning failing as well as the complete lack of documentation (as you can see obviously).

S5 Max however is a different story. Roborock greatly cut costs with that model and caused by that, there is no safe rooting method available (yet)

As you can see [here](https://github.com/Hypfer/Valetudo/commit/da53214c0b60deef15d08dcb18e8d7fc69b2a603) the docs have been updated to also include the S6 in the #584 branch. This wasn't merged yet, because there's still much to do which I don't have the time for atm.
The time and energy I do have is spent on 1st level support issues which adds to the overall frustration (as you can also see)

Personally, I would try getting an S5 since installing valetudo on it is just so much easier.

Hypfer, 2020-10-05:
https://github.com/Hypfer/Valetudo/issues/615

Look at it. It happend again _4 minutes_ after I've written this. 

I mean I'm really happy that I apparently can help people with achieving whatever they might want to do with their vacuum robots but still. _4 Minutes._

At least I do have a canned response now. That was a very good suggestion which I did get before but never got to implement previously

nununo, 2020-10-06:
@Hypfer, I fully understand your frustration. Sorry. And above all thank you for your amazing work.

Still, if you put yourself in my shoes, there is no clear way for me to understand that I should not be asking that question there.

The canned response is a great solution.

But, if you feel overwhelmed (and I'm sure no one wants you to feel that way) by these random support requests, (and if you also get tired of pasting your new canned response 😄) I  suggest that you should just ignore them (the issue is closed anyway). That way you don't become angry and frustrated and at the same time you also won't hold it against someone who just arrived here and has no clear way to understand he/she should not be asking these questions here.

Again, thanks and sorry for burdening you with this!!