Hypfer/Valetudo #604: Viomi V7: No map visible - Script not automatically executed

XobrekX, 2020-09-23:
## Describe the bug

## To Reproduce

### Screenshots

### Vacuum Model

Viomi V7

### Valetudo Version

0.6.1

### Firmware Version

### User-Agent

Chrome

## Expected behavior

I want the script to be automatically executed as well as the map to be displayed
[log.txt](https://github.com/Hypfer/Valetudo/files/5271892/log.txt)

I really hope you can help me. I have no clue what else to try.

## Additional context

Here are a few logs:

```
<30>Sep 24 04:51:14 valetudo[981]: [2020-09-23T20:51:14.814Z] [DEBUG] <<< local: {"result":["ok"],"id":2004}
<30>Sep 24 04:53:58 valetudo[1012]: [2020-09-23T20:53:58.957Z] [INFO] Loading configuration file: /mnt/UDISK/config/valetudo.json
<30>Sep 24 04:53:58 valetudo[1012]: [2020-09-23T20:53:58.989Z] [INFO] Starting Valetudo for Vacuum Model viomi.vacuum.v7
<30>Sep 24 04:53:58 valetudo[1012]: [2020-09-23T20:53:58.990Z] [INFO] DeviceId 278012635
<30>Sep 24 04:53:58 valetudo[1012]: [2020-09-23T20:53:58.991Z] [INFO] IP 192.168.178.28
<30>Sep 24 04:53:58 valetudo[1012]: [2020-09-23T20:53:58.991Z] [INFO] CloudSecret G8rnkYGRmjrzfN3W
<30>Sep 24 04:53:58 valetudo[1012]: [2020-09-23T20:53:58.992Z] [INFO] LocalSecret YGS3XrwjACcDT0Qv
<30>Sep 24 04:53:58 valetudo[1012]: [2020-09-23T20:53:58.993Z] [INFO] JS Runtime Version v14.4.0
<30>Sep 24 04:53:58 valetudo[1012]: [2020-09-23T20:53:58.995Z] [INFO] Max Heap Size: 33.5 MiB
<30>Sep 24 04:53:58 valetudo[1012]: [2020-09-23T20:53:58.995Z] [INFO] Node Flags: --max-old-space-size=32
<30>Sep 24 04:53:59 valetudo[1012]: [2020-09-23T20:53:59.105Z] [DEBUG] >>> local: HandshakePacket()
<30>Sep 24 04:53:59 valetudo[1012]: [2020-09-23T20:53:59.206Z] [INFO] Dummycloud is spoofing 110.43.0.83:8053 on 127.0.0.1:8053
<30>Sep 24 04:53:59 valetudo[1012]: [2020-09-23T20:53:59.210Z] [INFO] Webserver running on port 8080
<30>Sep 24 04:53:59 valetudo[1012]: [2020-09-23T20:53:59.225Z] [DEBUG] <<< local* {"stamp":1600894439}
<30>Sep 24 04:53:59 valetudo[1012]: [2020-09-23T20:53:59.226Z] [DEBUG] >>> local* {"stamp":1600894439}
<30>Sep 24 04:53:59 valetudo[1012]: [2020-09-23T20:53:59.231Z] [DEBUG] <<= local: handshake complete
<30>Sep 24 04:53:59 valetudo[1012]: [2020-09-23T20:53:59.240Z] [TRACE] >>> local: {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"],"id":1}
<30>Sep 24 04:53:59 valetudo[1012]: [2020-09-23T20:53:59.243Z] [DEBUG] >>> local: {"method":"set_language","params":[2],"id":2}
<30>Sep 24 04:53:59 valetudo[1012]: [2020-09-23T20:53:59.246Z] [DEBUG] <<< local* {"stamp":1600894439}
<30>Sep 24 04:53:59 valetudo[1012]: [2020-09-23T20:53:59.247Z] [DEBUG] local: Discarding pong
<30>Sep 24 04:53:59 valetudo[1012]: [2020-09-23T20:53:59.738Z] [DEBUG] local request 1 get_prop timed out
<30>Sep 24 04:53:59 valetudo[1012]: [2020-09-23T20:53:59.741Z] [DEBUG] >>> local: HandshakePacket()
<30>Sep 24 04:53:59 valetudo[1012]: [2020-09-23T20:53:59.743Z] [DEBUG] local request 2 set_language timed out
<30>Sep 24 04:53:59 valetudo[1012]: [2020-09-23T20:53:59.746Z] [DEBUG] <<< local* {"stamp":1600894440}
<30>Sep 24 04:53:59 valetudo[1012]: [2020-09-23T20:53:59.747Z] [DEBUG] >>> local* {"stamp":1600894440}
<30>Sep 24 04:53:59 valetudo[1012]: [2020-09-23T20:53:59.748Z] [DEBUG] <<= local: handshake complete
<30>Sep 24 04:53:59 valetudo[1012]: [2020-09-23T20:53:59.751Z] [TRACE] >>> local: {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"],"id":2003}
<30>Sep 24 04:53:59 valetudo[1012]: [2020-09-23T20:53:59.754Z] [DEBUG] >>> local: {"method":"set_language","params":[2],"id":2004}
<30>Sep 24 04:53:59 valetudo[1012]: [2020-09-23T20:53:59.757Z] [DEBUG] <<< local* {"stamp":1600894440}
<30>Sep 24 04:53:59 valetudo[1012]: [2020-09-23T20:53:59.757Z] [DEBUG] local: Discarding pong
<30>Sep 24 04:54:00 valetudo[1012]: [2020-09-23T20:54:00.251Z] [DEBUG] local request 2003 get_prop timed out
<30>Sep 24 04:54:00 valetudo[1012]: [2020-09-23T20:54:00.257Z] [DEBUG] >>> local: HandshakePacket()
<30>Sep 24 04:54:00 valetudo[1012]: [2020-09-23T20:54:00.263Z] [DEBUG] local request 2004 set_language timed out
<30>Sep 24 04:54:00 valetudo[1012]: [2020-09-23T20:54:00.277Z] [DEBUG] <<< local* {"stamp":1600894440}
<30>Sep 24 04:54:00 valetudo[1012]: [2020-09-23T20:54:00.279Z] [DEBUG] local: Discarding pong
<30>Sep 24 04:54:00 valetudo[1012]: [2020-09-23T20:54:00.560Z] [DEBUG] >>> local: HandshakePacket()
<30>Sep 24 04:54:00 valetudo[1012]: [2020-09-23T20:54:00.563Z] [DEBUG] <<< local* {"stamp":1600894441}
<30>Sep 24 04:54:00 valetudo[1012]: [2020-09-23T20:54:00.564Z] [DEBUG] >>> local* {"stamp":1600894441}
[... 122 lines cut ...]
      "deviceId": 278012635,
      "cloudSecret": "G8rnkYGRmjrzfN3W",
      "localSecret": "YGS3XrwjACcDT0Qv"
    }
  },
  "debug": {
    "memoryStatInterval": false
  }
}
```

```
root@TinaLinux:~# cat /etc/rc.d/S51valetudo
iptables         -F OUTPUT
iptables  -t nat -F OUTPUT
# for local development enter your local development host here
# and change port to 8080
dest=127.0.0.1
port=8080
for host in 110.43.0.83 110.43.0.85; do
  iptables  -t nat -A OUTPUT -p tcp --dport 80   -d $host -j DNAT --to-destination $dest:$port
  iptables  -t nat -A OUTPUT -p udp --dport 8053 -d $host -j DNAT --to-destination $dest:8053
  iptables         -A OUTPUT                     -d $host/32  -j REJECT
done
```

schannall, 2020-10-05:
Hi,
for the map upload host you have:
 "map_upload_host": "http://110.43.0.83",
It _should_ not make a difference as iptables will overwrite it - but you might want to change it to default (127.0.0.1).

Also your S51 Script is executable? I guess so as you can execute it but just to be safe :)

XobrekX, 2020-10-05:
Hi,
Good point, I'll try to change it to local this evening.

Yes it is executable :D

schannall, 2020-10-07:
@XobrekX : were you able to verify that the error still persists for you? Can you also Check if all your Settings are as in the Trobuleshooting?

https://valetudo.cloud/pages/troubleshooting.html

Thanks

rumpeltux, 2020-10-10:
Per your logs the cloud service doesn’t connect.
Which is expected since when the iptables script isn’t executed on boot, the robot will connect to the original xiaomi cloud and use different IPs from then.

Make sure that the /etc/rc.d/S51valetudo is correctly setup. It looks like it’s missing the first line which should be
```
#!/bin/sh
```

And double-check that it’s executable
```
# ls -al /etc/rc.d/S51valetudo
-rwxr-xr-x    1 root     root           423 Oct  7 18:56 /etc/rc.d/S51valetudo
```

closed 2020-10-10 by rumpeltux

XobrekX, 2020-10-11:
Sorry for not answering. I was kind of busy the last days. 
I reallly forgot the "#!/bin/bash in front of the S51 script. Thank thins works perfectly now!

On the other side, I am still seeing "no map data" in the map tab, even after a full cleaning. :(

I followed the stets under troubleshooting and changed the map upload path, but is has had no effect I think
By the way, for the V7 the path is different: /mnt/UDISK/config/valetudo.json