Hypfer/Valetudo #605: Reconnecting to MQTT broker takes hours after connection lost

definitio, 2020-09-25:
## Describe the bug

Valetudo needs hours to reconnect to MQTT broker after connection lost.
## To Reproduce
Steps to reproduce the behavior:
1. Stop MQTT broker
2. Wait a few minutes (it may not be necessary)
3. Wait until Valetudo will connect to broker again

### Screenshots

[image]

### Vacuum Model

Gen1

### Valetudo Version

0.6.1

### Firmware Version

4018

## Expected behavior

It should take seconds or minutes.

## Additional context

Hypfer, 2020-09-25:
When was the firmware built?

definitio, 2020-09-25:
28.07.2020

Hypfer, 2020-09-25:
This behaviour was caused by an invalid valetudo start script.

Commits to fix this were added on the 10.6.2020
https://github.com/zvldz/vacuum/commit/298a0c35d797c58476a3ed8b8e2b1b8640ce94f9#diff-97ec49e0e3b6e41ecb330776387add14

Depending on whether or not you've pulled the latest image builder, you may not have those.

So basically what happened there is that the script killed valetudo hard with no way to properly shutdown and the old mqtt connection timed out _after_ the new one was established, which caused the availability topic to receive the last will of the old connection, overwriting the new "online" status.

Please check your `/etc/init/S11valetudo` for a `-9` flag for the `killall` command

Hypfer, 2020-09-25:
[image]

I can restart my mqtt broker multiple times and it always reconnects properly

definitio, 2020-09-25:
I can restart broker too and it will reconnect, but if I disable broker for 5 minutes it will never connect again.

Hypfer, 2021-01-11:
Since pretty much the whole codebase changed recently, please re-check if this is still an issue in [the latest beta](https://github.com/Hypfer/Valetudo/releases/tag/2021.01.0b0) and if yes open a new issue.

Thank you!

closed 2021-01-11 by Hypfer