Hypfer/Valetudo #606: S9: no map data is available

ni-ko-b, 2020-09-27:
## Describe the bug
No map data is shown

## To Reproduce

1. Go to valetudo in your browser (http://192.168.../)
2. Click on "map"
4. Note how no map is shown. Instead "no map data" appears

### Screenshots

### Vacuum Model
Roborock S6

### Valetudo Version
0.6.1

### Firmware Version
1998

### User-Agent
Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/85.0.4183.121 Safari/537.36

## Expected behavior
After docking the robot should display the last created map

## Additional context

- Persistent data has been disabled, map data reset and enabled again

```
[root@rockrobo ~]# cat /mnt/data/valetudo/config.json
{
  "spots": [],
  "mqtt": {
    "enabled": true,
    "server": "*removed*",
    "port": "1883",
    "clientId": "roborock s6",
    "username": "",
    "password": "",
    "usetls": false,
    "ca": "",
    "clientCert": "",
    "clientKey": "",
    "qos": 0,
    "identifier": "rockrobo",
    "topicPrefix": "valetudo",
    "autoconfPrefix": "homeassistant",
    "provideMapData": true
  },
  "webserver": {
    "port": 80
  },
  "dummycloud": {
    "spoofedIP": "203.0.113.1",
    "bindIP": "127.0.0.1"
  },
  "httpAuth": {
    "enabled": false,
    "username": "valetudo",
    "password": "valetudo"
  },
  "allowSSHKeyUpload": true,
  "map_upload_host": "http://127.0.0.1",
  "logLevel": "trace",
  "model": {
    "type": "auto",
    "embedded": true
  },
[... 220 lines cut ...]
2980265 rrlogd T488 found /mnt/data/rockrobo/rrlog/000103.20200927152426314_R0314S01400694_2020060901REL/APP_PROXY_normal.tar. will not delete
2980266 rrlogd T488 found /mnt/data/rockrobo/rrlog/000102.20200927142335387_R0314S01400694_2020060901REL/APP_PROXY_normal.tar. will not delete
2980266 rrlogd T488 found /mnt/data/rockrobo/rrlog/000101.20200927125259843_R0314S01400694_2020060901REL/watchdog.pl3.rr2.xz. will not delete
2980266 rrlogd T488 found /mnt/data/rockrobo/rrlog/000100.20200927111254543_R0314S01400694_2020060901REL/000100_1601168707989_v1.pl9.rr.stat. will not delete
2980267 rrlogd T488 found /mnt/data/rockrobo/rrlog/000099.20200927100254751_R0314S01400694_2020060901REL/APP_PROXY_normal.tar. will not delete
2980267 rrlogd T488 found /mnt/data/rockrobo/rrlog/000098.20200927010456565_R0314S01400694_2020060901REL/watchdog.pl3.rr2.xz. will not delete
2980267 rrlogd T488 found /mnt/data/rockrobo/rrlog/000097.20200926154014711_R0314S01400694_2020060901REL/watchdog.pl3.rr2.xz. will not delete
2980268 rrlogd T488 found /mnt/data/rockrobo/rrlog/000096.20200926153131587_R0314S01400694_2020060901REL/APP_PROXY_normal.tar. will not delete
2980269 rrlogd T488 found /mnt/data/rockrobo/rrlog/000108.20200927171707443_R0314S01400694_2020060901REL/watchdog.pl3.rr2.xz. will not delete
2980270 rrlogd T488 found /mnt/data/rockrobo/rrlog/000107.20200927163502106_R0314S01400694_2020060901REL/APP_PROXY_normal.tar. will not delete
2980270 rrlogd T488 found /mnt/data/rockrobo/rrlog/000106.20200927161208622_R0314S01400694_2020060901REL/watchdog.pl3.rr2.xz. will not delete
2980271 rrlogd T488 found /mnt/data/rockrobo/rrlog/000105.20200927161139850_R0314S01400694_2020060901REL/APP_PROXY_normal.tar. will not delete
2980271 rrlogd T488 found /mnt/data/rockrobo/rrlog/000104.20200927152555588_R0314S01400694_2020060901REL/watchdog.pl3.rr2.xz. will not delete
2980271 rrlogd T488 found /mnt/data/rockrobo/rrlog/000103.20200927152426314_R0314S01400694_2020060901REL/APP_PROXY_normal.tar. will not delete
2980272 rrlogd T488 found /mnt/data/rockrobo/rrlog/000102.20200927142335387_R0314S01400694_2020060901REL/APP_PROXY_normal.tar. will not delete
2980272 rrlogd T488 found /mnt/data/rockrobo/rrlog/000101.20200927125259843_R0314S01400694_2020060901REL/watchdog.pl3.rr2.xz. will not delete
2980272 rrlogd T488 found /mnt/data/rockrobo/rrlog/000100.20200927111254543_R0314S01400694_2020060901REL/000100_1601168707989_v1.pl9.rr.stat. will not delete
2980273 rrlogd T488 found /mnt/data/rockrobo/rrlog/000099.20200927100254751_R0314S01400694_2020060901REL/APP_PROXY_normal.tar. will not delete
2980273 rrlogd T488 found /mnt/data/rockrobo/rrlog/000098.20200927010456565_R0314S01400694_2020060901REL/watchdog.pl3.rr2.xz. will not delete
2980274 rrlogd T488 found /mnt/data/rockrobo/rrlog/000097.20200926154014711_R0314S01400694_2020060901REL/watchdog.pl3.rr2.xz. will not delete
2980274 rrlogd T488 found /mnt/data/rockrobo/rrlog/000096.20200926153131587_R0314S01400694_2020060901REL/APP_PROXY_normal.tar. will not delete
3007515 rrlogd T456 fl_notify_self: msglen=4, offset=4, ret=4, sig=4
```

- It seems like the robot does not upload the map data to valetudo ("cannot get log server, abort upload"). Also no attempts by valetudo itself seem to be logged. I assume there is some issue with the iptables configuration.

ni-ko-b, 2020-09-27:
[valetudo.log](https://github.com/Hypfer/Valetudo/files/5288757/valetudo.log)

Hypfer, 2020-09-27:
Please try this solution

```
Dgi, [27.08.20 21:58]
[Forwarded from Dgi]
#!/bin/bash

echo ssid=\"MySSID\" > /mnt/data/miio/wifi.conf
echo psk=\"MyPassword\" >> /mnt/data/miio/wifi.conf
echo key_mgmt=\"WPA\" >> /mnt/data/miio/wifi.conf
echo uid=0 >> /mnt/data/miio/wifi.conf
echo region=us >> /mnt/data/miio/wifi.conf
echo cfg_by=miot >> /mnt/data/miio/wifi.conf
echo 0 > /mnt/data/miio/device.uid
echo "us" > /mnt/data/miio/device.country

Dgi, [27.08.20 21:59]
(replace ssid, password and region, if needed)
```

ni-ko-b, 2020-09-27:
Sorry I forgot to mention that the robot was already provisioned:

```
[root@rockrobo ~]# cat /mnt/data/miio/wifi.conf
ssid="*removed*"
psk="*removed*"
key_mgmt=WPA
uid=0
region=
cfg_by=miot
```

Hypfer, 2020-09-27:
Try adding the region then. Afaik the missing value there is causing the miio client to not connect at all

ni-ko-b, 2020-09-27:
ok thank you. I just patched the files and will report back

ni-ko-b, 2020-09-27:
it works now. thank you very much for the incredible fast support and this awesome software :) is it possible to donate?

Hypfer, 2020-09-27:
The best way to donate is to donate your time by supporting other users :)

This missing region issue is indeed a bug btw. Valetudo is missing something in the provisioning command

ni-ko-b, 2020-09-27:
Awesome. Glad I could help identify this :D

Hypfer, 2021-01-03:
Should be fixed with 28483abd4431cf0d38953aefeb3995e9d45ea2b8

closed 2021-01-03 by Hypfer