Hypfer/Valetudo #607: Map doesn't update when I use custom configured zones cleaning mode - viomi.vacuum.v8

Kaan88, 2020-09-28:
## Describe the bug

In full cleaning or room cleaning mode I can use the map page to follow robot's progress. But in zone cleaning mod the map doesn't update. And on the main page status is in cleaning mode at start but after a couple of seconds display changes to "paused," even though the device is cleaning.

I have the global version of the vacuum so I don't have the latest update the Chinese version has. I don't know if this issue is because of this.  3.5.3_0017 is the latest for me.

Also, the same happens with python-miio. Vacuum state is in paused like Valetudo.

## To Reproduce

Choose any zones made via zones page.

### Vacuum Model

STYTJ02YM
### Valetudo Version

 0.6.1 - latest git
### Firmware Version

 3.5.3_0017

## Expected behavior

It should update robot's position on the map like in other modes.

<details><summary>Trace Log</summary>
<p>

```
Thu Jan  1 08:04:56 1970 daemon.info valetudo[2448]: [1970-01-01T00:04:56.871Z] [INFO] Starting Valetudo for Vacuum Model viomi.vacuum.v8
Thu Jan  1 08:04:56 1970 daemon.info valetudo[2448]: [1970-01-01T00:04:56.872Z] [INFO] DeviceId 328986625
Thu Jan  1 08:04:56 1970 daemon.info valetudo[2448]: [1970-01-01T00:04:56.872Z] [INFO] IP 127.0.0.1
Thu Jan  1 08:04:56 1970 daemon.info valetudo[2448]: [1970-01-01T00:04:56.873Z] [INFO] CloudSecret h9e9Bi9fnoPJJvkb
Thu Jan  1 08:04:56 1970 daemon.info valetudo[2448]: [1970-01-01T00:04:56.876Z] [INFO] LocalSecret 6fryEm0ZYM750PlB
Thu Jan  1 08:04:56 1970 daemon.info valetudo[2448]: [1970-01-01T00:04:56.876Z] [INFO] JS Runtime Version v14.4.0
Thu Jan  1 08:04:56 1970 daemon.info valetudo[2448]: [1970-01-01T00:04:56.878Z] [INFO] Max Heap Size: 33.5 MiB
Thu Jan  1 08:04:56 1970 daemon.info valetudo[2448]: [1970-01-01T00:04:56.878Z] [INFO] Node Flags: --max-old-space-size=32
Thu Jan  1 08:04:57 1970 daemon.info valetudo[2448]: [1970-01-01T00:04:57.091Z] [INFO] Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053
Thu Jan  1 08:04:57 1970 daemon.info valetudo[2448]: [1970-01-01T00:04:57.095Z] [INFO] Webserver running on port 80
Thu Jan  1 08:05:01 1970 daemon.info valetudo[2448]: [1970-01-01T00:05:01.171Z] [INFO] Get FanSpeeds
Thu Jan  1 08:05:03 1970 daemon.info valetudo[2448]: [1970-01-01T00:05:03.753Z] [INFO] Cloud connected
Thu Jan  1 08:05:08 1970 daemon.info valetudo[2448]: [1970-01-01T00:05:08.633Z] [INFO] << cloud: ignoring response for non-pending request {"result":["ok"],"id":1}
Thu Jan  1 08:05:23 1970 daemon.info valetudo[2448]: [1970-01-01T00:05:23.234Z] [INFO] Get FanSpeeds
Thu Jan  1 08:05:40 1970 daemon.info valetudo[2448]: [1970-01-01T00:05:40.620Z] [INFO] Get FanSpeeds
Thu Jan  1 08:07:44 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:44.993Z] [TRACE] take 12@area head 9c6a8 <Buffer b7 f8 48 5f 01 00 00 00 01 00 00 00>
Thu Jan  1 08:07:44 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:44.993Z] [TRACE] AreaClean [ 1598617783, 1, 1 ]
Thu Jan  1 08:07:44 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:44.994Z] [TRACE] take 92@area content 9c6b4 <Buffer 00 00 00 00 00 00 00 00 04 00 00 00 9a 99 79 c0 cd cc 4c 3d 9a 99 79 c0 9a 99 09 c0 66 66 26 bf 9a 99 09 c0 66 66 26 bf cd cc 4c 3d 00 00 00 00 00 00 ... 42 more bytes>
Thu Jan  1 08:07:44 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:44.994Z] [TRACE] take 20@navigate 9c710 <Buffer b7 f8 48 5f 00 00 00 00 00 80 89 44 00 80 89 44 00 00 00 00>
Thu Jan  1 08:07:44 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:44.995Z] [TRACE] take 21@realtime 9c724 <Buffer b7 f8 48 5f 00 00 00 00 00 dc a1 67 c0 94 33 22 40 c2 ad ea bc>
Thu Jan  1 08:07:44 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:44.995Z] [TRACE] take 8@unknown8 9c739 <Buffer b7 f8 48 5f 00 00 00 01>
Thu Jan  1 08:07:44 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:44.996Z] [TRACE] take 4@room header 9c741 <Buffer b7 f8 48 5f>
Thu Jan  1 08:07:44 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:44.997Z] [TRACE] take 4@mapArg: 9c746 <Buffer 01 00 00 00>
Thu Jan  1 08:07:44 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:44.997Z] [TRACE] take 4@room length 9c74a <Buffer 06 00 00 00>
Thu Jan  1 08:07:44 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:44.998Z] [TRACE] room10: room1 9c74e 640846
Thu Jan  1 08:07:44 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:44.998Z] [TRACE] room11: Bedroom 9c75e 640862
Thu Jan  1 08:07:44 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:44.999Z] [TRACE] room12: Kitty toilet 9c770 640880
Thu Jan  1 08:07:44 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:44.999Z] [TRACE] room15: room0 9c787 640903
Thu Jan  1 08:07:45 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:44.999Z] [TRACE] room13: room2 9c797 640919
Thu Jan  1 08:07:45 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:45.000Z] [TRACE] room14: room3 9c7a7 640935
Thu Jan  1 08:07:45 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:45.000Z] [TRACE] take 6@unknown 9c7b7 <Buffer 01 01 00 00 00 00>
Thu Jan  1 08:07:45 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:45.000Z] [TRACE] take 12@rooms2head 9c7bd <Buffer b7 f8 48 5f 01 00 00 00 06 00 00 00>
Thu Jan  1 08:07:45 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:45.001Z] [TRACE] take 4@rooms2 length 9c7c9 <Buffer 01 00 00 00>
Thu Jan  1 08:07:45 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:45.001Z] [TRACE] take 2@room data 9c7cd <Buffer 00 00>
Thu Jan  1 08:07:45 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:45.002Z] [TRACE] take 4@unknown2 9c7cf <Buffer 00 00 04 00>
Thu Jan  1 08:07:45 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:45.002Z] [TRACE] take 50@unknown50 9c7d0 <Buffer 00 04 00 00 00 03 09 0e c0 68 91 87 40 03 09 0e c0 68 91 87 40 0b 24 38 bf 8c b9 87 40 0b 24 38 bf 8c b9 87 40 00 00 00 00 00 00 00 00 00 00 00 00 00>
Thu Jan  1 08:07:45 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:45.003Z] [TRACE] take 5@unknown5 9c802 <Buffer 00 00 00 00 00>
Thu Jan  1 08:07:45 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:45.003Z] [TRACE] take 4@pose length 9c807 <Buffer 00 00 00 00>
Thu Jan  1 08:07:45 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:45.004Z] [TRACE] end 9c80b remaining: 10920
Thu Jan  1 08:07:45 1970 daemon.info valetudo[3180]: [1970-01-01T00:07:45.049Z] [TRACE] >>> cloud: {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"],"id":7}
[... 60 lines cut ...]
Thu Jan  1 08:08:11 1970 daemon.info valetudo[3180]: [1970-01-01T00:08:11.554Z] [TRACE] take 4@room header 9c74a <Buffer b7 f8 48 5f>
Thu Jan  1 08:08:11 1970 daemon.info valetudo[3180]: [1970-01-01T00:08:11.554Z] [TRACE] take 4@mapArg: 9c74f <Buffer 01 00 00 00>
Thu Jan  1 08:08:11 1970 daemon.info valetudo[3180]: [1970-01-01T00:08:11.555Z] [TRACE] take 4@room length 9c753 <Buffer 06 00 00 00>
Thu Jan  1 08:08:11 1970 daemon.info valetudo[3180]: [1970-01-01T00:08:11.555Z] [TRACE] room10: room1 9c757 640855
Thu Jan  1 08:08:11 1970 daemon.info valetudo[3180]: [1970-01-01T00:08:11.556Z] [TRACE] room11: Bedroom 9c767 640871
Thu Jan  1 08:08:11 1970 daemon.info valetudo[3180]: [1970-01-01T00:08:11.556Z] [TRACE] room12: Kitty toilet 9c779 640889
Thu Jan  1 08:08:11 1970 daemon.info valetudo[3180]: [1970-01-01T00:08:11.557Z] [TRACE] room15: room0 9c790 640912
Thu Jan  1 08:08:11 1970 daemon.info valetudo[3180]: [1970-01-01T00:08:11.557Z] [TRACE] room13: room2 9c7a0 640928
Thu Jan  1 08:08:11 1970 daemon.info valetudo[3180]: [1970-01-01T00:08:11.557Z] [TRACE] room14: room3 9c7b0 640944
Thu Jan  1 08:08:11 1970 daemon.info valetudo[3180]: [1970-01-01T00:08:11.558Z] [TRACE] take 6@unknown 9c7c0 <Buffer 01 01 00 00 00 00>
Thu Jan  1 08:08:11 1970 daemon.info valetudo[3180]: [1970-01-01T00:08:11.558Z] [TRACE] take 12@rooms2head 9c7c6 <Buffer b7 f8 48 5f 01 00 00 00 06 00 00 00>
Thu Jan  1 08:08:11 1970 daemon.info valetudo[3180]: [1970-01-01T00:08:11.559Z] [TRACE] take 4@rooms2 length 9c7d2 <Buffer 01 00 00 00>
Thu Jan  1 08:08:11 1970 daemon.info valetudo[3180]: [1970-01-01T00:08:11.559Z] [TRACE] take 2@room data 9c7d6 <Buffer 00 00>
Thu Jan  1 08:08:11 1970 daemon.info valetudo[3180]: [1970-01-01T00:08:11.559Z] [TRACE] take 4@unknown2 9c7d8 <Buffer 00 00 04 00>
Thu Jan  1 08:08:11 1970 daemon.info valetudo[3180]: [1970-01-01T00:08:11.560Z] [TRACE] take 50@unknown50 9c7d9 <Buffer 00 04 00 00 00 03 09 0e c0 68 91 87 40 03 09 0e c0 68 91 87 40 0b 24 38 bf 8c b9 87 40 0b 24 38 bf 8c b9 87 40 00 00 00 00 00 00 0 [... cut ...]
Thu Jan  1 08:08:11 1970 daemon.info valetudo[3180]: [1970-01-01T00:08:11.562Z] [TRACE] end 9c814 remaining: 10920
Thu Jan  1 08:08:16 1970 daemon.info valetudo[3180]: [1970-01-01T00:08:16.022Z] [TRACE] >>> cloud: {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"],"id":18}
Thu Jan  1 08:08:16 1970 daemon.info valetudo[3180]: [1970-01-01T00:08:16.064Z] [TRACE] <<< cloud: {"result":[4,0,2104,100,1,0,0,0,0,12,1,1,0,0],"id":18}

```

</p>
</details>

rumpeltux, 2020-09-28:
The logfile appears incomplete (I don’t see any commands issued to the robot?), but I remember running into the issue myself:
I believe this is a bug in the robot firmware unrelated to Valetudo in which it doesn’t correctly report status in room or zone cleaning mode.

closed 2020-09-28 by rumpeltux

rumpeltux, 2020-09-28:
Just re-checked. This seems to be the case only for zone cleans. Not for room cleans.

Kaan88, 2020-09-28:
A bit unrelated question, how do I get a complete log? Isn't trace the highest level?

rumpeltux, 2020-09-29:
> A bit unrelated question, how do I get a complete log? Isn't trace the highest level?

Yes, but maybe messages are being dropped if logging via udp. Never had this problem myself though.