Hypfer/Valetudo #611: Wrong area calculation in Viomi v8

PaskoMoto, 2020-10-04:
I have realized that the 'Area' calculation in my Viomi V8 is more accurate if AreaCleanupStatsAttribute.value is divided by 1.000 instead of 10.000. i.e:
Changing

`                ("00" + (AreaCleanupStatsAttribute.value / 10000).toFixed(2)).slice(-6) + " m²";`
by

`                ("00" + (AreaCleanupStatsAttribute.value / 1000).toFixed(2)).slice(-6) + " m²";`

in line 305 in the file client/home.js.

I made the change locally and the measurement is accurate to the dimensions of mi apartment, but I am not confident to make a PR since the changes coud probably affect other device models.

Maybe there's another model-specific point in the code where I cuold make the change and test it before issuing a PR?

Best Regards.
PaskoMoto.

schannall, 2020-10-04:
FWIW: I can confirm; the area seems to be 1/10 of what it should be for me as well.

Hypfer, 2020-10-07:
While this bug is indeed valid, please don't delete the provided issue template but rather fill out all requested fields.

It makes things much easier.

fhorinek, 2020-11-21:
The problem is not in client/home.js, but in lib/devices/Viomi.js, during the conversion from m² to cm².
Please look here https://github.com/Hypfer/Valetudo/pull/628#discussion_r528195988

closed 2021-01-09 by Hypfer