Hypfer/Valetudo #612: handle door becoming a barrier

anthonyk91, 2020-10-04:
When I do a room clean with the door closed using a roborock S50 it then updates the map so the door is a permanent obstacle, and afterwards if you ask it to go to a different room it complains that it can't find a route to get there.

Is there a way to mark doors as passable regions, sort of the opposite of a forbidden zone? Or is it possible to prevent a temporary obstacle being marked like a wall?

Hypfer, 2020-10-04:
Valetudo doesn't touch any of the vacuum logic.

Also, this issue tracker is _not_ for support requests like these.

closed 2020-10-04 by Hypfer

anthonyk91, 2020-10-04:
This was not a "support request" it was a feature request for valetudo. If you know that it is not possible to include with the valetudo interface and it can only be done by modifying the firmware it would be better to explain that

Hypfer, 2020-10-04:
The issue tracker is also not for feature requests.

There's a reason why none of the two issue templates matched your issue.

Hypfer, 2020-10-04:
The problem is, that the exact same things are asked/discussed over and over again which very tedious to deal with.
While I do of course understand that newcomers do have these questions, I really can't answer them every week.

Therefore, to direct this elsewhere, this issue tracker has been limited to bugs only.
If I wouldn't instantly close all these duplicate/support issues, the issue tracker would quickly become a complete mess of redundant issues.