Hypfer/Valetudo #619: No menu to enable or select mopping

codingstube, 2020-10-11:
Hi,

I am not really sure. Currently I've build the actually version of valetudo, but I am missing in the main menu the options for mopping. 

I thought I have seen this in previous versions?

Thanks

Hypfer, 2020-10-11:
It's the "off" setting.

Also, please don't use this issue tracker for support requests.

closed 2020-10-11 by Hypfer

codingstube, 2020-10-11:
Sorry I dont get it. 
I searched the settings but I dont find any option to turn it off or on?!

rumpeltux, 2020-10-11:
Mopping is automatically enabled when a mop and water tank is present and disabled otherwise.
If water tank but not dustbin is installed, Valetudo will disable vacuuming and use the mop-only mode.

Note the functionality is currently broken, but it’s fixed in https://github.com/Hypfer/Valetudo/pull/608