Hypfer/Valetudo #626: Web interface wont load if a previously configured MQTT server becomes unavailable

kquinsland, 2020-11-14:
## Describe the bug

As the title indicates; I can 'break' the web UI by taking my MQTT server down.

## To Reproduce

0. Set up valetudo
1. configure MQTT credentials via web UI
2. sever the connection to the MQTT server. In my case, I had some stability issues with the kubernetes cluster that the MQTT server is hosted on, but how you sever the Vac <=> MQTT connection does not appear to matter
3. wait a while. in testing, it's less than 30min, but i have not explicitly timed it.
4. try to load the valetudo web UI. You'll observe your browser's default 'connection refused' message.

Notes: for several min after killing the MQTT connection, the web UI is still accessible. I can go to the settings, view the map... etc.
My 'clue' that valetudo is about to 'die' is when i'll see the 'network error while fetching the resource' pop up on the bottom of the valetudo web interface. Just after that, i'll refresh the page and get the 'unable to connect' message in my browser.

The web page will remain unavailable until i restore connection to the MQTT server. Then, within a few seconds, I can refresh the valetudo page and the 'could not connect' error will be replaced with the web UI.

### Vacuum Model

model: roborock.vacuum.s6
fw version: 1998
valetudo version: 0.6.1

### User-Agent

I can reproduce this with the latest chrome and FF on OSX (Catalina) and Android (version 11)

## Expected behavior

I expect the web UI to still be available if the MQTT server is not available. 
Ideally the web UI would also indicate that the MQTT server could not be reached.

I'm not sure what other information I can provide. If you give me some log locations (or the name of a service to feed into`journalctl`, i'm more than happy to oblige.)

zombielinux, 2020-12-25:
+1 I have this same bug. 

Instead of Kubernetes, I'm running DockerSwarm and the Eclipse MQTT container. 

I have similar "dropouts" with other mqtt clients as well. My MQTT broker is fronted by an HAProxy loadbalancer. 

This is on an S5-Max.

Hypfer, 2021-01-11:
Since pretty much the whole codebase changed recently, please re-check if this is still an issue in [the latest beta](https://github.com/Hypfer/Valetudo/releases/tag/2021.01.0b0) and if yes open a new issue.

Thank you!

closed 2021-01-11 by Hypfer

kquinsland, 2021-01-11:
```The MQTT interface was also rewritten to support different subsets of capabilities.
Instead of having a single topic, which contains all the information available, data is now split up onto different topics
based on capabilities.```

<3. Thanks for that!

I'll aim to update in the next few days and see if i can reproduce