Hypfer/Valetudo #638: Valetudo master (923f941e76e2e41be7a714e47f3a08428d09cfd4) doesn't start

cryptomilk, 2021-01-01:
## Describe the bug

Here is the trace from the logfile:

```
[2021-01-01T08:53:22.807Z] [INFO] Loading configuration file: /mnt/data/valetudo/valetudo_config.json
[2021-01-01T08:53:22.827Z] [INFO] Autodetected RoborockS5ValetudoRobot
[2021-01-01T08:53:22.975Z] [INFO] Starting Valetudo 2020.12.0
[2021-01-01T08:53:22.976Z] [INFO] Robot: Beijing Roborock Technology Co., Ltd. S5 (RoborockS5ValetudoRobot)
[2021-01-01T08:53:22.976Z] [INFO] JS Runtime Version v14.4.0
[2021-01-01T08:53:22.978Z] [INFO] Max Heap Size: 33.5 MiB
[2021-01-01T08:53:22.978Z] [INFO] Node Flags: --max-old-space-size=32
[2021-01-01T08:53:22.980Z] [INFO] DeviceId 117948289
[2021-01-01T08:53:22.980Z] [INFO] IP 127.0.0.1
[2021-01-01T08:53:22.981Z] [INFO] CloudSecret rAuI7KBd8otaO5Ua
[2021-01-01T08:53:22.982Z] [INFO] LocalSecret rk4KK6AEGJykfWv0
/snapshot/valetudo/lib/webserver/WebServer.js:43
        if (this.webserverConfig.basicAuth.enabled === true) {
                                           ^

TypeError: Cannot read property 'enabled' of undefined
    at new WebServer (/snapshot/valetudo/lib/webserver/WebServer.js:43:44)
    at new Valetudo (/snapshot/valetudo/lib/Valetudo.js:41:26)
    at Object.<anonymous> (/snapshot/valetudo/index.js:5:16)
    at Module._compile (internal/modules/cjs/loader.js:1198:30)
    at Module._compile (pkg/prelude/bootstrap.js:1327:32)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1218:10)
    at Module.load (internal/modules/cjs/loader.js:1047:32)
    at Function.Module._load (internal/modules/cjs/loader.js:935:14)
    at Function.Module.runMain (pkg/prelude/bootstrap.js:1375:12)
    at internal/main/run_main_module.js:17:47
```

It should first check if the config entry exists ...

Hypfer, 2021-01-01:
Huh

This should never happen, because the Configuration is always initialized and missing values are filled with the [default_config.json](https://github.com/Hypfer/Valetudo/blob/master/lib/res/default_config.json) as can be seen [here](https://github.com/Hypfer/Valetudo/blob/923f941e76e2e41be7a714e47f3a08428d09cfd4/lib/Configuration.js#L24-L27)

So.. how?!
There are many places in the code which don't check for the existence of config keys due to that logic

Hypfer, 2021-01-01:
After giving it some thought, I think I know what is happening here.

The default assignment is not nested, meaning that if there is a key named `webserver`, it wont be replaced with the default, even if it's missing some keys.

I assume that you may have copied over your old configuration file?
Since the format changed quite a bit, that's not possible and is most likely why Valetudo crashes here.

There's a note about that in the release notes, however as of now they are still a draft on my desktop.

Sorry for the inconvenience

cryptomilk, 2021-01-01:
Yes, I copied the old config file as it complained there doesn't exist one ...

Hypfer, 2021-01-12:
Is this issue solved? it should start with a new and clean config file

cryptomilk, 2021-01-12:
Yep, thanks

closed 2021-01-12 by cryptomilk