Hypfer/Valetudo #640: Add 'clean' parameter to MQTT configuration (allow persistent connection)

bkk87, 2021-01-03:
When MQTT settings are configured, Valetudo will **subscribe** to several topics during start-up, e.g. valetudo/rockrobo/**command**.

In Valetudo, it is possible to specify a 'global' QoS parameter. The QoS parameter is correctly set when subscribing or publishing to a topic.

However, what is currently not possible is to set the 'clean' parameter of mqtt.js for topic subscription.
See: https://github.com/mqttjs/MQTT.js

The current default in Valetudo is `clean = true` which means that all messages are deleted when the robot re-establishes a new connection at the MQTT broker.

Being able to set `clean = false` is especially useful when setting QoS to 1 or 2 because this would allow for **offline** messages. 
Example:
 - due to a flacky Wifi, the robot loses the network connection temporarily 
 - a user/automation trigger sends the command to start cleaning
 - robot comes back online
With `clean = false` the robot would see the previous command and start cleaning.
With `clean = true` (current default) , the command message would be deleted at the broker when the robot comes back online and re-establishes the connection.

Hypfer, 2021-01-09:
This is not abug

Hypfer, 2021-01-12:
`With clean = false the robot would see the previous command and start cleaning.`

Would it ever stop executing that previous command or just repeat that every time a connection to the broker is established?

bkk87, 2021-01-12:
If I read the specs correctly then the robot would just start cleaning once and the message is removed/"processed".

The "clean" parameter is different to the "retain" parameter in mqtt. "retain" will always hold the last message and not delete it.

Also, when using the clean parameter, one can configure in e.g. mosquitto that the session will at some point be removed in case the robot does not connect back to the broker (persistent_client_expiration).

commit a3905106 (2021-04-10): feat!(mqtt): Homie-compatible implementation (#787)

* mqtt: Rename old MQTT package to start Homie refactor

* mqtt: Implement basic structure for new MQTT handlers

* mqtt: Refactor handle(r)s into common base class

* mqtt: Refactor config to make space for Homie

* mqtt: Move more stuff into base MqttHandle class

* mqtt: Move handles into subpackage

* mqtt: Add initial MQTT subscription-only support

async-mqtt is a new dependency, which helps me not want to stab my eyes every time I call MQTT from an async context.

* mqtt: Handle MQTT state and disconnection

* mqtt: Handle publication [... cut]
closed 2021-04-10 by Hypfer