Hypfer/Valetudo #641: No infomation about battery - area - time.

panoslouk, 2021-01-07:
## Describe the bug

I successfully installed valetudo but i can't find out why i can't retrieve information about battery status, area and time. I believe this issue have to  do with miio (maybe i have done wrong about firewall rules or something wrong with config file).
Any advice what can i do? 

You can find some log files and screenshots bellow. 

### Screenshots

[image]

```
<30>Jan  8 05:51:36 valetudo[712]: [1970-01-07T21:51:36.445Z] [DEBUG] local request 2 get_consumable timed out
<30>Jan  8 05:51:36 valetudo[712]: [1970-01-07T21:51:36.457Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":1003}
<30>Jan  8 05:51:36 valetudo[712]: [1970-01-07T21:51:36.961Z] [DEBUG] local request 1003 get_consumable timed out
<30>Jan  8 05:51:36 valetudo[712]: [1970-01-07T21:51:36.965Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":2004}
<30>Jan  8 05:51:37 valetudo[712]: [1970-01-07T21:51:37.467Z] [DEBUG] local request 2004 get_consumable timed out
<30>Jan  8 05:51:37 valetudo[712]: [1970-01-07T21:51:37.476Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":3005}
<30>Jan  8 05:51:37 valetudo[712]: [1970-01-07T21:51:37.981Z] [DEBUG] local request 3005 get_consumable timed out
<30>Jan  8 05:51:37 valetudo[712]: [1970-01-07T21:51:37.991Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":4006}
<30>Jan  8 05:51:38 valetudo[712]: [1970-01-07T21:51:38.497Z] [DEBUG] local request 4006 get_consumable timed out
<30>Jan  8 05:51:38 valetudo[712]: [1970-01-07T21:51:38.500Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":5007}
<30>Jan  8 05:51:39 valetudo[712]: [1970-01-07T21:51:39.002Z] [DEBUG] local request 5007 get_consumable timed out
<30>Jan  8 05:51:39 valetudo[712]: [1970-01-07T21:51:39.004Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":6008}
<30>Jan  8 05:51:39 valetudo[712]: [1970-01-07T21:51:39.507Z] [DEBUG] local request 6008 get_consumable timed out
<30>Jan  8 05:51:39 valetudo[712]: [1970-01-07T21:51:39.514Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":7009}
<30>Jan  8 05:51:40 valetudo[712]: [1970-01-07T21:51:40.020Z] [DEBUG] local request 7009 get_consumable timed out
<30>Jan  8 05:51:40 valetudo[712]: [1970-01-07T21:51:40.023Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":8010}
<30>Jan  8 05:51:40 valetudo[712]: [1970-01-07T21:51:40.525Z] [DEBUG] local request 8010 get_consumable timed out
<30>Jan  8 05:51:40 valetudo[712]: [1970-01-07T21:51:40.533Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":9011}
<30>Jan  8 05:51:41 valetudo[712]: [1970-01-07T21:51:41.038Z] [DEBUG] local request 9011 get_consumable timed out
<27>Jan  8 05:51:41 valetudo[712]: [1970-01-07T21:51:41.041Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 10, method: 'get_consumable', args: [] }
<30>Jan  8 05:51:41 valetudo[712]: [1970-01-07T21:51:41.053Z] [DEBUG] >>> local: HandshakePacket()
<30>Jan  8 05:51:41 valetudo[712]: [1970-01-07T21:51:41.057Z] [DEBUG] <<< local* {"stamp":23}
<30>Jan  8 05:51:41 valetudo[712]: [1970-01-07T21:51:41.057Z] [DEBUG] >>> local* {"stamp":23}
<30>Jan  8 05:51:41 valetudo[712]: [1970-01-07T21:51:41.058Z] [DEBUG] <<= local: handshake complete
<30>Jan  8 05:51:41 valetudo[712]: [1970-01-07T21:51:41.060Z] [DEBUG] <<< local* {"stamp":23}
<30>Jan  8 05:51:41 valetudo[712]: [1970-01-07T21:51:41.061Z] [DEBUG] local: Discarding pong
<30>Jan  8 05:51:41 valetudo[712]: [1970-01-07T21:51:41.356Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":10012}
<30>Jan  8 05:51:41 valetudo[712]: [1970-01-07T21:51:41.858Z] [DEBUG] local request 10012 get_consumable timed out
<30>Jan  8 05:51:41 valetudo[712]: [1970-01-07T21:51:41.861Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":11013}
<30>Jan  8 05:51:42 valetudo[712]: [1970-01-07T21:51:42.363Z] [DEBUG] local request 11013 get_consumable timed out
<30>Jan  8 05:51:42 valetudo[712]: [1970-01-07T21:51:42.365Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":12014}
<30>Jan  8 05:51:42 valetudo[712]: [1970-01-07T21:51:42.867Z] [DEBUG] local request 12014 get_consumable timed out
<30>Jan  8 05:51:42 valetudo[712]: [1970-01-07T21:51:42.869Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":13015}
<30>Jan  8 05:51:43 valetudo[712]: [1970-01-07T21:51:43.371Z] [DEBUG] local request 13015 get_consumable timed out
<30>Jan  8 05:51:43 valetudo[712]: [1970-01-07T21:51:43.378Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":14016}
<30>Jan  8 05:51:43 valetudo[712]: [1970-01-07T21:51:43.882Z] [DEBUG] local request 14016 get_consumable timed out
<30>Jan  8 05:51:43 valetudo[712]: [1970-01-07T21:51:43.885Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":15017}
<30>Jan  8 05:51:44 valetudo[712]: [1970-01-07T21:51:44.385Z] [DEBUG] local request 15017 get_consumable timed out
<30>Jan  8 05:51:44 valetudo[712]: [1970-01-07T21:51:44.387Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":16018}
<30>Jan  8 05:51:44 valetudo[712]: [1970-01-07T21:51:44.889Z] [DEBUG] local request 16018 get_consumable timed out
<30>Jan  8 05:51:44 valetudo[712]: [1970-01-07T21:51:44.897Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":17019}
<30>Jan  8 05:51:45 valetudo[712]: [1970-01-07T21:51:45.401Z] [DEBUG] local request 17019 get_consumable timed out
<30>Jan  8 05:51:45 valetudo[712]: [1970-01-07T21:51:45.404Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":18020}
<30>Jan  8 05:51:45 valetudo[712]: [1970-01-07T21:51:45.905Z] [DEBUG] local request 18020 get_consumable timed out
<30>Jan  8 05:51:45 valetudo[712]: [1970-01-07T21:51:45.908Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":19021}
<30>Jan  8 05:51:46 valetudo[712]: [1970-01-07T21:51:46.410Z] [DEBUG] local request 19021 get_consumable timed out
<27>Jan  8 05:51:46 valetudo[712]: [1970-01-07T21:51:46.412Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 20, method: 'get_consumable', args: [] }
<30>Jan  8 05:51:46 valetudo[712]: [1970-01-07T21:51:46.414Z] [DEBUG] >>> local: HandshakePacket()
<30>Jan  8 05:51:46 valetudo[712]: [1970-01-07T21:51:46.416Z] [DEBUG] <<< local* {"stamp":28}
<30>Jan  8 05:51:46 valetudo[712]: [1970-01-07T21:51:46.417Z] [DEBUG] >>> local* {"stamp":28}
<30>Jan  8 05:51:46 valetudo[712]: [1970-01-07T21:51:46.418Z] [DEBUG] <<= local: handshake complete
<30>Jan  8 05:51:46 valetudo[712]: [1970-01-07T21:51:46.419Z] [DEBUG] <<< local* {"stamp":28}
<30>Jan  8 05:51:46 valetudo[712]: [1970-01-07T21:51:46.420Z] [DEBUG] local: Discarding pong
<30>Jan  8 05:51:46 valetudo[712]: [1970-01-07T21:51:46.718Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":20022}
<30>Jan  8 05:51:47 valetudo[712]: [1970-01-07T21:51:47.219Z] [DEBUG] local request 20022 get_consumable timed out
<30>Jan  8 05:51:47 valetudo[712]: [1970-01-07T21:51:47.222Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":21023}
<30>Jan  8 05:51:47 valetudo[712]: [1970-01-07T21:51:47.724Z] [DEBUG] local request 21023 get_consumable timed out
<30>Jan  8 05:51:47 valetudo[712]: [1970-01-07T21:51:47.726Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":22024}
[... 94 lines cut ...]
ff02::1 ip6-allnodes
ff02::2 ip6-allrouters
110.43.0.83 ot.io.mi.com ott.io.mi.com
110.43.0.83 de.ot.io.mi.com de.ott.io.mi.com
110.43.0.83 ea.ot.io.mi.com ea.ott.io.mi.com
110.43.0.83 in.ot.io.mi.com in.ott.io.mi.com
110.43.0.83 pv.ot.io.mi.com pv.ott.io.mi.com
110.43.0.83 ru.ot.io.mi.com ru.ott.io.mi.com
110.43.0.83 sg.ot.io.mi.com sg.ott.io.mi.com
110.43.0.83 st.ot.io.mi.com st.ott.io.mi.com
110.43.0.83 tw.ot.io.mi.com tw.ott.io.mi.com
110.43.0.83 us.ot.io.mi.com us.ott.io.mi.com

```

### Vacuum Model

Viomi.v8 - STYTJ02YM
### Valetudo Version

2021.01.0b0
### Firmware Version

3.5.3_0017
### User-Agent

rumpeltux, 2021-01-08:
Please also post your valetudo.conf.

panoslouk, 2021-01-08:
Thanks for the replay! here is my config file.
```
{
  "embedded": true,
  "robot": {
    "implementation": "auto",
    "implementationSpecificConfig": {
      "ip": "127.0.0.1"
    }
  },
  "webserver": {
    "port": 80,
    "basicAuth": {
      "enabled": false,
      "username": "valetudo",
      "password": "valetudo"
    }
  },
  "zonePresets": {},
  "goToLocationPresets": {},
  "mqtt": {
    "enabled": false,
    "server": "",
    "port": "1883",
    "clientId": "",
    "username": "",
    "password": "",
    "usetls": false,
    "ca": "",
    "clientCert": "",
    "clientKey": "",
    "qos": 0,
    "identifier": "robot",
    "topicPrefix": "valetudo",
    "autoconfPrefix": "homeassistant",
    "provideMapData": true,
    "base64EncodeMapData": true
  },
  "allowSSHKeyUpload": true,
  "logLevel": "trace",
  "debug": {
    "memoryStatInterval": false,
    "enableRawCommandCapability": false
  }
}
```

rumpeltux, 2021-01-08:
Looks like a port mismatch:
config says 80, but iptables says 8080
see also https://valetudo.cloud/pages/installation/viomi.html

panoslouk, 2021-01-10:
Actually i changed the port to 8080 and i got the same issue. After that i changed the port in iptables to 80 from 8080 and the port in config file from 8080 to 80 and i got the same issue again.

panoslouk, 2021-01-11:
I found out that the data response from miio (??) cannot be mapped to the STATE_PROPERTIES object. After a few changes within ViomiValetudoRobot.js files i made it to work.

Since i don't have big knowledge of nodejs, im not sure if what i did is right. I thought that is good to mention what i did to solve the problem

Hypfer, 2021-01-12:
What changes exactly?

w1an, 2021-01-12:
> ViomiValetudoRobot.js

I have exactly the same issue with a viomi v7. My debug log looks very much like above. I'm very much interested in the fix. If I can provide debug logs, please let me know what (and how).
```
[2020-09-26T18:42:27.208Z] [INFO] Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053
[2020-09-26T18:42:27.211Z] [INFO] Webserver running on port 80
[2020-09-26T18:42:27.225Z] [INFO] Map Upload Server running on port 8087
[2020-09-26T18:42:27.230Z] [DEBUG] >>> local: HandshakePacket()
[2020-09-26T18:42:27.241Z] [DEBUG] <<< local* {"stamp":1601145748}
[2020-09-26T18:42:27.242Z] [DEBUG] >>> local* {"stamp":1601145748}
[2020-09-26T18:42:27.244Z] [DEBUG] <<= local: handshake complete
[2020-09-26T18:42:27.247Z] [DEBUG] <<< local* {"stamp":1601145748}
[2020-09-26T18:42:27.248Z] [DEBUG] local: Discarding pong
[2020-09-26T18:42:27.250Z] [DEBUG] <<< local* {"stamp":1601145748}
[2020-09-26T18:42:27.251Z] [DEBUG] local: Discarding pong
[2020-09-26T18:42:27.682Z] [INFO] Connected successfully to mqtt server
[2020-09-26T18:42:27.750Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":2}
[2020-09-26T18:42:28.241Z] [DEBUG] local request 1 get_prop timed out
[2020-09-26T18:42:28.251Z] [DEBUG] local request 2 get_consumable timed out
[2020-09-26T18:42:28.255Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":2004}
[2020-09-26T18:42:28.747Z] [DEBUG] local request 1003 get_prop timed out
[2020-09-26T18:42:28.755Z] [DEBUG] local request 2004 get_consumable timed out
[2020-09-26T18:42:28.759Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":4006}
[2020-09-26T18:42:28.782Z] [DEBUG] <<< local: {"result":[5,0,2105,100,3,1,0,0,1,12,1,0,1,0],"id":1}
[2020-09-26T18:42:28.783Z] [INFO] << local: ignoring response for non-pending request {"result":[5,0,2105,100,3,1,0,0,1,12,1,0,1,0],"id":1}
[2020-09-26T18:42:29.250Z] [DEBUG] local request 3005 get_prop timed out
[2020-09-26T18:42:29.259Z] [DEBUG] local request 4006 get_consumable timed out
[2020-09-26T18:42:29.262Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":6008}
[2020-09-26T18:42:29.751Z] [DEBUG] local request 5007 get_prop timed out
[2020-09-26T18:42:29.763Z] [DEBUG] local request 6008 get_consumable timed out
[2020-09-26T18:42:29.769Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":8010}
[2020-09-26T18:42:30.254Z] [DEBUG] local request 7009 get_prop timed out
[2020-09-26T18:42:30.270Z] [DEBUG] local request 8010 get_consumable timed out
[2020-09-26T18:42:30.273Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":10012}
[2020-09-26T18:42:30.287Z] [DEBUG] <<< local: {"result":[5,0,2105,100,3,1,0,0,1,12,1,0,1,0],"id":5007}
[2020-09-26T18:42:30.288Z] [INFO] << local: ignoring response for non-pending request {"result":[5,0,2105,100,3,1,0,0,1,12,1,0,1,0],"id":5007}
[2020-09-26T18:42:30.757Z] [DEBUG] local request 9011 get_prop timed out
[2020-09-26T18:42:30.774Z] [DEBUG] local request 10012 get_consumable timed out
[2020-09-26T18:42:30.781Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":12014}
[2020-09-26T18:42:31.265Z] [DEBUG] local request 11013 get_prop timed out
[2020-09-26T18:42:31.280Z] [DEBUG] local request 12014 get_consumable timed out
[2020-09-26T18:42:31.283Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":14016}
[2020-09-26T18:42:31.769Z] [DEBUG] local request 13015 get_prop timed out
[2020-09-26T18:42:31.784Z] [DEBUG] local request 14016 get_consumable timed out
[2020-09-26T18:42:31.787Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":16018}
[2020-09-26T18:42:31.807Z] [DEBUG] <<< local: {"result":[5,0,2105,100,3,1,0,0,1,12,1,0,1,0],"id":11013}
[2020-09-26T18:42:31.808Z] [INFO] << local: ignoring response for non-pending request {"result":[5,0,2105,100,3,1,0,0,1,12,1,0,1,0],"id":11013}
[2020-09-26T18:42:32.271Z] [DEBUG] local request 15017 get_prop timed out
[2020-09-26T18:42:32.291Z] [DEBUG] local request 16018 get_consumable timed out
[2020-09-26T18:42:32.301Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":18020}
[2020-09-26T18:42:32.784Z] [DEBUG] local request 17019 get_prop timed out
[2020-09-26T18:42:32.788Z] [WARN] Token is okay, however we're unable to reach the vacuum {
  retries: 10,
  method: 'get_prop',
  args: [
    'run_state',     'mode',
    'err_state',     'battary_life',
    'box_type',      'mop_type',
    's_time',        's_area',
    'suction_grade', 'water_grade',
    'remember_map',  'has_map',
    'is_mop',        'has_newmap'
  ]
}
[2020-09-26T18:42:32.819Z] [DEBUG] >>> local: HandshakePacket()
[2020-09-26T18:42:32.821Z] [DEBUG] local request 18020 get_consumable timed out
[2020-09-26T18:42:32.823Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 10, method: 'get_consumable', args: [] }
[2020-09-26T18:42:32.826Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":19021}
[2020-09-26T18:42:32.828Z] [DEBUG] <<< local* {"stamp":1601145753}
[2020-09-26T18:42:32.829Z] [DEBUG] >>> local* {"stamp":1601145753}
[... 21 lines cut ...]
[2020-09-26T18:42:36.363Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":30032}
[2020-09-26T18:42:36.865Z] [DEBUG] local request 30032 get_consumable timed out
[2020-09-26T18:42:36.868Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":31033}
[2020-09-26T18:42:37.369Z] [DEBUG] local request 31033 get_consumable timed out
[... cut ...]
[2020-09-26T18:42:39.208Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":35037}
[2020-09-26T18:42:39.709Z] [DEBUG] local request 35037 get_consumable timed out
[2020-09-26T18:42:39.712Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":36038}
[2020-09-26T18:42:39.901Z] [DEBUG] <<< cloud* {"stamp":0}
[2020-09-26T18:42:39.902Z] [DEBUG] ^-- initial timesync packet
[2020-09-26T18:42:39.903Z] [DEBUG] >>> Responding to timesync request
[2020-09-26T18:42:39.905Z] [INFO] Cloud connected
```

panoslouk, 2021-01-12:
Hi there

i have changed the **pollState()** function to

```
    async pollState() {
        const response = await this.sendCommand("get_prop", STATE_PROPERTIES);
        if (response) {
            this.parseAndUpdateState(response);
        }
        return this.state;
    }
```

and i added to the begging of the function **parseAndUpdateState** and after the **let newStateAttr;** the following

```
       data["run_state"] = data[0];
       data["mode"] = data[1];
       data["err_state"] = data[2];
       data["battary_life"] = data[3];
       data["box_type"] = data[4];
       data["mop_type"] = data[5];
       data["s_time"] = data[6];
       data["s_area"] = data[7];
       data["suction_grade"] = data[8];
       data["water_grade"] = data[9];
       data["remember_map"] = data[10];
       data["has_map"] = data[11];
       data["is_mop"] = data[12];
       data["has_newmap"] = data[13];
```
After those changes UI started to receive the stats from the vacuum.
I'm saying again, i m not sure about those changes. Seems that this isn't the way that the code doing the mapping of the data but its a quick fix until i have a better view of the code.

v0d0r, 2021-01-16:
Same problem for me and also seems no maps generate.
I am also running:  Vacuum Model Viomi.v8 - STYTJ02YM Valetudo Version 2021.01.0b0

I have not tried workaround changes by panoslouk from above.

panoslouk, 2021-01-16:
I have the same vacuum robot, you must change the port in /etc/rc.d/S51valetudo from 80 to 8080. After that change the webserver port in config file to 8080 to receive map data.

v0d0r, 2021-01-17:
> I have the same vacuum robot, you must change the port in /etc/rc.d/S51valetudo from 80 to 8080. After that change the webserver port in config file to 8080 to receive map data.

Thanks panoslouk I will give this a try and do another full clean to see if the map appears.
On another note: Where exactly do I find the ViomiValetudoRobot.js to try out your changes?  Or do I have to edit locally and rebuild?

panoslouk, 2021-01-17:
Download the pre-release from here https://github.com/Hypfer/Valetudo/archive/2021.01.0b0.zip
Unzip that file and do run npm install inside valetudo folder just to Install dependencies, then change the /lib/robots/viomi/ViomiValetudoRobot.js file and run npm run build to generate a new valetudo binary file and transfer that file to your vacuum. 

More details and instructions you will find here https://valetudo.cloud/pages/development/building-and-modifying-valetudo.html

v0d0r, 2021-01-18:
> Download the pre-release from here https://github.com/Hypfer/Valetudo/archive/2021.01.0b0.zip
> Unzip that file and do run npm install inside valetudo folder just to Install dependencies, then change [...]

Thanks for this. I can confirm everything looks good now.  I just need to get the map after a full clean.

[image]

Hypfer, 2021-01-18:
Please share a trace level log with this modification

v0d0r, 2021-01-18:
> Please share a trace level log with this modification

See my trace log output below.  Let me know if you want any other info.

<30>Jan  2 14:21:36 logread[423]: Logread connected to 192.168.128.1:8054
<30>Jan  2 14:21:44 valetudo[1025]: [1970-01-02T06:21:44.579Z] [INFO] Loading configuration file: /mnt/UDISK/valetudo_config.json
<30>Jan  2 14:21:44 valetudo[1025]: [1970-01-02T06:21:44.605Z] [TRACE] Trying to open device.conf at /mnt/default/device.conf
<30>Jan  2 14:21:44 valetudo[1025]: [1970-01-02T06:21:44.609Z] [TRACE] cannot read /mnt/default/device.conf Error: ENOENT: no such file or directory, open '/mnt/default/device.conf'
<30>Jan  2 14:21:44 valetudo[1025]:     at Object.openSync (fs.js:461:3)
<30>Jan  2 14:21:44 valetudo[1025]:     at Object.fs.openSync (pkg/prelude/bootstrap.js:487:32)
<30>Jan  2 14:21:44 valetudo[1025]:     at Object.readFileSync (fs.js:364:35)
<30>Jan  2 14:21:44 valetudo[1025]:     at Object.fs.readFileSync (pkg/prelude/bootstrap.js:683:36)
<30>Jan  2 14:21:44 valetudo[1025]:     at Function.READ_DEVICE_CONF (/snapshot/Valetudo/lib/robots/MiioValetudoRobot.js:326:29)
<30>Jan  2 14:21:44 valetudo[1025]:     at Function.IMPLEMENTATION_AUTO_DETECTION_HANDLER (/snapshot/Valetudo/lib/robots/roborock/RoborockV1ValetudoRobot.js:33:46)
<30>Jan  2 14:21:44 valetudo[1025]:     at /snapshot/Valetudo/lib/core/ValetudoRobotFactory.js:39:27
<30>Jan  2 14:21:44 valetudo[1025]:     at Array.some (<anonymous>)
<30>Jan  2 14:21:44 valetudo[1025]:     at Function.autodetectRobotImplementation (/snapshot/Valetudo/lib/core/ValetudoRobotFactory.js:38:29)
<30>Jan  2 14:21:44 valetudo[1025]:     at Function.getRobotImplementation (/snapshot/Valetudo/lib/core/ValetudoRobotFactory.js:16:45) {
<30>Jan  2 14:21:44 valetudo[1025]:   errno: -2,
<30>Jan  2 14:21:44 valetudo[1025]:   syscall: 'open',
<30>Jan  2 14:21:44 valetudo[1025]:   code: 'ENOENT',
<30>Jan  2 14:21:44 valetudo[1025]:   path: '/mnt/default/device.conf'
<30>Jan  2 14:21:44 valetudo[1025]: }
<30>Jan  2 14:21:44 valetudo[1025]: [1970-01-02T06:21:44.629Z] [TRACE] Failed to read device.conf
<30>Jan  2 14:21:44 valetudo[1025]: [1970-01-02T06:21:44.629Z] [TRACE] Trying to open device.conf at /mnt/default/device.conf
<30>Jan  2 14:21:44 valetudo[1025]: [1970-01-02T06:21:44.631Z] [TRACE] cannot read /mnt/default/device.conf Error: ENOENT: no such file or directory, open '/mnt/default/device.conf'
<30>Jan  2 14:21:44 valetudo[1025]:     at Object.openSync (fs.js:461:3)
<30>Jan  2 14:21:44 valetudo[1025]:     at Object.fs.openSync (pkg/prelude/bootstrap.js:487:32)
<30>Jan  2 14:21:44 valetudo[1025]:     at Object.readFileSync (fs.js:364:35)
<30>Jan  2 14:21:44 valetudo[1025]:     at Object.fs.readFileSync (pkg/prelude/bootstrap.js:683:36)
<30>Jan  2 14:21:44 valetudo[1025]:     at Function.READ_DEVICE_CONF (/snapshot/Valetudo/lib/robots/MiioValetudoRobot.js:326:29)
<30>Jan  2 14:21:44 valetudo[1025]:     at Function.IMPLEMENTATION_AUTO_DETECTION_HANDLER (/snapshot/Valetudo/lib/robots/roborock/RoborockS5ValetudoRobot.js:36:46)
<30>Jan  2 14:21:44 valetudo[1025]:     at /snapshot/Valetudo/lib/core/ValetudoRobotFactory.js:39:27
<30>Jan  2 14:21:44 valetudo[1025]:     at Array.some (<anonymous>)
<30>Jan  2 14:21:44 valetudo[1025]:     at Function.autodetectRobotImplementation (/snapshot/Valetudo/lib/core/ValetudoRobotFactory.js:38:29)
<30>Jan  2 14:21:44 valetudo[1025]:     at Function.getRobotImplementation (/snapshot/Valetudo/lib/core/ValetudoRobotFactory.js:16:45) {
<30>Jan  2 14:21:44 valetudo[1025]:   errno: -2,
<30>Jan  2 14:21:44 valetudo[1025]:   syscall: 'open',
<30>Jan  2 14:21:44 valetudo[1025]:   code: 'ENOENT',
<30>Jan  2 14:21:44 valetudo[1025]:   path: '/mnt/default/device.conf'
<30>Jan  2 14:21:44 valetudo[1025]: }
<30>Jan  2 14:21:44 valetudo[1025]: [1970-01-02T06:21:44.634Z] [TRACE] Failed to read device.conf
<30>Jan  2 14:21:44 valetudo[1025]: [1970-01-02T06:21:44.636Z] [TRACE] Trying to open device.conf at /mnt/default/device.conf
<30>Jan  2 14:21:44 valetudo[1025]: [1970-01-02T06:21:44.637Z] [TRACE] cannot read /mnt/default/device.conf Error: ENOENT: no such file or directory, open '/mnt/default/device.conf'
<30>Jan  2 14:21:44 valetudo[1025]:     at Object.openSync (fs.js:461:3)
<30>Jan  2 14:21:44 valetudo[1025]:     at Object.fs.openSync (pkg/prelude/bootstrap.js:487:32)
<30>Jan  2 14:21:44 valetudo[1025]:     at Object.readFileSync (fs.js:364:35)
<30>Jan  2 14:21:44 valetudo[1025]:     at Object.fs.readFileSync (pkg/prelude/bootstrap.js:683:36)
<30>Jan  2 14:21:44 valetudo[1025]:     at Function.READ_DEVICE_CONF (/snapshot/Valetudo/lib/robots/MiioValetudoRobot.js:326:29)
<30>Jan  2 14:21:44 valetudo[1025]:     at Function.IMPLEMENTATION_AUTO_DETECTION_HANDLER (/snapshot/Valetudo/lib/robots/roborock/RoborockS6ValetudoRobot.js:34:46)
<30>Jan  2 14:21:44 valetudo[1025]:     at /snapshot/Valetudo/lib/core/ValetudoRobotFactory.js:39:27
<30>Jan  2 14:21:44 valetudo[1025]:     at Array.some (<anonymous>)
<30>Jan  2 14:21:44 valetudo[1025]:     at Function.autodetectRobotImplementation (/snapshot/Valetudo/lib/core/ValetudoRobotFactory.js:38:29)
<30>Jan  2 14:21:44 valetudo[1025]:     at Function.getRobotImplementation (/snapshot/Valetudo/lib/core/ValetudoRobotFactory.js:16:45) {
<30>Jan  2 14:21:44 valetudo[1025]:   errno: -2,
<30>Jan  2 14:21:44 valetudo[1025]:   syscall: 'open',
<30>Jan  2 14:21:44 valetudo[1025]:   code: 'ENOENT',
<30>Jan  2 14:21:44 valetudo[1025]:   path: '/mnt/default/device.conf'
[... cut ...]
<30>Jan  2 14:21:46 valetudo[1025]: [1970-01-02T06:21:46.982Z] [DEBUG] local: Discarding pong
<30>Jan  2 14:21:46 valetudo[1025]: [1970-01-02T06:21:46.984Z] [DEBUG] <<< local* {"stamp":863}
<30>Jan  2 14:21:46 valetudo[1025]: [1970-01-02T06:21:46.984Z] [DEBUG] local: Discarding pong
<30>Jan  2 14:21:46 valetudo[1025]: [1970-01-02T06:21:46.986Z] [DEBUG] <<< local* {"stamp":863}
<30>Jan  2 14:21:46 valetudo[1025]: [1970-01-02T06:21:46.986Z] [DEBUG] local: Discarding pong
<30>Jan  2 14:21:47 valetudo[1025]: [1970-01-02T06:21:47.213Z] [TRACE] Initial handshake successful

Hypfer, 2021-01-18:
@v0d0r It was missing the part with a successful get_prop command. The full raw request & reply is what I'm looking for

v0d0r, 2021-01-18:
> @v0d0r It was missing the part with a successful get_prop command. The full raw request & reply is what I'm looking for
I copied the log to quick. This showed up a few seconds later. 

<30>Jan  2 14:27:13 valetudo[1025]: [1970-01-02T06:27:13.519Z] [TRACE] Trying to read token file at /etc/miio/device.token
<27>Jan  2 14:27:13 valetudo[1025]: [1970-01-02T06:27:13.521Z] [WARN] Token is okay, however we're unable to reach the vacuum {
<27>Jan  2 14:27:13 valetudo[1025]:   retries: 10,
<27>Jan  2 14:27:13 valetudo[1025]:   method: 'get_prop',
<27>Jan  2 14:27:13 valetudo[1025]:   args: [
<27>Jan  2 14:27:13 valetudo[1025]:     'run_state',     'mode',
<27>Jan  2 14:27:13 valetudo[1025]:     'err_state',     'battary_life',
<27>Jan  2 14:27:13 valetudo[1025]:     'box_type',      'mop_type',
<27>Jan  2 14:27:13 valetudo[1025]:     's_time',        's_area',
<27>Jan  2 14:27:13 valetudo[1025]:     'suction_grade', 'water_grade',
<27>Jan  2 14:27:13 valetudo[1025]:     'remember_map',  'has_map',
<27>Jan  2 14:27:13 valetudo[1025]:     'is_mop',        'has_newmap'
<27>Jan  2 14:27:13 valetudo[1025]:   ]
<27>Jan  2 14:27:13 valetudo[1025]: }
<30>Jan  2 14:27:13 valetudo[1025]: [1970-01-02T06:27:13.528Z] [DEBUG] >>> local: HandshakePacket()
<30>Jan  2 14:27:13 valetudo[1025]: [1970-01-02T06:27:13.830Z] [DEBUG] >>> local: HandshakePacket()
<30>Jan  2 14:27:14 valetudo[1025]: [1970-01-02T06:27:14.132Z] [DEBUG] >>> local: HandshakePacket()
<30>Jan  2 14:27:14 valetudo[1025]: [1970-01-02T06:27:14.435Z] [DEBUG] >>> local: HandshakePacket()
<30>Jan  2 14:27:14 valetudo[1025]: [1970-01-02T06:27:14.654Z] [DEBUG] <<< local* {"stamp":1191}
<30>Jan  2 14:27:14 valetudo[1025]: [1970-01-02T06:27:14.657Z] [DEBUG] >>> local* {"stamp":1191}
<30>Jan  2 14:27:14 valetudo[1025]: [1970-01-02T06:27:14.661Z] [DEBUG] <<= local: handshake complete
<30>Jan  2 14:27:14 valetudo[1025]: [1970-01-02T06:27:14.669Z] [DEBUG] <<< local* {"stamp":1191}
<30>Jan  2 14:27:14 valetudo[1025]: [1970-01-02T06:27:14.671Z] [DEBUG] local: Discarding pong
<30>Jan  2 14:27:14 valetudo[1025]: [1970-01-02T06:27:14.678Z] [DEBUG] <<< local* {"stamp":1191}
<30>Jan  2 14:27:14 valetudo[1025]: [1970-01-02T06:27:14.680Z] [DEBUG] local: Discarding pong
<30>Jan  2 14:27:14 valetudo[1025]: [1970-01-02T06:27:14.683Z] [DEBUG] <<< local* {"stamp":1191}
<30>Jan  2 14:27:14 valetudo[1025]: [1970-01-02T06:27:14.684Z] [DEBUG] local: Discarding pong
<30>Jan  2 14:27:14 valetudo[1025]: [1970-01-02T06:27:14.686Z] [DEBUG] <<< local* {"stamp":1191}
<30>Jan  2 14:27:14 valetudo[1025]: [1970-01-02T06:27:14.686Z] [DEBUG] local: Discarding pong
<30>Jan  2 14:27:14 valetudo[1025]: [1970-01-02T06:27:14.746Z] [TRACE] >>> local: {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"],"id":10001}

v0d0r, 2021-01-18:
For what its worth. My maps still did not work after changing to port 8080.
However I never saw the following log entry ever:
 <30>Jan  2 15:13:34 valetudo[1292]: [1970-01-02T07:13:34.920Z] [INFO] Cloud connected

I then did a tcpdump and just excluded port 22.

I then saw 2 new IP addresses.
15:05:10.991951 IP 110.43.0.85.80 > 192.168.128.21.46157: Flags [.], ack 195, win 948, length 0
15:05:11.420475 IP 110.43.0.83.80 > 192.168.128.21.51736: Flags [.], ack 195, win 948, length 0

I am not sure why these werenot caught by the host file.
I added them to my  /etc/rc.d/S51valetudo with the existing hosts as follows:
for host in 203.0.113.1 203.0.113.5 110.43.0.85 110.43.0.83 ; do

iptables now caters for these IP's

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination
DNAT       tcp  --  0.0.0.0/0            203.0.113.1          tcp dpt:80 to:127.0.0.1:8080
DNAT       udp  --  0.0.0.0/0            203.0.113.1          udp dpt:8053 to:127.0.0.1:8053
DNAT       tcp  --  0.0.0.0/0            203.0.113.5          tcp dpt:80 to:127.0.0.1:8080
DNAT       udp  --  0.0.0.0/0            203.0.113.5          udp dpt:8053 to:127.0.0.1:8053
DNAT       tcp  --  0.0.0.0/0            110.43.0.85          tcp dpt:80 to:127.0.0.1:8080
DNAT       udp  --  0.0.0.0/0            110.43.0.85          udp dpt:8053 to:127.0.0.1:8053
DNAT       tcp  --  0.0.0.0/0            110.43.0.83          tcp dpt:80 to:127.0.0.1:8080
DNAT       udp  --  0.0.0.0/0            110.43.0.83          udp dpt:8053 to:127.0.0.1:8053

And my MAP now shows after the cloud connect is in the log.

[image]

panoslouk, 2021-01-18:
not sure if that trace log will help you out but im posting here just for the record. 

```
[2021-01-18T20:20:24.246Z] [TRACE] fpos:abb8b take 1@trailing <Buffer ff>
[2021-01-18T20:20:24.521Z] [DEBUG] Sending command using CLOUD: Method ->"get_prop" Paramsrun_state,mode,err_state,battary_life,box_type,mop_type,s_time,s_area,suction_grade,water_grade,remember_map,has_map,is_mop,has_newmap Options: [object Object]
[2021-01-18T20:20:24.523Z] [DEBUG] Received command {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"]}
[2021-01-18T20:20:24.526Z] [TRACE] >>> cloud: {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"],"id":11}
[2021-01-18T20:20:24.554Z] [DEBUG] get_prop <<< cloud: {"result":[5,0,2105,100,1,0,26,20.04,3,12,1,1,0,1],"id":11}
Start Mapping data: 5,0,2105,100,1,0,26,20.04,3,12,1,1,0,1
run_state 5 mode 0 err_state 2105 battary_life 100 box_type 1 mop_type 0
s_time 26 s_area 20.04 suction_grade 3 water_grade 12 remember_map 1
has_map 1 is_mop 0 has_newmap 1
[2021-01-18T20:20:24.653Z] [DEBUG] prop.run_state <<< cloud: {"method":"prop.run_state","params":[5]}
---------------------------->{"method":"prop.run_state","params":[5]}
[2021-01-18T20:20:24.713Z] [DEBUG] prop.box_type <<< cloud: {"method":"prop.box_type","params":[1]}
---------------------------->{"method":"prop.box_type","params":[1]}
[2021-01-18T20:20:24.715Z] [DEBUG] prop.mop_type <<< cloud: {"method":"prop.mop_type","params":[0]}
---------------------------->{"method":"prop.mop_type","params":[0]}
[2021-01-18T20:20:24.716Z] [DEBUG] prop.err_state <<< cloud: {"method":"prop.err_state","params":[2105]}
---------------------------->{"method":"prop.err_state","params":[2105]}
[2021-01-18T20:20:24.718Z] [DEBUG] prop.suction_grade <<< cloud: {"method":"prop.suction_grade","params":[3]}
---------------------------->{"method":"prop.suction_grade","params":[3]}
[2021-01-18T20:20:24.719Z] [DEBUG] prop.is_charge <<< cloud: {"method":"prop.is_charge","params":[0]}
---------------------------->{"method":"prop.is_charge","params":[0]}
[2021-01-18T20:20:24.721Z] [DEBUG] prop.is_work <<< cloud: {"method":"prop.is_work","params":[1]}
---------------------------->{"method":"prop.is_work","params":[1]}
[2021-01-18T20:20:25.015Z] [DEBUG] Sending command using CLOUD: Method ->"get_prop" Paramsrun_state,mode,err_state,battary_life,box_type,mop_type,s_time,s_area,suction_grade,water_grade,remember_map,has_map,is_mop,has_newmap Options: [object Object]
[2021-01-18T20:20:25.016Z] [DEBUG] Received command {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"]}
[2021-01-18T20:20:25.017Z] [TRACE] >>> cloud: {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"],"id":12}
[2021-01-18T20:20:25.038Z] [DEBUG] get_prop <<< cloud: {"result":[5,0,2105,100,1,0,26,20.04,3,12,1,1,0,1],"id":12}
Start Mapping data: 5,0,2105,100,1,0,26,20.04,3,12,1,1,0,1
run_state 5 mode 0 err_state 2105 battary_life 100 box_type 1 mop_type 0
s_time 26 s_area 20.04 suction_grade 3 water_grade 12 remember_map 1
has_map 1 is_mop 0 has_newmap 1
[2021-01-18T20:20:25.523Z] [DEBUG] Sending command using CLOUD: Method ->"get_prop" Paramsrun_state,mode,err_state,battary_life,box_type,mop_type,s_time,s_area,suction_grade,water_grade,remember_map,has_map,is_mop,has_newmap Options: [object Object]
[2021-01-18T20:20:25.524Z] [DEBUG] Received command {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"]}
[2021-01-18T20:20:25.526Z] [TRACE] >>> cloud: {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"],"id":13}
[2021-01-18T20:20:25.543Z] [DEBUG] get_prop <<< cloud: {"result":[5,0,2105,100,1,0,26,20.04,3,12,1,1,0,1],"id":13}
Start Mapping data: 5,0,2105,100,1,0,26,20.04,3,12,1,1,0,1
run_state 5 mode 0 err_state 2105 battary_life 100 box_type 1 mop_type 0
s_time 26 s_area 20.04 suction_grade 3 water_grade 12 remember_map 1
has_map 1 is_mop 0 has_newmap 1
[2021-01-18T20:20:26.019Z] [DEBUG] Sending command using CLOUD: Method ->"get_prop" Paramsrun_state,mode,err_state,battary_life,box_type,mop_type,s_time,s_area,suction_grade,water_grade,remember_map,has_map,is_mop,has_newmap Options: [object Object]
[2021-01-18T20:20:26.020Z] [DEBUG] Received command {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"]}
[2021-01-18T20:20:26.021Z] [TRACE] >>> cloud: {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"],"id":14}
[... cut ...]
[2021-01-18T20:20:42.087Z] [TRACE] >>> cloud: {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"],"id":46}
[2021-01-18T20:20:42.098Z] [DEBUG] get_prop <<< cloud: {"result":[5,0,2105,100,1,0,26,20.04,3,12,1,1,0,1],"id":46}
Start Mapping data: 5,0,2105,100,1,0,26,20.04,3,12,1,1,0,1
run_state 5 mode 0 err_state 2105 battary_life 100 box_type 1 mop_type 0
s_time 26 s_area 20.04 suction_grade 3 water_grade 12 remember_map 1
has_map 1 is_mop 0 has_newmap 1

```

Hypfer, 2021-01-18:
In the old code, the response array was mapped to the requested properties before then parsing it:
https://github.com/Hypfer/Valetudo/blob/6db8b76bddfea636303d1169fa7441ac21a5deca/lib/devices/Viomi.js#L416-L423

which would provide a working Valetudo on the robots of the users in this issue since it looks like this:
```
>>> cloud: {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"],"id":46}
<<< cloud: {"result":[5,0,2105,100,1,0,26,20.04,3,12,1,1,0,1],"id":46}
```

In the current codebase there's no such thing happening
https://github.com/Hypfer/Valetudo/blob/cd76abe952cda43fd260eda2b217aff677b98d29/lib/robots/viomi/ViomiValetudoRobot.js#L121-L129

which apparently also works on at least @depau's viomi?

Why is that?
It's not the model since there are both v7 and v8 with the same issue in this thread.
Is it the fw version?

panoslouk, 2021-01-18:
If i understand correctly you mean that maybe the fw version of the vacuum was create that issue?

btw according to the old code i changed the pollstate() function. Now the parseAndUpdateState() function seems to work good.

```
    async pollState() {
        const response = await this.sendCommand("get_prop", STATE_PROPERTIES);
        let statusDict = {};
        STATE_PROPERTIES.forEach((key, index) => statusDict[key] = response[index]);
        if (response) {
            this.parseAndUpdateState(statusDict);
        }

        return this.state;
    }
```

```
[2021-01-18T21:25:24.871Z] [TRACE] >>> cloud: {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"],"id":8}
[2021-01-18T21:25:24.916Z] [DEBUG] get_prop <<< cloud: {"result":[5,0,2105,100,1,0,26,20.04,3,12,1,1,0,1],"id":8}
[2021-01-18T21:25:25.369Z] [DEBUG] Sending command using CLOUD: Method ->"get_prop" Paramsrun_state,mode,err_state,battary_life,box_type,mop_type,s_time,s_area,suction_grade,water_grade,remember_map,has_map,is_mop,has_newmap Options: [object Object]
[2021-01-18T21:25:25.369Z] [DEBUG] Received command {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"]}
[2021-01-18T21:25:25.371Z] [TRACE] >>> cloud: {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"],"id":9}
[2021-01-18T21:25:25.382Z] [DEBUG] get_prop <<< cloud: {"result":[5,0,2105,100,1,0,26,20.04,3,12,1,1,0,1],"id":9}
[2021-01-18T21:25:25.870Z] [DEBUG] Sending command using CLOUD: Method ->"get_prop" Paramsrun_state,mode,err_state,battary_life,box_type,mop_type,s_time,s_area,suction_grade,water_grade,remember_map,has_map,is_mop,has_newmap Options: [object Object]
[2021-01-18T21:25:25.871Z] [DEBUG] Received command {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"]}
[2021-01-18T21:25:25.873Z] [TRACE] >>> cloud: {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"],"id":10}
[2021-01-18T21:25:25.891Z] [DEBUG] get_prop <<< cloud: {"result":[5,0,2105,100,1,0,26,20.04,3,12,1,1,0,1],"id":10}
```
[image]

v0d0r, 2021-01-20:
Not to hijack this issue but out of interest @panoslouk does your Zones work at all?  The only thing I get working in Zones is restricted areas.  As this is my first time on Valentudo I am not sure if this is meant to work and/or if it work in previous releases.  Another thing that also does not work for me is the setting of the vacuum from max,medium low etc.  Curious to hear if yours are working or if its just an issue on my end.

panoslouk, 2021-01-20:
No zones and fan speed doesn't doesn't work. :/ Maybe this issue have to do with pre-release version of valetudo. I'm waiting for the next stable version to try it out.

v0d0r, 2021-01-20:
> No zones and fan speed doesn't doesn't work. :/ Maybe this issue have to do with pre-release version of valetudo. I'm waiting for the next stable version to try it out.

Thanks for confirming. I will also wait and see how things pan out on the next release. For now I am super happy with the features working as is.  Well done to the entire dev team.

atd, 2021-01-24:
I can also confirm that using the old code fixes the issue for me

Running 3.5.3_0047

Depau, 2021-01-25:
> In the old code, the response array was mapped to the requested properties before then parsing it: [...]

Hi, I can't give you a proper answer right now, but from what I can remember, yes, you send "get_prop" with all the data you need and it replies with that array.

I think the issue though might have to do with this line:

```js
         this.parseAndUpdateState(response[0]);  // why [0]?
```

I don't remember any issues with the status polling code but I may have f'd up and never realized, and it looks like this might be the case.

Keep in mind that when I wrote the code the UI wasn't wired up to the capabilities system yet so I never got the chance to test it properly.

<small>I'm planning to finish up with the Viomi capabilities soon™ after I finish up with what I'm working right now ;)</small>

rromrrom, 2021-01-28:
Hi,
I have same issue with my viomi.v8
When valetudo requests vacuum state (pollState()) it got something like this

response: [5,0,2105,100,1,0,26,20.04,3,12,1,1,0,1]

but on the next step in parseAndUpdateState() it expects dictionary with named fields instead of array of values like data["battary_life"] and so on.
So there is no way this would work

I can confirm that @panoslouk fix above with statusDict mapping solves this particular issue for me.

rumpeltux, 2021-01-31:
This particular issue should be fixed in current HEAD. Much of the rest of the UI still isn’t functional though, looking forward to @Depau’s implementation of the missing capabilities.

Note that [installation instructions](https://valetudo.cloud/pages/installation/viomi.html#robot-setup) have [changed](https://github.com/Hypfer/Valetudo/commit/a38987affba30f37e0cd09a67627a51adf695679#diff-4005931a921cf15818e21388070f3da9fb515b88b9f2296f35012870bab3c3cd) so when redeploying a new binary you’ll need some adjustments.

closed 2021-01-31 by rumpeltux