Hypfer/Valetudo #643: Feature request: browser notification

OhSoGood, 2021-01-08:
Hi All,
For a standalone installation (no HomeAssistant, no mqtt) where the vacuuum is used only though Valetudo web interface, would it be possible to get notification from the website in the browser ( a standard web push - https://www.w3.org/TR/push-api/ )? Example of scenario: I start the vacuum in Valetudo web interface, and I'd like to get informed with a browser popup when the vaccum has finished its work. 

Is that doable? Or is there already a existing way to get this?

Hypfer, 2021-01-08:
https://developer.mozilla.org/en-US/docs/Web/API/Notifications_API/Using_the_Notifications_API

```
Secure context
This feature is available only in secure contexts (HTTPS), in some or all supporting browsers.
```

Since there's no feasible way to have proper SSL Certificates with auto-renewal on the robot inside your home network, this isn't feasible.

I'd suggest using something like Home Assistant do implement notifications. You can even build actionable notifications there as can be seen here:
https://companion.home-assistant.io/docs/notifications/actionable-notifications/

closed 2021-01-08 by Hypfer

rumpeltux, 2021-01-09:
> Since there's no feasible way to have proper SSL Certificates with auto-renewal on the robot inside your home network, this isn't feasible.

Actually, SSL support would be quite nice and it may be feasible: I think one can achieve this by configuring a free dns domain that supports TXT records and using letsencrypt’s DNS validation.