Hypfer/Valetudo #657: Can't change Wifi network from ui

ministryofsillywalks, 2021-01-20:
## Describe the bug

When setting up the robot I added it to my general Network. Now that its up and running I wanted to add it to a different Wifi running on its own VLAN.
I can input the new network and password in the gui.
When I think click on "Save new Wifi configuration" I get the prompts to confirm and the Alert that the changes were successfully applied.
However the robot never changes the network and just stays on the old wifi.
After reloading the page (same IP) the SSID has reverted back to the previous Network.

## To Reproduce

Steps to reproduce the behavior:
1. Go to Settings
2. Click on Wifi
3. Try and change the SSID
4. See error

### Screenshots

### Vacuum Model

Gen1

### Valetudo Version

0.6.1
### Firmware Version

? (ui only shows "?")
### User-Agent

Mozilla Firefox, Chromium, tried on different browsers
## Expected behavior

Inputting a new SSID should have the robot change to the new network.
## Additional context

Is a restart needed? If so this should be written somewhere in one of the prompts.

Hypfer, 2021-01-24:
This is most likely caused by an invalid SSID and/or password either because it was entered incorrectly or because it contains characters like the `\`

ministryofsillywalks, 2021-01-25:
Sorry to report this but, the SSID is 100% corrent and so is the password. The PW doesnt contain "/" but it does contain a blank space " " could that be the issue?

Hypfer, 2021-01-25:
Spaces should work.

Looking at the overall download count of 0.6.1 as well as my personal experience, I can assure you that this is not a bug in Valetudo

johnhamelink, 2021-01-30:
I'm also experiencing this issue, my SSID also has spaces in it. My SSID is `The Promised LAN`. Changing this to an SSID without spaces did indeed allow my Roborock S55 to join the LAN.

ministryofsillywalks, 2021-01-30:
For me the SSID doesnt have spaces but the password does.
I switched to rrand fork and now its working...Kinda weird.

Hypfer, 2021-01-30:
So we're at 

"It works with Valetudo but not RE" as well as "It works with RE but not Valetudo".

As I've said before: This is not an issue with any codebase but rather with the Wifi, the user etc.
You may also rephrase it as a Layer 8 error.

ministryofsillywalks, 2021-01-30:
I'm not too stupid to enter my ssid and pw correctly. I tried it 10 times in valetudo.
Then flashed rrand and first try it works and connects to the new WiFi. 
But if you are so sure the problem is in front of the screen then I guess that's an easy solution for you.
I think the download numbers for 0.6.1 mean little because what percentage of the people who download it will switch WiFi if the bot is already connected to a WiFi? I'm guessing less than 1%.

Hypfer, 2021-06-19:
Whoops.
You were right about this.

Valetudo does fail to set new Wifi Credentials when running on the robot itself, which is why it was never discovered during local testing, since that only uses the local interface.

The firmware seems to simply ignore the `miIO.config_router` command when it's received from the cloud socket which does make sense. Unfortunately at some point, Valetudo started automatically redirecting all communication through the cloud socket if it's available because that worked much more reliably.

Apparently that wasn't without issues.
Sorry about that

It was even visible in the log. Oh well
`[2021-06-19T11:31:14.169Z] [DEBUG] <<< cloud: {"id":14,"result":"unknown_method"}`

closed 2021-06-19 by Hypfer

d3wy, 2021-06-21:
Thank god this got reopened, I'm here as a result of googling that I gave up on 2 months ago where I saw ministry get roasted for reporting this thinking to myself there is no way two of us are 'making a mistake entering the ssid
@Hypfer I note this was closed right after, does that mean a fix can't be/won't be worked on for this? Do we have any work around?
I am facing the same issue as @ministryofsillywalks with the same hardware.

Hypfer, 2021-06-21:
This way closed by a commit which contains a fix.

commit 32a12cf8 (2021-06-19): fix(miio): Allow setting new wifi credentials when provisioned

fixes #657