Hypfer/Valetudo #658: Issue or expected behaviour?

xefil, 2021-01-20:
Hi all,

I ask here to understand if what I'm noticing is an issue or expected behaviour, and a little help.

I was trying to install Valetudo on my Gen1 Vacuum Xiaomi. After some tries without success (errors like Error: `Unable to parse message payload`), I've seen that my firmware was maybe too new to be replaced with the Valetudo ones. The info query has worked as always, like here:

```
mirobo --token MY-TOKEN --ip 192.168.3.61
State: Charging
Battery: 100 %
Fanspeed: 90 %
Cleaning since: 0:54:30
Cleaned area: 57.485 m²
Water box attached: False
```
or

```
# mirobo --token MY-TOKEN--ip 192.168.3.61 info
rockrobo.vacuum.v1 v3.5.4_004007 (34:CE:00:E8:7E:46) @ 192.168.3.61 - token: MY-TOKEN
```

So I've read that with a factory reset I should then have an old firmware to start with.
I've made a factory reset and then add the device to my Mii Home app succesfully. To discover then the new token I've used the Token Extractor from the Cloud:

https://github.com/PiotrMachowski/Xiaomi-cloud-tokens-extractor

Doin so I was able to query the device back again:

```
# mirobo --token MY-NEW-TOKEN --ip 192.168.3.61 info                                                            
rockrobo.vacuum.v1 v3.3.6_003062 (34:CE:00:E8:7E:46) @ 192.168.3.61 - token: MY-NEW-TOKEN
```

Ok, then, trying to upload the firmware it worked as expected, wow!
I was so able to ssh into the vacuu, and open the web interface, as well as discover it into home assistant.

Now, the query gives me the following output about the firmware:

```
# mirobo --token MY-NEW-TOKEN --ip 192.168.3.61 info
rockrobo.vacuum.v1 v3.5.8_004018 (34:CE:00:E8:7E:46) @ 192.168.3.61 - token: MY-NEW-TOKEN
```

And ssh works!

```
Welcome to Ubuntu 14.04.3 LTS (GNU/Linux 3.4.39 armv7l)

 * Documentation:  https://help.ubuntu.com/
Last login: Wed Jan 20 07:01:43 2021 from 192.168.3.114

          _______  _______                    _______ 
|\     /|(  ___  )(  ____ \|\     /||\     /|(       )
| )   ( || (   ) || (    \/| )   ( || )   ( || || || |
| |   | || (___) || |      | |   | || |   | || || || |
( (   ) )|  ___  || |      | |   | || |   | || ||_|| |
 \ \_/ / | (   ) || |      | |   | || |   | || |   | |
  \   /  | )   ( || (____/\| (___) || (___) || )   ( |
   \_/   |/     \|(_______/(_______)(_______)|/     \|
                                              20210119
======================================================
MODEL...........: rockrobo.vacuum.v1
SERIAL..........: MY-SERIAL
PRODUCTION DATE.: UNKNOWN
FIRMWARE........: 3.5.8_004018
BUILD NUMBER....: 2020052901REL
MIIO VERSION....: 3.3.9
REGION..........: 
IP..............: 192.168.3.61
MAC.............: 34:CE:00:E8:7E:46
TOKEN...........: MY-NEW-TOKEN
DID.............: MY-DID
KEY.............: MY-KEY
======================================================
```

Well, then what is the issue?

I've noticed that opening the Mii Home app, I get a notice that wants to update the firmware to 3.3.9_003194 or higher. I can only quit or update. Of course actually I'm quitting. But in this way I cannot change the default language spoken from the robot which is chinese.

The questions are then:

1) Is the firmware really on 3.5.8 or does it have a 3.3 version?

2) I've read that the algorythm that cleans the house is better on latest firmware, so is that then applied?

3) how cna i change the language then?

4) and is then the vacuum definitly no more controllable via Xiaomi App? Which would be ok from me.

Thanks for the help!

Simon

Hypfer, 2021-01-20:
Hi there,

your issue has been closed, because it is either a support request or something else that would usually be handled by 1st level support.

Since this is not a commercial project, but rather just something I've built for myself and then shared with the world for free, I really can't provide said support.

Therefore, please direct this request somewhere else, where other users might be able to help you.
The Telegram group would be one option but certainly isn't the only one.

Thank you for understanding as well as your interest in the project.

Hypfer, 2021-01-20:
Furthermore, whatever you're doing there isn't even remotely related to the installation procedure described in the docs.

Not entirely sure why you're doing that but even if this Issue Tracker was for support, it would only provide support for deployments as documented

closed 2021-01-20 by Hypfer

xefil, 2021-01-20:
Hi @Hypfer . First of all thanks for your great work, that's awesome. Due I'm new to this world, the question was exactly to understand if this *IS* an issue for some reasons of it's expected. You answer gives me to think it's expected, ok. Then, about the other questions, you'll have for sure better to do instead to give support for free :-) 
You mentioned a telegram group, ok, could then give me some hints on where ask for my questions? Or where to join the telegram group?
I wish you a nice day, Tschüß und danke sehr nochmals, servus, Simon