Hypfer/Valetudo #662: Eufy Vacuum Support

oxr463, 2021-01-22:
Placeholder for now.

Need to research what is required to make this happen.

Hypfer, 2021-01-22:
The only interesting Model seems to be the eufy Robovac L70.

Here are some teardown photos of that one. Thank you, FCC

https://fccid.io/2AOKB-T2190/Internal-Photos/Internal-Photos-4373122

It doesn't seem to be a rebranded 3irobotix vacuum robot which is refreshing

oxr463, 2021-01-22:
I have one of these from Walmart:

- [T2123](https://www.eufylife.com/products/variant/robovac-25c/T2123111)

Hypfer, 2021-01-22:
There's no point in supporting the RoboVac 25C in Valetudo since it has no understanding of its surroundings.

Valetudo is made for (Vacuum) robots with maps

oxr463, 2021-01-22:
> There's no point in supporting the RoboVac 25C in Valetudo since it has no understanding of its surroundings.
> 
> Valetudo is made for (Vacuum) robots with maps

Oh I see. Not for bump navigation. Should we close this issue then? If someone with a valid model would like to see it supported, then they can either make a new issue or ask to re-open this one.

closed 2021-01-22 by Hypfer