Hypfer/Valetudo #674: Firefox not showing webpage (blank page)

deku-m, 2021-02-02:
## Describe the bug

When going to the valetudo webpage with firefox. Firefox isnt showing the page completly.
Only shows hamburger menu icon, home and below navitgation. The rest of functions arent shown.
Looks like error in js file for firefox?

## To Reproduce

Steps to reproduce the behavior:
1. Go to Firefox
2. Got to ip of valetudo interface
3. See blank page.
4. See error with debugger console. (added as image)

### Screenshots

[image]
[image]

### Vacuum Model

Xiaomi Miija STYJ02YM 

### Valetudo Version

2021.01.1
### Firmware Version

2021.01.1

### User-Agent

Firefox 85.0 / Win10
## Expected behavior

If possible to see a working display on firefox.

## Additional context

Workaround is to use Edge

Hypfer, 2021-02-02:
Development is done using Firefox on W10 so this should definitely work.

Maybe there are Add-Ons interfering? Unstable network connection? Try a hard refresh (ctrl + f5) and please also check the network tab for possibly missing resources

deku-m, 2021-02-02:
Thanks Hypfer for the fast response and help pointing me to the solution.
Its because my anti-virus is scanning/interfering the js files that i didnt know about.

closed 2021-02-02 by deku-m

Hypfer, 2021-02-02:
Just for reference: Which antivirus software is it?

deku-m, 2021-02-02:
Kaspersky Antivirus. Located under network settings. 
I disabled scan secure traffic and inject script into webtraffic.

In the debug on webconsole of firefox there is mentioning a kaspersky... js file.