Hypfer/Valetudo #676: Problem with map_data over MQTT after 2021.01.1 update

alceasan, 2021-02-03:
## Describe the bug
After 2021.01.1 update my Roborock v1 sends incorrect data over MQTT for mapping.

## To Reproduce

1. Make a clean update following official guide.
2. Activate MQTT server and provide map data, but not base64 encoded compressed
3. Configure MQTT sensor for map_data on Home Assistant:
```
  - platform: mqtt
    state_topic: "valetudo/roborock/state"
    json_attributes_topic: "valetudo/roborock/map_data"
    name: roborock_map
```
4. Home Assistant MQTT sensor for map data complains about 'Can't decode payload'

### Screenshots

[image]

### Vacuum Model

Xiaomi Roborock v1

### Valetudo Version

2021.01.1

### Firmware Version

3.5.8_004018

## Expected behavior

Map data was provided over MQTT in version 0.6.1 using this sensor on Home Assistant. I have other roborock v2 on 0.6.1 version that I haven't updated and is still working with the same configuration.

## Additional context

MQTT connection with Home Assistant is working, I can control vacuum over MQTT and receive other information, but have this problem with map data.

Hypfer, 2021-02-03:
You need to update your sensor configuration.

Since ~August 2020, maps are deflated before they are sent
https://github.com/Hypfer/Valetudo/commit/118705d4a98a3e41f29e5965a4bd757513926126

That wasn't mentioned in the release notes though. sorry about that.

Check out
https://github.com/TheLastProject/lovelace-valetudo-map-card

and ctrl + f "In this case the sensor should look like this:"

I'm currently thinking about how this process could be improved

alceasan, 2021-02-03:
Thank you, now it's working, but I have to activate base64 encoded compressed option for it to work

I would be nice to add this information to documentation or changelog.

closed 2021-02-03 by alceasan

Hypfer, 2021-02-17:
This has been improved drastically in [2021.02.0](https://github.com/Hypfer/Valetudo/releases/tag/2021.02.0)