Hypfer/Valetudo #678: Valetudo not able to reach the robot

ebrithiljonas, 2021-02-03:
## Describe the bug

Valetudo starts up all fine with the following output:

```
[2021-02-03T14:34:04.609Z] [INFO] Loading configuration file: /mnt/data/valetudo_config.json
[2021-02-03T14:34:04.640Z] [INFO] Autodetected RoborockS6PureValetudoRobot
[2021-02-03T14:34:04.807Z] [INFO] Starting Valetudo 2021.01.1
[2021-02-03T14:34:04.808Z] [INFO] Robot: Beijing Roborock Technology Co., Ltd. S6 Pure (RoborockS6PureValetudoRobot)
[2021-02-03T14:34:04.809Z] [INFO] JS Runtime Version v14.4.0
[2021-02-03T14:34:04.810Z] [INFO] Max Heap Size: 33.5 MiB
[2021-02-03T14:34:04.810Z] [INFO] Node Flags: --max-old-space-size=32
[2021-02-03T14:34:04.812Z] [INFO] DeviceId 322058607
[2021-02-03T14:34:04.812Z] [INFO] IP 127.0.0.1
[2021-02-03T14:34:04.813Z] [INFO] CloudSecret *********************
[2021-02-03T14:34:04.814Z] [INFO] LocalSecret ********************
[2021-02-03T14:34:04.983Z] [INFO] Dummycloud is spoofing 203.0.113.1:8053 on 203.0.113.1:8053
[2021-02-03T14:34:04.985Z] [INFO] Webserver running on port 80
[2021-02-03T14:34:04.992Z] [INFO] Map Upload Server running on port 8079
```

But after connecting to the Web interface it shows the following errors:
```
[2021-02-03T14:34:19.183Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 10, method: 'get_status', args: {} }
[2021-02-03T14:34:19.206Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 20, method: 'get_status', args: {} }
.
.
.
[2021-02-03T14:34:21.188Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 10000, method: 'get_status', args: {} }
[2021-02-03T14:34:21.341Z] [ERROR] Unable to reach vacuum
[2021-02-03T14:34:21.343Z] [INFO] exiting with code 1...
[2021-02-03T14:34:21.346Z] [ERROR] Stacktrace that lead to the process exiting: Error
    at process.<anonymous> (/snapshot/Valetudo/index.js:34:70)
    at process.emit (events.js:315:20)
    at process.exit (internal/process/per_thread.js:167:15)
    at Timeout._onTimeout (/snapshot/Valetudo/lib/miio/RetryWrapper.js:125:29)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
[root@rockrobo ~]# rr_boot_flags -s
```

## To Reproduce

1. Install latest image from [Dustbuilder](https://builder.dontvacuum.me/_a08.html)
2. Open the Web interface
3. Read the logfile

### Vacuum Model

Roborock S6 Pure

### Valetudo Version

2021.01.1

### Firmware Version

ver 0938, 10/2020

### User-Agent

Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:84.0) Gecko/20100101 Firefox/84.0

## Expected behavior

Valetudo should connect to the robot and should show the information about it on the Web interface instead of crashing.

bmhorn, 2021-02-05:
I just did the update of valetudo the dustbuilder firmware and i can see the same warning in the logs. But not the error you receiving.

ebrithiljonas, 2021-02-08:
@bmhorn What does your configuration file look like?

bmhorn, 2021-02-09:
{
  "embedded": true,
  "robot": {
    "implementation": "auto",
    "implementationSpecificConfig": {
      "ip": "127.0.0.1"
    }
  },
  "webserver": {
    "port": 80,
    "basicAuth": {
      "enabled": false,
      "username": "valetudo",
      "password": "valetudo"
    }
  },
  "zonePresets": {
  },
  "goToLocationPresets": {},
  "mqtt": {
    "enabled": true,
    "server": "MY_SECRET_BROKER_IP",
    "port": "1883",
    "clientId": "valetudo",
    "username": "MY_SECRET_USERNAME",
    "password": "MY_SECRET_PASSWORD",
    "usetls": false,
    "ca": "",
    "clientCert": "",
    "clientKey": "",
    "qos": 0,
    "identifier": "robot",
    "topicPrefix": "valetudo",
    "autoconfPrefix": "homeassistant",
    "provideMapData": false,
    "base64EncodeMapData": true
  },
  "allowSSHKeyUpload": true,
  "logLevel": "info",
  "debug": {
    "memoryStatInterval": false,
    "enableRawCommandCapability": false
  }
}

bmhorn, 2021-02-11:
Some days ago I installed the newest valetudo version. The warning does not reappear in the log.
Maybe you also do a reinstall.

ebrithiljonas, 2021-02-11:
I have re flashed the robot multiple times, do you also have an S6 Pure and how do you exactly reinstall valetudo?

bmhorn, 2021-02-11:
Yes I also have an S6 Pure.
I deployed valetudo on my PC in VSCode and than just copy&paste (via scp) on the robot.
You should check that the binary with new firmwares has to be named "valetudo". Old firmware versions expect "valetudo.bin".

ebrithiljonas, 2021-02-11:
I have directly downloaded the valetudo binary and made it executable. And when I run it I get these errors.

This is my _rootsh file:

if [[ -f /mnt/data/valetudo ]]; then
#	iptables         -F OUTPUT
#	ip6tables        -F OUTPUT
#	iptables  -t nat -F OUTPUT
#	iptables  -t nat -A OUTPUT -p tcp --dport 80   -d 203.0.113.1 -j DNAT --to-destination 127.0.0.1:8053
#	iptables  -t nat -A OUTPUT -p udp --dport 8053 -j DNAT --to-destination 127.0.0.1:8053
#	iptables         -A OUTPUT                     -d 203.0.113.1/32  -j REJECT
#	ip6tables        -A OUTPUT                     -d 2001:db8::1/128 -j DROP
	ip addr add 203.0.113.1 dev lo
	VALETUDO_CONFIG_PATH=/mnt/data/valetudo_config.json /mnt/data/valetudo >> /tmp/valetudo.log 2>&1 &

bmhorn, 2021-02-11:
Okay that is exactly the same as the one I use.

Which stock firmware version was installed at your robot?
I did an firmware upgrade with the Roborock app before I started "working" with the robot.

ebrithiljonas, 2021-02-11:
I did that too, it was on the latest patch when I gained ssh access and installed the modified Firmware. 
You mentioned "deploying Valetudo on your PC with VSCode", what exactly do you mean with this? Have you done more steps then just copying the binary over and installing the root.sh file?

bmhorn, 2021-02-11:
Okay everything sounds identical to the steps i did.

Deploying means in my case:

1. clone this repository
2. install the dependencies (i did that in a container, but it can also be done global)
3. open it in vscode and execute the build script.
4. copy the binary to the robot.

ebrithiljonas, 2021-02-11:
Ok, I haven't built it myself I used the prebuilt release binary, I will try building it myself later this day, thank you a lot so far for your valuable help.

ebrithiljonas, 2021-02-11:
Ok I did it, now at least there is a different error now:

```
[2021-02-11T19:47:42.923Z] [INFO] No configuration file present. Creating one at: /tmp/valetudo_config.json
[2021-02-11T19:47:42.951Z] [INFO] Autodetected RoborockS6PureValetudoRobot
[2021-02-11T19:47:43.099Z] [INFO] Starting Valetudo 2021.01.1
[2021-02-11T19:47:43.101Z] [INFO] Robot: Beijing Roborock Technology Co., Ltd. S6 Pure (RoborockS6PureValetudoRobot)
[2021-02-11T19:47:43.101Z] [INFO] JS Runtime Version v14.4.0
[2021-02-11T19:47:43.103Z] [INFO] Max Heap Size: 33.5 MiB
[2021-02-11T19:47:43.104Z] [INFO] Node Flags: --max-old-space-size=32
[2021-02-11T19:47:43.113Z] [INFO] DeviceId 322058607
[2021-02-11T19:47:43.114Z] [INFO] IP 127.0.0.1
[2021-02-11T19:47:43.115Z] [INFO] CloudSecret ***********************
[2021-02-11T19:47:43.116Z] [INFO] LocalSecret ************************
[2021-02-11T19:47:43.214Z] [INFO] exiting with code 1...
[2021-02-11T19:47:43.217Z] [ERROR] Stacktrace that lead to the process exiting: Error
    at process.<anonymous> (/snapshot/Valetudo/index.js:34:70)
    at process.emit (events.js:315:20)
    at process._fatalException (internal/process/execution.js:176:19)
events.js:292
      throw er; // Unhandled 'error' event
      ^

Error: bind EADDRINUSE 127.0.0.1:8053
    at dgram.js:337:20
    at processTicksAndRejections (internal/process/task_queues.js:85:21)
    at process.runNextTicks [as _tickCallback] (internal/process/task_queues.js:66:3)
    at Function.Module.runMain (pkg/prelude/bootstrap.js:1376:13)
    at internal/main/run_main_module.js:17:47
Emitted 'error' event on Socket instance at:
    at dgram.js:339:14
    at processTicksAndRejections (internal/process/task_queues.js:85:21)
    [... lines matching original stack trace ...]
    at internal/main/run_main_module.js:17:47 {
  errno: -98,
  code: 'EADDRINUSE',
  syscall: 'bind',
  address: '127.0.0.1',
  port: 8053
}

```

bmhorn, 2021-02-11:
Did you checked via htop that only one instance of valetudo is running?
The error "Error: bind EADDRINUSE 127.0.0.1:8053" looks a little bit like multiple instances are running.
Maybe you first kill valetudo via htop and then start it new from the commandline. You can just use this command for that
`VALETUDO_CONFIG_PATH=/mnt/data/valetudo_config.json /mnt/data/valetudo >> /tmp/valetudo.log 2>&1 &`

You also started valetudo without a config. I dont think that this should result in an error, so this just an information
`No configuration file present. Creating one at: /tmp/valetudo_config.json`

ebrithiljonas, 2021-02-11:
You were right, I stopped valetudo with htop and then it could start again, but now I am back to the same error again, and I tried compiling the current master and the commit you were on, on both came the exact same error again.

bmhorn, 2021-02-12:
Hmm okay. That is strange. Maybe you can change the loglevel via the config-file. Allowed values should be {"trace"|"debug"|"info"|"warn"|"error"}. Hopefully this gives us more information about the issue.

ebrithiljonas, 2021-02-12:
I have done that and this is the output, there isn't really more information in it as it seems:

```

[2021-02-12T14:27:45.303Z] [INFO] Loading configuration file: /tmp/valetudo_config.json
[2021-02-12T14:27:45.327Z] [TRACE] Trying to open device.conf at /mnt/default/device.conf
[2021-02-12T14:27:45.330Z] [TRACE] Trying to open device.conf at /mnt/default/device.conf
[2021-02-12T14:27:45.332Z] [TRACE] Trying to open device.conf at /mnt/default/device.conf
[2021-02-12T14:27:45.334Z] [TRACE] Trying to open device.conf at /mnt/default/device.conf
[2021-02-12T14:27:45.336Z] [TRACE] Trying to open device.conf at /mnt/default/device.conf
[2021-02-12T14:27:45.337Z] [INFO] Autodetected RoborockS6PureValetudoRobot
[2021-02-12T14:27:45.436Z] [TRACE] Trying to read token file at /mnt/data/miio/device.token
[2021-02-12T14:27:45.449Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:45.458Z] [TRACE] Trying to open device.conf at /mnt/default/device.conf
[2021-02-12T14:27:45.461Z] [TRACE] Trying to open device.conf at /mnt/default/device.conf
[2021-02-12T14:27:45.496Z] [INFO] Starting Valetudo 2021.01.1
[2021-02-12T14:27:45.498Z] [INFO] Robot: Beijing Roborock Technology Co., Ltd. S6 Pure (RoborockS6PureValetudoRobot)
[2021-02-12T14:27:45.498Z] [INFO] JS Runtime Version v14.4.0
[2021-02-12T14:27:45.501Z] [INFO] Max Heap Size: 33.5 MiB
[2021-02-12T14:27:45.501Z] [INFO] Node Flags: --max-old-space-size=32
[2021-02-12T14:27:45.504Z] [DEBUG] Starting NTP Query for { server: 'pool.ntp.org', port: 123, timeout: 10000 }
[2021-02-12T14:27:45.525Z] [TRACE] Trying to open device.conf at /mnt/default/device.conf
[2021-02-12T14:27:45.527Z] [INFO] DeviceId 322058607
[2021-02-12T14:27:45.528Z] [INFO] IP 127.0.0.1
[2021-02-12T14:27:45.528Z] [TRACE] Trying to open device.conf at /mnt/default/device.conf
[2021-02-12T14:27:45.529Z] [INFO] CloudSecret ****************
[2021-02-12T14:27:45.530Z] [TRACE] Trying to read token file at /mnt/data/miio/device.token
[2021-02-12T14:27:45.531Z] [INFO] LocalSecret ****************
[2021-02-12T14:27:45.636Z] [INFO] Dummycloud is spoofing 127.0.0.1:8053 on 127.0.0.1:8053
[2021-02-12T14:27:45.640Z] [TRACE] Intercepting dns.lookup call for pool.ntp.org
[2021-02-12T14:27:45.646Z] [INFO] Webserver running on port 80
[2021-02-12T14:27:45.649Z] [INFO] Map Upload Server running on port 8079
[2021-02-12T14:27:45.675Z] [DEBUG] Got Time from NTP Server: 2021-02-12T14:27:45.889Z
[2021-02-12T14:27:45.005Z] [INFO] Successfully set the robot time via NTP to 2021-02-12T14:27:45.889Z
[2021-02-12T14:27:45.007Z] [DEBUG] Next NTP sync in 28800000 ms
[2021-02-12T14:27:45.060Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:45.365Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:45.668Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:45.972Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:46.276Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:46.580Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:46.884Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:47.188Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:47.491Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:47.794Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:48.097Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:48.401Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:48.705Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:49.008Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:49.311Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:49.615Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:49.918Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:50.222Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:50.525Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:50.829Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:51.132Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:51.435Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:51.738Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:52.052Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:52.355Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:52.658Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:52.967Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:53.269Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:53.573Z] [DEBUG] >>> local: HandshakePacket()
[2021-02-12T14:27:53.741Z] [TRACE] Trying to read token file at /mnt/data/miio/device.token
[2021-02-12T14:27:53.744Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 10, method: 'get_status', args: {} }
[2021-02-12T14:27:53.751Z] [TRACE] Trying to read token file at /mnt/data/miio/device.token
[2021-02-12T14:27:53.752Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 20, method: 'get_status', args: {} }
[2021-02-12T14:27:53.755Z] [TRACE] Trying to read token file at /mnt/data/miio/device.token
[2021-02-12T14:27:53.757Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 30, method: 'get_status', args: {} }
...
[2021-02-12T14:27:56.951Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 9990, method: 'get_status', args: {} }
[2021-02-12T14:27:56.953Z] [TRACE] Trying to read token file at /mnt/data/miio/device.token
[... cut ...]
[2021-02-12T14:27:57.108Z] [ERROR] Unable to reach vacuum
[2021-02-12T14:27:57.110Z] [INFO] exiting with code 1...
[2021-02-12T14:27:57.112Z] [ERROR] Stacktrace that lead to the process exiting: Error
    at process.<anonymous> (/snapshot/Valetudo/index.js:34:70)
    at process.emit (events.js:315:20)
    at process.exit (internal/process/per_thread.js:167:15)
    at Timeout._onTimeout (/snapshot/Valetudo/lib/miio/RetryWrapper.js:125:29)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
```

bmhorn, 2021-02-13:
Yeah you are right. It looks like something is wrong with the Token.
Unfortunately I do not have further ideas at the moment. Sorry.

akselii, 2021-02-16:
I had this same problem with my new S5 Max, the valetudo web page just saying Loading state and log file giving that Token is okay... line.
I reflashed and tried different binaries of valetudo and finally what helped was just resetting the wifi...

Hypfer, 2021-02-16:
> Dummycloud is spoofing 203.0.113.1:8053 on 203.0.113.1:8053

As you've already figured out, this is the issue. It happens, because the S5Max etc root guide assumes a newer Valetudo version than the one that is available in the releases section. This will be fixed with the next release and should hopefully just work.

Until then you can either wait or build the latest master yourself and deploy that

ebrithiljonas, 2021-02-16:
I just recompiled it with the current master branch, then it gave the same error but didn't crash. Then I di a Wifi reset and everything worked. Thanks for all the help here.

closed 2021-02-16 by ebrithiljonas