Hypfer/Valetudo #693: MQTT & HA: not all items populated

Kugelfang666, 2021-02-17:
## Describe the bug

Valetudo 2021_2 installed and connected to MQTT Server on HA, I can see in MQTT Topics for the Consumables, however they don't become populated in Home Assistant (see screenshot)

## To Reproduce

### Screenshots

### Vacuum Model

Gen 1 
### Valetudo Version

2021_0
[image]
2
## Expected behavior

## Additional context

Hypfer, 2021-02-17:
Yep that happens. It's not a bug in Valetudo though but rather in the home assistant autodiscovery logic.
My hypothesis is, that it doesn't handle newly appearing entities with the same device info properly.

Unfortunately, there's nothing I can do about that

Deleting the device and restarting valetudo + ha a few times is the only workaround available

closed 2021-02-17 by Hypfer

Kugelfang666, 2021-02-17:
ok i See rebooting the robot a few Times indeed fixed it for me

Hypfer, 2021-02-17:
See also: #686