Hypfer/Valetudo #696: Roborock S6 fresh install: Token is okay, however we're unable to reach the vacuum

gschnitzler, 2021-02-17:
## Describe the bug
WebUI is stuck in state loading with a timeout error. Log contains this:

[2021-02-17T11:07:01.499Z] [INFO] Firmware Version: 3.5.8_2516
[2021-02-17T11:07:01.673Z] [INFO] Dummycloud is spoofing 127.0.0.1:8053 on 127.0.0.1:8053
[2021-02-17T11:07:01.681Z] [INFO] Webserver running on port 80
[2021-02-17T11:07:01.688Z] [INFO] Map Upload Server running on port 8079
[2021-02-17T11:07:01.003Z] [INFO] Successfully set the robot time via NTP to 2021-02-17T11:07:01.890Z
[2021-02-17T11:07:43.829Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 10, method: 'get_status', args: {} }
[2021-02-17T11:07:43.844Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 20, method: 'get_status', args: {} }
[2021-02-17T11:07:43.848Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 30, method: 'get_status', args: {} }
[2021-02-17T11:07:43.852Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 40, method: 'get_status', args: {} }
[2021-02-17T11:07:43.855Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 50, method: 'get_status', args: {} }
[2021-02-17T11:07:43.858Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 60, method: 'get_status', args: {} }
[2021-02-17T11:07:43.861Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 70, method: 'get_status', args: {} }
[2021-02-17T11:07:43.864Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 80, method: 'get_status', args: {} }
[2021-02-17T11:07:43.867Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 90, method: 'get_status', args: {} }
[2021-02-17T11:07:43.871Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 100, method: 'get_status', args: {} }
[2021-02-17T11:07:43.873Z] [ERROR] Unable to reach vacuum. Giving up after 100 tries

On the serial console, i can see this (but i think this is unrelated):

[3.37] start logging
[3.41] init network
[3.43] insmod kernel modules
[3.54] mount all fs
[3.94] play boot ring
[3.97] init watchdoge
kernel.softlockup_panic = 1
[4.99] init adbd
1+0 records in
1+0 records out
1 byte (1 B) copied, 0.000974458 s, 1.0 kB/s
1+0 records in
1+0 records out
1 byte (1 B) copied, 0.00134017 s, 0.7 kB/s
/etc/init/S10rc_local: line 38: log_begin_msg: not found
ip6tables v1.4.21: Couldn't load target `REJECT':No such file or directory

Try `ip6tables -h' or 'ip6tables --help' for more information.
/etc/init/S10rc_local: line 38: log_end_msg: not found
[5.13] start valetudo-daemon
install_listener('tcp:5037','*smartsocket*')

rockrobo login:

## To Reproduce
I rooted a new S6 using Dennis’s Dustbuilder. I only used the Roborock app to hook up wifi.
Above errors occur once i open the WebUI. Rebooted several times and also reset wifi. No dice.

### Screenshots

### Vacuum Model
Roborock S6

### Valetudo Version
2021.1.1 and 2022.2.0

## Expected behavior

## Additional context

Hypfer, 2021-02-17:
Please check `/mnt/data/miio/wifi.conf` (or similar) for the value of `cfg_by`

gschnitzler, 2021-02-17:
Its cfg_by=rriot

Hypfer, 2021-02-17:
That should be `cfg_by=miot`.

Also, make sure that your `device.country` file is not set to `xx`. A valid value could be `de` or `us`

closed 2021-02-17 by Hypfer