Hypfer/Valetudo #702: Can't get any properties from vacuum

HydrelioxGitHub, 2021-02-18:
## Describe the bug

I've installed Valetudo through the script from @rumpeltux . I have update the link and the sha256 to download the last version of valetudo. I can reach the web interface but nothing can be retreive from the vacuum.
## To Reproduce

### Screenshots

[image]

### Vacuum Model

STYJ02YM

### Valetudo Version

2021.02.0

## Expected behavior

Get info from vacuum in the Home screen
## Additional context

This is my log file
```
[2021-02-18T13:16:11.968Z] [INFO] Set Logfile to /tmp/valetudo.log
[2021-02-18T13:16:11.982Z] [INFO] Autodetected ViomiV7ValetudoRobot
[2021-02-18T13:16:12.134Z] [INFO] Starting Valetudo 2021.02.0
[2021-02-18T13:16:12.135Z] [INFO] Configuration file: /tmp/valetudo_config.json
[2021-02-18T13:16:12.135Z] [INFO] Logfile: /tmp/valetudo.log
[2021-02-18T13:16:12.136Z] [INFO] Robot: Viomi Technology Co., Ltd Xiaomi Mijia STYJ02YM (ViomiV7ValetudoRobot)
[2021-02-18T13:16:12.137Z] [INFO] JS Runtime Version v14.4.0
[2021-02-18T13:16:12.138Z] [INFO] Max Heap Size: 33.5 MiB
[2021-02-18T13:16:12.139Z] [INFO] Node Flags: --max-old-space-size=32
[2021-02-18T13:16:12.147Z] [INFO] DeviceId 278000172
[2021-02-18T13:16:12.148Z] [INFO] IP 127.0.0.1
[2021-02-18T13:16:12.149Z] [INFO] CloudSecret cUHLI2UwMQacciV5
[2021-02-18T13:16:12.150Z] [INFO] LocalSecret uvwDFufs1oJxJL2C
[2021-02-18T13:16:12.296Z] [INFO] Dummycloud is spoofing 127.0.0.1:8053 on 127.0.0.1:8053
[2021-02-18T13:16:12.304Z] [INFO] Webserver running on port 80
[2021-02-18T13:16:12.310Z] [INFO] Map Upload Server running on port 8079
[2021-02-18T13:16:12.341Z] [WARN] Error during time sync: Error: queryA ECONNREFUSED pool.ntp.org
    at QueryReqWrap.onresolve [as oncomplete] (dns.js:203:19) {
  errno: undefined,
  code: 'ECONNREFUSED',
  syscall: 'queryA',
  hostname: 'pool.ntp.org'
}
[2021-02-18T13:16:31.338Z] [ERROR] Unhandled WebServer Error Error [ERR_HTTP_HEADERS_SENT]: Cannot set headers after they are sent to the client
    at ServerResponse.setHeader (_http_outgoing.js:533:11)
    at ServerResponse.header (/snapshot/Valetudo/node_modules/express/lib/response.js:771:10)
    at ServerResponse.send (/snapshot/Valetudo/node_modules/express/lib/response.js:170:12)
    at ServerResponse.json (/snapshot/Valetudo/node_modules/express/lib/response.js:267:15)
    at ServerResponse.send (/snapshot/Valetudo/node_modules/express/lib/response.js:158:21)
    at ViomiV7ValetudoRobot.handleHttpDnsRequest (/snapshot/Valetudo/lib/robots/MiioValetudoRobot.js:231:25)
    at /snapshot/Valetudo/lib/webserver/WebServer.js:105:28
    at Layer.handle [as handle_request] (/snapshot/Valetudo/node_modules/express/lib/router/layer.js:95:5)
    at next (/snapshot/Valetudo/node_modules/express/lib/router/route.js:137:13)
    at Route.dispatch (/snapshot/Valetudo/node_modules/express/lib/router/route.js:112:3) {
  code: 'ERR_HTTP_HEADERS_SENT'
}
[2021-02-18T13:16:31.411Z] [INFO] Cloud connected
[2021-02-18T13:16:31.519Z] [INFO] Unknown cloud message received: {"id":2099693185,"method":"props","params":{"ota_state":"idle"}}
[2021-02-18T13:16:36.851Z] [INFO] Unknown cloud message received: {"id":2099693185,"method":"props","params":{"ota_state":"idle"}}
[2021-02-18T13:16:36.936Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-02-18T13:16:39.855Z] [INFO] Unknown cloud message received: {"id":2099693185,"method":"props","params":{"ota_state":"idle"}}
[2021-02-18T13:16:39.861Z] [INFO] Cloud connected
[2021-02-18T13:16:40.854Z] [INFO] << cloud: ignoring response for non-pending request {"id":1,"error":{"code":-9999,"message":"user ack timeout"}}
[2021-02-18T13:16:45.385Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-02-18T13:16:48.850Z] [INFO] << cloud: ignoring response for non-pending request {"id":2,"error":{"code":-9999,"message":"user ack timeout"}}
[2021-02-18T13:16:48.854Z] [INFO] Cloud connected
[2021-02-18T13:16:54.370Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-02-18T13:16:57.850Z] [INFO] << cloud: ignoring response for non-pending request {"id":3,"error":{"code":-9999,"message":"user ack timeout"}}
[2021-02-18T13:16:57.855Z] [INFO] Cloud connected
[2021-02-18T13:17:03.371Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-02-18T13:17:06.851Z] [INFO] << cloud: ignoring response for non-pending request {"id":4,"error":{"code":-9999,"message":"user ack timeout"}}
[2021-02-18T13:17:06.855Z] [INFO] Cloud connected
[2021-02-18T13:17:12.412Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-02-18T06:14:30.002Z] [INFO] Successfully set the robot time via NTP to 2021-02-18T14:14:30.890Z
[2021-02-18T06:14:33.012Z] [INFO] << cloud: ignoring response for non-pending request {"id":5,"error":{"code":-9999,"message":"user ack timeout"}}
[2021-02-18T06:14:33.016Z] [INFO] Cloud connected
[2021-02-18T06:15:07.899Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-02-18T06:15:08.699Z] [INFO] Cloud connected
[2021-02-18T06:15:11.011Z] [INFO] << cloud: ignoring response for non-pending request {"id":6,"error":{"code":-9999,"message":"user ack timeout"}}
[2021-02-18T06:23:43.403Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-02-18T06:23:47.011Z] [INFO] << cloud: ignoring response for non-pending request {"id":7,"error":{"code":-9999,"message":"user ack timeout"}}
[2021-02-18T06:23:47.016Z] [INFO] Cloud connected
[2021-02-18T06:24:40.626Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-02-18T06:24:44.010Z] [INFO] << cloud: ignoring response for non-pending request {"id":8,"error":{"code":-9999,"message":"user ack timeout"}}
[2021-02-18T06:24:44.015Z] [INFO] Cloud connected
[2021-02-18T06:25:53.848Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-02-18T06:25:57.010Z] [INFO] << cloud: ignoring response for non-pending request {"id":9,"error":{"code":-9999,"message":"user ack timeout"}}
[2021-02-18T06:25:57.015Z] [INFO] Cloud connected
[2021-02-18T06:26:00.525Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
```

rumpeltux, 2021-02-21:
Before doing the upgrade, did you follow the instructions to 
[uninstalling](https://github.com/Hypfer/Valetudo/blob/0.5.3/docs/_pages/installation/viomi.md#uninstall-valetudo) and
then [reinstalling](https://valetudo.cloud/pages/installation/viomi.html#robot-setup)?

rumpeltux, 2021-02-21:
For proper debugging, please set your log level to trace and also post your /mnt/data/valetudo/config.json

Christian-Schl, 2021-02-26:
Same Problem on my side.

some additional informations

- Problem only happend when i have active the MQTT Integration
- The most of the Rest Api calls returning an HTTP 500
- SSH is not working (timeout)

Workaround on my side
- Shutdown the MQTT broker 
- wait some time (> 15 minutes)
- access the UI and disable the MQTT Integration

After disable the MQTT Integration everything is working again Rest API, SSH UI ...

Vacuum Model
S5

Valetudo Version
2021.02.0

rumpeltux, 2021-03-06:
Closing issue due to lack of reporter response.

closed 2021-03-06 by rumpeltux

shouya, 2021-07-18:
I had the same problem, it may be caused by `robotManager` not starting. To test it out, run `/etc/init.d/robotManager restart` and see the problem is gone.

If it's indeed the case, run `cd /etc/rc.d; ln -s ../init.d/robotManager S90robotManager` to make sure robotManager is started on boot.