Hypfer/Valetudo #706: loosing data after reboot

lukich48, 2021-02-20:
## Describe the bug
After reboot all settings are loosed. I saved /tmp/valetudo_config.json. Where should I put it to apply the settings?
I used firmware vacuum_4028_valetudo_2021_02_0.pkg 
from there https://vacuumz.info/download/gen1/

## To Reproduce

just run reboot command in ssh terminal

### Screenshots

### Vacuum Model

Xiaomi Mi Robot Vacuum 1

### Valetudo Version

## Additional context

```
[2021-02-20T15:02:19.602Z]` [INFO] Set Logfile to /tmp/valetudo.log
[2021-02-20T15:02:19.610Z] [INFO] Autodetected RoborockV1ValetudoRobot
[2021-02-20T15:02:19.759Z] [INFO] Starting Valetudo 2021.02.0
[2021-02-20T15:02:19.760Z] [INFO] Configuration file: /tmp/valetudo_config.json
[2021-02-20T15:02:19.761Z] [INFO] Logfile: /tmp/valetudo.log
[2021-02-20T15:02:19.762Z] [INFO] Robot: Beijing Roborock Technology Co., Ltd. Xiaomi Mi SDJQR02RR (RoborockV1ValetudoRobot)
[2021-02-20T15:02:19.763Z] [INFO] JS Runtime Version v14.4.0
[2021-02-20T15:02:19.765Z] [INFO] Max Heap Size: 33.5 MiB
[2021-02-20T15:02:19.766Z] [INFO] Node Flags: --max-old-space-size=32
[2021-02-20T15:02:19.775Z] [INFO] DeviceId 82651588
[2021-02-20T15:02:19.776Z] [INFO] IP 127.0.0.1
[2021-02-20T15:02:19.778Z] [INFO] CloudSecret ogp36NPfP2wzhvHe
[2021-02-20T15:02:19.779Z] [INFO] LocalSecret f3Z7yeLTOiDEJSju
[2021-02-20T15:02:19.949Z] [INFO] Dummycloud is spoofing 127.0.0.1:8053 on 127.0.0.1:8053
[2021-02-20T15:02:19.957Z] [INFO] Webserver running on port 80
[2021-02-20T15:02:19.964Z] [INFO] Map Upload Server running on port 8079
[2021-02-20T15:02:19.981Z] [WARN] Error during time sync: Error: queryA ECONNREFUSED pool.ntp.org
    at QueryReqWrap.onresolve [as oncomplete] (dns.js:203:19) {
  errno: undefined,
  code: 'ECONNREFUSED',
  syscall: 'queryA',
  hostname: 'pool.ntp.org'
}
[2021-02-20T15:02:27.873Z] [INFO] Cloud connected
[2021-02-20T15:03:19.004Z] [INFO] Successfully set the robot time via NTP to 2021-02-20T15:03:19.890Z
```

Hypfer, 2021-02-20:
Sounds like the prebuilt images are broken? @zvldz ?

lukich48, 2021-02-20:
@Hypfer How can I restore my settings? I have saved valetudo_config.json but when I replace the default config in /tmp directory nothin happens ((

darootler, 2021-02-21:
I am also experiencing this issue and i am also using the pre built images from vacuumz.info. I think the problem is that "VALETUDO_CONFIG_PATH" env variable is not set.

Regards
Richard

closed 2021-02-22 by lukich48