Hypfer/Valetudo #712: Valetudo crashing after some time on S5E

makuser, 2021-02-23:
## Describe the bug

Valetudo crashes (exits) randomly and needs to be restarted

## To Reproduce

I do not know yet, unfortunately

### Screenshots

### Vacuum Model

Roborock S5 Max

### Valetudo Version

2021.02.0

## Expected behavior

Run until killed

## Additional context

```log
[2021-02-23T20:22:18.862Z] [DEBUG] map_upload_handler { '31248862475': '' }
[2021-02-23T20:22:19.689Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":223}
[2021-02-23T20:22:19.736Z] [DEBUG] <<< cloud* {"stamp":1614111738}
[2021-02-23T20:22:19.737Z] [DEBUG] >>> cloud* {"stamp":1614111738}
[2021-02-23T20:22:19.740Z] [DEBUG] <<< cloud: {"id":223,"result":["map_upload_handler"],"exe_time":2}
[2021-02-23T20:22:19.879Z] [DEBUG] map_upload_handler { '31248910141': '' }
[2021-02-23T20:22:20.879Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":224}
[2021-02-23T20:22:20.937Z] [DEBUG] <<< cloud: {"id":224,"result":["map_upload_handler"],"exe_time":1}
[2021-02-23T20:22:21.072Z] [DEBUG] map_upload_handler { '31248937433': '' }
[2021-02-23T20:22:21.874Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":225}
[2021-02-23T20:22:21.899Z] [DEBUG] <<< cloud: {"id":225,"result":["map_upload_handler"],"exe_time":8}
[2021-02-23T20:22:22.152Z] [DEBUG] map_upload_handler { '31254883058': '' }
[2021-02-23T20:22:23.066Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":226}
[2021-02-23T20:22:23.129Z] [DEBUG] <<< cloud: {"id":226,"result":["map_upload_handler"],"exe_time":1}
[2021-02-23T20:22:23.230Z] [DEBUG] map_upload_handler { '31255094141': '' }
[2021-02-23T20:22:24.099Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":227}
[2021-02-23T20:22:24.269Z] [DEBUG] <<< cloud: {"id":227,"result":["map_upload_handler"],"exe_time":2}
[2021-02-23T20:22:24.282Z] [DEBUG] map_upload_handler { '31255145725': '' }
[2021-02-23T20:22:24.572Z] [DEBUG] <<< cloud* {"stamp":1614111743}
[2021-02-23T20:22:24.573Z] [DEBUG] cloud: Discarding pong
[2021-02-23T20:22:25.269Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":228}
[2021-02-23T20:22:25.446Z] [DEBUG] >>> cloud: {"method":"get_consumable","params":[],"id":229}
[2021-02-23T20:22:27.323Z] [DEBUG] <<< cloud: {"id":228,"result":["map_upload_handler"],"exe_time":287}
[2021-02-23T20:22:27.420Z] [DEBUG] <<< cloud: {"id":229,"result":[{"main_brush_work_time":511833,"side_brush_work_time":511833,"filter_work_time":511833,"filter_element_work_time":122720,"sensor_dirty_time":216271}],"exe_time":727}
[2021-02-23T20:22:28.464Z] [DEBUG] map_upload_handler { '31255170433': '' }
[2021-02-23T20:22:32.383Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":230}
[2021-02-23T20:22:35.403Z] [DEBUG] cloud request 230 get_map_v1 timed out
[2021-02-23T20:22:37.212Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-02-23T20:22:41.155Z] [DEBUG] <<< cloud: {"id":230,"result":["map_upload_handler"],"exe_time":4949}
[2021-02-23T20:22:41.286Z] [INFO] << cloud: ignoring response for non-pending request {"id":230,"result":["map_upload_handler"],"exe_time":4949}
[2021-02-23T20:22:41.318Z] [INFO] Cloud connected
[2021-02-23T20:22:42.216Z] [DEBUG] map_upload_handler { '31260636600': '' }
[2021-02-23T20:22:45.208Z] [DEBUG] <<< cloud* {"stamp":1614111763}
[2021-02-23T20:22:45.326Z] [DEBUG] >>> cloud* {"stamp":1614111763}
[2021-02-23T20:22:45.656Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":231}
[2021-02-23T20:22:45.997Z] [DEBUG] <<< cloud: {"id":231,"result":["map_upload_handler"],"exe_time":137}
[2021-02-23T20:22:46.262Z] [DEBUG] map_upload_handler { '31260709642': '' }
[2021-02-23T20:22:48.309Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":232}
[2021-02-23T20:22:48.783Z] [DEBUG] <<< cloud: {"id":232,"result":["map_upload_handler"],"exe_time":150}
[2021-02-23T20:22:49.235Z] [DEBUG] map_upload_handler { '31235413850': '' }
[2021-02-23T20:22:51.061Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":233}
[2021-02-23T20:22:51.422Z] [DEBUG] <<< cloud: {"id":233,"result":["map_upload_handler"],"exe_time":121}
[2021-02-23T20:22:52.193Z] [DEBUG] map_upload_handler { '31235718016': '' }
[2021-02-23T20:22:54.686Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":234}
[2021-02-23T20:22:56.667Z] [DEBUG] cloud request 234 get_map_v1 timed out
[2021-02-23T20:22:56.892Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-02-23T20:22:57.317Z] [DEBUG] <<< cloud: {"id":234,"result":["map_upload_handler"],"exe_time":1034}
[2021-02-23T20:22:57.317Z] [INFO] << cloud: ignoring response for non-pending request {"id":234,"result":["map_upload_handler"],"exe_time":1034}
[2021-02-23T20:22:57.329Z] [INFO] Cloud connected
[2021-02-23T20:22:59.344Z] [DEBUG] map_upload_handler { '31235776225': '' }
[2021-02-23T20:23:03.921Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":235}
[2021-02-23T20:23:07.449Z] [DEBUG] <<< cloud* {"stamp":1614111783}
[2021-02-23T20:23:07.548Z] [DEBUG] >>> cloud* {"stamp":1614111783}
[2021-02-23T20:23:07.584Z] [DEBUG] cloud request 235 get_map_v1 timed out
[2021-02-23T20:23:07.592Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-02-23T20:23:08.067Z] [DEBUG] <<< cloud: {"id":235,"result":["map_upload_handler"],"exe_time":121}
[2021-02-23T20:23:08.068Z] [INFO] << cloud: ignoring response for non-pending request {"id":235,"result":["map_upload_handler"],"exe_time":121}
[2021-02-23T20:23:08.069Z] [INFO] Cloud connected
[2021-02-23T20:23:09.323Z] [DEBUG] map_upload_handler { '31235817433': '' }
[2021-02-23T20:23:14.664Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":236}
[2021-02-23T20:23:18.669Z] [DEBUG] cloud request 236 get_map_v1 timed out
[2021-02-23T20:23:19.976Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-02-23T20:23:23.936Z] [DEBUG] <<< cloud: {"id":236,"result":["map_upload_handler"],"exe_time":233}
[2021-02-23T20:23:24.088Z] [INFO] << cloud: ignoring response for non-pending request {"id":236,"result":["map_upload_handler"],"exe_time":233}
[2021-02-23T20:23:24.088Z] [INFO] Cloud connected
[2021-02-23T20:23:24.652Z] [DEBUG] <<< cloud* {"stamp":1614111803}
[2021-02-23T20:23:24.653Z] [DEBUG] >>> cloud* {"stamp":1614111803}
[2021-02-23T20:23:25.286Z] [DEBUG] map_upload_handler { '31242062017': '' }
[2021-02-23T20:23:31.150Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":237}
[2021-02-23T20:23:38.321Z] [DEBUG] >>> cloud: {"method":"get_consumable","params":[],"id":238}
[2021-02-23T20:23:49.282Z] [DEBUG] cloud request 237 get_map_v1 timed out
[2021-02-23T20:23:50.731Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-02-23T20:23:52.658Z] [DEBUG] cloud request 238 get_consumable timed out
[2021-02-23T20:23:56.333Z] [ERROR] Failed to poll Attributes MiioTimeoutError: request timed out:{"method":"get_consumable","params":[],"id":238}
    at Timeout.timeout [as _onTimeout] (/snapshot/Valetudo/lib/miio/MiioSocket.js:185:32)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
[2021-02-23T20:24:10.304Z] [DEBUG] <<< cloud* {"stamp":1614111849}
[2021-02-23T20:24:10.487Z] [DEBUG] >>> cloud* {"stamp":1614111849}
[2021-02-23T20:24:10.515Z] [INFO] Cloud connected
[2021-02-23T20:24:11.547Z] [DEBUG] <<< cloud: {"id":237,"result":["retry"],"exe_time":1626}
[2021-02-23T20:24:11.548Z] [INFO] << cloud: ignoring response for non-pending request {"id":237,"result":["retry"],"exe_time":1626}
[2021-02-23T20:24:12.823Z] [DEBUG] map_upload_handler { '31242188808': '' }
[2021-02-23T20:24:16.058Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":239}
[2021-02-23T20:24:18.353Z] [DEBUG] cloud request 239 get_map_v1 timed out
[2021-02-23T20:24:18.501Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-02-23T20:24:20.544Z] [DEBUG] <<< cloud: {"id":239,"result":["map_upload_handler"],"exe_time":108}
[2021-02-23T20:24:20.758Z] [INFO] << cloud: ignoring response for non-pending request {"id":239,"result":["map_upload_handler"],"exe_time":108}
[2021-02-23T20:24:20.758Z] [INFO] Cloud connected
[2021-02-23T20:24:21.056Z] [DEBUG] map_upload_handler { '31242237808': '' }
[2021-02-23T20:24:25.642Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":240}
[2021-02-23T20:24:26.347Z] [DEBUG] <<< cloud: {"id":240,"result":["map_upload_handler"],"exe_time":295}
[2021-02-23T20:24:28.020Z] [DEBUG] map_upload_handler { '31242277350': '' }
[2021-02-23T20:24:29.046Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":241}
```

Hypfer, 2021-02-23:
Not seeing any stacktrace in the log may indicate OOM issues.

I've just added [os.freemem()](https://nodejs.org/api/os.html#os_os_freemem) to the `memoryStatInterval` feature.

You may want to try the latest master and enable the feature by setting an interval (in ms) for `debug.memoryStatInterval` in the config file

creatvty, 2021-03-12:
Hi, I seem to have a similar issue. I'll contribute my logs here:
```[2021-03-12T10:49:32.007Z] [INFO] Successfully set the robot time via NTP to 2021-03-12T10:49:32.890Z
[2021-03-12T16:33:03.622Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-03-12T16:33:20.896Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 10, method: 'get_consumable', args: [] }
[2021-03-12T16:33:21.197Z] [INFO] Cloud connected
[2021-03-12T16:33:43.744Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-03-12T16:34:01.368Z] [INFO] Cloud connected
[2021-03-12T16:35:23.450Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-03-12T16:35:23.460Z] [ERROR] Failed to poll Attributes MiioTimeoutError: request timed out:{"method":"get_consumable","params":[],"id":7156}
    at Timeout.timeout [as _onTimeout] (/snapshot/Valetudo/lib/miio/MiioSocket.js:185:32)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
[2021-03-12T16:35:41.874Z] [INFO] Cloud connected
[2021-03-12T16:39:24.411Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-03-12T16:39:25.747Z] [ERROR] Failed to poll Attributes MiioTimeoutError: request timed out:{"method":"get_consumable","params":[],"id":7384}
    at Timeout.timeout [as _onTimeout] (/snapshot/Valetudo/lib/miio/MiioSocket.js:185:32)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
[2021-03-12T16:39:32.659Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 10, method: 'get_status', args: {} }
[2021-03-12T16:39:42.543Z] [INFO] Cloud connected
[2021-03-12T16:42:29.288Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-03-12T16:42:33.125Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 10, method: 'get_consumable', args: [] }
[2021-03-12T16:42:38.073Z] [INFO] << cloud: ignoring response for non-pending request {"id":7474,"result":["map_upload_handler"]}
[2021-03-12T16:42:38.197Z] [INFO] Cloud connected
[2021-03-12T16:43:01.959Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-03-12T16:43:03.779Z] [INFO] << cloud: ignoring response for non-pending request {"id":7479,"result":[{"msg_ver":3,"msg_seq":284,"state":5,"battery":78,"clean_time":960,"clean_area":22650000,"error_code":0,"map_present":1,"in_cleaning":1,"in_returning":0,"in_fresh_state":0,"lab_status":1,"water_box_status":0,"fan_power":102,"dnd_enabled":0,"map_status":3,"lock_status":0}]}
[2021-03-12T16:43:03.996Z] [INFO] Cloud connected
[2021-03-12T16:44:24.904Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-03-12T16:44:50.964Z] [INFO] Cloud connected
[2021-03-12T16:47:06.056Z] [INFO] Loading configuration file: /mnt/data/valetudo/valetudo_config.json
[2021-03-12T16:47:06.087Z] [INFO] Set Logfile to /tmp/valetudo.log
[2021-03-12T16:47:06.095Z] [INFO] Autodetected RoborockS5ValetudoRobot
[2021-03-12T16:47:06.266Z] [INFO] Starting Valetudo 2021.02.0
[2021-03-12T16:47:06.267Z] [INFO] Configuration file: /mnt/data/valetudo/valetudo_config.json
[2021-03-12T16:47:06.268Z] [INFO] Logfile: /tmp/valetudo.log
[2021-03-12T16:47:06.272Z] [INFO] Robot: Beijing Roborock Technology Co., Ltd. S5 (RoborockS5ValetudoRobot)
[2021-03-12T16:47:06.273Z] [INFO] JS Runtime Version v14.4.0
[2021-03-12T16:47:06.275Z] [INFO] Max Heap Size: 33.5 MiB
[2021-03-12T16:47:06.276Z] [INFO] Node Flags: --max-old-space-size=32
[2021-03-12T16:47:06.287Z] [INFO] DeviceId xxxxxx
[2021-03-12T16:47:06.288Z] [INFO] IP 127.0.0.1
[2021-03-12T16:47:06.289Z] [INFO] CloudSecret xxxxxxxxxxxx
[2021-03-12T16:47:06.290Z] [INFO] LocalSecret xxxxxxxxxxxxxxx
[2021-03-12T16:47:06.292Z] [INFO] Firmware Version: 3.5.8_002020
[2021-03-12T16:47:06.961Z] [INFO] Dummycloud is spoofing 127.0.0.1:8053 on 127.0.0.1:8053
[2021-03-12T16:47:06.970Z] [INFO] Webserver running on port 80
[2021-03-12T16:47:06.991Z] [INFO] Map Upload Server running on port 8079
[2021-03-12T16:46:58.005Z] [INFO] Successfully set the robot time via NTP to 2021-03-12T16:46:58.890Z
[2021-03-12T16:46:58.368Z] [INFO] Connected successfully to mqtt server
[2021-03-12T16:47:02.497Z] [INFO] Cloud connected
```
My model is a S5. It happened while I was running a cleanup. if that helps.

makuser, 2021-03-16:
> Not seeing any stacktrace in the log may indicate OOM issues.
>
> I've just added [os.freemem()](https://nodejs.org/api/os.html#os_os_freemem) to the `memoryStatInterval` feature.
>
> You may want to try [...]

Unfortunately it has not happened ever since... Anyway, I enabled the function and I will be getting back once I find something

sfspeiser, 2021-04-06:
The same thing happens to me. The S5E crashes after a while.
I am not  running the lowmem version yet, but seeing that the free mem went down to 2MB it might be a good idea. After some while, the ssh shell crashed out as well.

However, i could not find any oom_killer logs in the messages or syslog files. The ``vm.min_free_kbytes`` is set to 2038, which would align with the ~2MB of mem left.

Here is the valetudo log. I will switch to the lowmem version of valetudo and see if that helps.
```
BusyBox v1.24.1 (2019-11-15 15:22:50 CST) built-in shell (ash)

[root@rockrobo ~]# tail -f /tmp/valetudo.log 
[2021-04-06T07:42:17.638Z] [DEBUG] <<< cloud: {"id":773,"result":["retry"]}
[2021-04-06T07:42:17.816Z] [DEBUG] map_upload_handler { '33943240311': '' }
[2021-04-06T07:42:22.253Z] [INFO] Memory Stats {
  totalHeapSize: '16.586 MiB',
  mallocedMemory: '0.008 MiB',
  peakMallocedMemory: '1.383 MiB',
  freeSystemMemory: '78.020 MiB'
}
[2021-04-06T07:42:24.947Z] [DEBUG] <<< cloud* {"stamp":1617694944}
[2021-04-06T07:42:24.947Z] [DEBUG] >>> cloud* {"stamp":1617694944}
[2021-04-06T07:42:27.256Z] [INFO] Memory Stats {
  totalHeapSize: '16.586 MiB',
  mallocedMemory: '0.008 MiB',
  peakMallocedMemory: '1.383 MiB',
  freeSystemMemory: '78.023 MiB'
}
[2021-04-06T07:42:32.260Z] [INFO] Memory Stats {
  totalHeapSize: '16.586 MiB',
  mallocedMemory: '0.008 MiB',
  peakMallocedMemory: '1.383 MiB',
  freeSystemMemory: '78.000 MiB'
}
[2021-04-06T07:42:37.264Z] [INFO] Memory Stats {
  totalHeapSize: '16.586 MiB',
  mallocedMemory: '0.008 MiB',
  peakMallocedMemory: '1.383 MiB',
  freeSystemMemory: '78.020 MiB'
}
[2021-04-06T07:42:42.267Z] [INFO] Memory Stats {
  totalHeapSize: '16.586 MiB',
  mallocedMemory: '0.008 MiB',
  peakMallocedMemory: '1.383 MiB',
  freeSystemMemory: '78.012 MiB'
}
[2021-04-06T07:42:44.947Z] [DEBUG] <<< cloud* {"stamp":1617694964}
[2021-04-06T07:42:44.947Z] [DEBUG] >>> cloud* {"stamp":1617694964}
[2021-04-06T07:42:47.267Z] [INFO] Memory Stats {
  totalHeapSize: '16.352 MiB',
  mallocedMemory: '0.008 MiB',
  peakMallocedMemory: '1.383 MiB',
  freeSystemMemory: '78.074 MiB'
}
[2021-04-06T07:42:48.927Z] [DEBUG] >>> cloud: {"method":"get_status","params":{},"id":774}
[2021-04-06T07:42:48.934Z] [DEBUG] <<< cloud: {"id":774,"result":[{"msg_ver":2,"msg_seq":792,"state":8,"battery":100,"clean_time":10,"clean_area":0,"error_code":0,"map_present":1,"in_cleaning":0,"in_returning":0,"in_fresh_state":1,"lab_status":1,"water_box_status":0,"fan_power":102,"dnd_enabled":0,"map_status":3,"is_locating":0,"lock_status":0,"water_box_mode":202,"water_box_carriage_status":0,"mop_forbidden_enable":0}],"exe_time":2}
[2021-04-06T07:42:51.686Z] [DEBUG] >>> cloud: {"method":"get_status","params":{},"id":775}
[2021-04-06T07:42:51.693Z] [DEBUG] <<< cloud: {"id":775,"result":[{"msg_ver":2,"msg_seq":793,"state":8,"battery":100,"clean_time":10,"clean_area":0,"error_code":0,"map_present":1,"in_cleaning":0,"in_returning":0,"in_fresh_state":1,"lab_status":1,"water_box_status":0,"fan_power":102,"dnd_enabled":0,"map_status":3,"is_locating":0,"lock_status":0,"water_box_mode":202,"water_box_carriage_status":0,"mop_forbidden_enable":0}],"exe_time":2}
[2021-04-06T07:42:52.267Z] [INFO] Memory Stats {
  totalHeapSize: '16.586 MiB',
  mallocedMemory: '0.008 MiB',
  peakMallocedMemory: '1.383 MiB',
  freeSystemMemory: '77.727 MiB'
}
[2021-04-06T07:42:57.271Z] [INFO] Memory Stats {
  totalHeapSize: '16.586 MiB',
  mallocedMemory: '0.008 MiB',
  peakMallocedMemory: '1.383 MiB',
  freeSystemMemory: '77.422 MiB'
}
[2021-04-06T07:43:02.244Z] [DEBUG] >>> cloud: {"method":"get_consumable","params":[],"id":776}
[2021-04-06T07:43:02.250Z] [DEBUG] <<< cloud: {"id":776,"result":[{"main_brush_work_time":38764,"side_brush_work_time":38764,"filter_work_time":38764,"filter_element_work_time":2203,"sensor_dirty_time":38676}],"exe_time":1}
[... 1676 lines cut ...]
[2021-04-06T07:56:12.629Z] [INFO] Memory Stats {
  totalHeapSize: '15.590 MiB',
  mallocedMemory: '0.008 MiB',
  peakMallocedMemory: '2.391 MiB',
  freeSystemMemory: '2.281 MiB'
}
[2021-04-06T07:56:19.332Z] [DEBUG] <<< cloud: {"id":1012,"error":{"code":-9999,"message":"user ack timeout"},"exe_time":8071}
[2021-04-06T07:56:20.198Z] [INFO] << cloud: ignoring response for non-pending request {"id":1012,"error":{"code":-9999,"message":"user ack timeout"},"exe_time":8071}
[2021-04-06T07:56:20.210Z] [INFO] Cloud connected
[2021-04-06T07:56:21.474Z] [DEBUG] <<< cloud* {"stamp":1617695741}
[2021-04-06T07:56:21.507Z] [DEBUG] cloud: Discarding pong
[2021-04-06T07:56:21.509Z] [DEBUG] <<< cloud* {"stamp":1617695753}
[2021-04-06T07:56:21.509Z] [DEBUG] >>> cloud* {"stamp":1617695753}
[2021-04-06T07:56:21.777Z] [DEBUG] <<< cloud* {"stamp":1617695775}
[2021-04-06T07:56:21.778Z] [DEBUG] >>> cloud* {"stamp":1617695775}
[2021-04-06T07:56:21.829Z] [DEBUG] local request 70045 get_consumable timed out
[2021-04-06T07:56:22.520Z] [DEBUG] >>> local: {"method":"get_consumable","params":[],"id":71046}
[2021-04-06T07:56:22.891Z] [INFO] Memory Stats {
  totalHeapSize: '15.352 MiB',
  mallocedMemory: '0.008 MiB',
  peakMallocedMemory: '2.391 MiB',
  freeSystemMemory: '2.105 MiB'
}

```

Hypfer, 2021-04-06:
The lowmem 2021.03.0 release or the latest master lowmem build? The latter should work much better

sfspeiser, 2021-04-06:
> The lowmem 2021.03.0 release or the latest master lowmem build? The latter should work much better

Neat,  i now tried the current master build. Works like a charm. Free memory hovers around 50MB now.

Hypfer, 2021-04-06:
@sfspeiser I'd assume the debug log now shows quite a few `Garbage collection forced` loglines?

sfspeiser, 2021-04-06:
@Hypfer yes, every few seconds during a large room cleaning.

creatvty, 2021-04-09:
Hey, I can also confirm that the lowmem build is much more stable. Thanks for that :+1:

sfspeiser, 2021-04-11:
Ok, aperantly the issue is not solved for me yet. After a few room cleanings the robot crashed again.
It looks like there is some memory leak, as the garbage collection can not re claim memory.
This is the last memory statement before the entire robot froze:
```
[2021-04-11T10:36:12.955Z] [INFO] Memory Stats {
  rss: '147.199 MiB',
  heapTotal: '8.523 MiB',
  heapUsed: '7.131 MiB',
  external: '111.550 MiB',
  arrayBuffers: '109.997 MiB',
  freeSystemMemory: '2.184 MiB'
}
[2021-04-11T10:36:13.518Z] [DEBUG] Garbage collection forced. rss: 154800128
```

The memory usage increased with each room cleaning an did not go back down after a cleaning was finished. After 3 cleanings the robot ran out of memory.
The robot is running the low mem build using the 5c8495ce2eff247a667e87b2649c98ee13bddf28 commit.

I am now thinking of running valetudo on my local server using x86... that would at least circumvent the issue of the robot crashing.

Hypfer, 2021-04-11:
It seems like we're pretty much pushing the limits of nodejs here. Documentation is getting quite sparse

global.gc() seems to accept additional parameters:

https://github.com/v8/v8/blob/a69bc99136d4397f7b9c23bff513bd04fcf29aa1/src/extensions/gc-extension.cc#L130-L162

https://github.com/v8/v8/commit/aa7c6e22f963ffcd49898521890cde8d78e4fcc5

This commit message contains somewhat more information on that. Apparently, it defaults to a major gc (whatever that means) so overriding it so that it does a minor one doesn't make much sense I suppose.
Interestingly, when googling, one of the few results that exist suggested the exact opposite.

(Un)fortunately, I'm not able to reproduce your issue on my robots.

Anyways, the only idea left would be to resort to doing manual memory management 👀
One place where buffers are constantly created and disposed again is this one:
https://github.com/Hypfer/Valetudo/blob/5c8495ce2eff247a667e87b2649c98ee13bddf28/lib/robots/MiioValetudoRobot.js#L70-L99

While calling Buffer.concat on all chunks is the first google result for doing file upload handling with expressjs, in our situation, it might make sense to work with a single buffer and write data to the offsets. Expressjs would probably still internally create more buffers, but we could still reduce that amount by a bit.
I'm of course just guessing here. It might not even make any difference at all

Hypfer, 2021-04-11:
271bac7ae0d6d7a6b3fb3d014847c2fcf56556cb

There we go.
Another thing to look into would be mqtt map publishing

@sfspeiser Do you have mqtt map data stuff enabled? Also, was the webinterface open during those three cleanups?

sfspeiser, 2021-04-11:
> [271bac7](https://github.com/Hypfer/Valetudo/commit/271bac7ae0d6d7a6b3fb3d014847c2fcf56556cb)
> 
> There we go.
> Another thing to look into would be mqtt map publishing
> 
> @sfspeiser Do you have mqtt map data stuff enabled? Also, was the webinterface open during those three cleanups?
Thanks for your commit, sadly it did not change anything.

The web interface was open during the runs (on my pc and my phone aswell).  But even when closing it, the memory is not reclaimed. I now disabled MQTT all together and still have the same issue. 
I have a feeling it might be an issue with the map. Tomorrow I will reset it and let it map my flat again. I can't  imagine why else this error is not reproducible...

Hypfer, 2021-06-02:
98b2757aec0af94392038577eead7749ef659e17 and 0d7d8364b98e950d186d961062b81fbfff5e920a may also improve lowmem performance especially when dealing with large maps

I however just guessed that since I can't reproduce the issue locally so further testing would be much appreciated

Hypfer, 2021-07-03:
Closing due to inactivity. Please reopen if this is still an issue

closed 2021-07-03 by Hypfer