Hypfer/Valetudo #715: Zone cleanup stops after several seconds due to "Some of reached areas are not cleaned"

tlpbu, 2021-02-27:
## Describe the bug

Starting zone cleanup from Home Assistant 
```
                action: call-service
                service: vacuum.send_command
                service_data:
                  entity_id: vacuum.robot
                  command: 'zoned_cleanup'
                  params:
                    'zone_ids': ['85318723-5b9e-4d45-9d33-09828dc3ee9d']
```
Vacuum starts cleaning with audial notification "Starting zone cleanup"
but then after several seconds return to dock saying "Some of reached areas are not cleaned".

## To Reproduce

Not really clear how to reproduce - zone cleanup works fine until it doesn't
### Screenshots

### Vacuum Model

Xiaomi Roborock Gen 1

### Valetudo Version

2021.2.0

## Additional context

Could be it is not even Valetudo bug as such, because here https://github.com/homebridge-xiaomi-roborock-vacuum/homebridge-xiaomi-roborock-vacuum/issues/295 it is happening in another context.

Hypfer, 2021-02-27:
> Could be it is not even Valetudo bug

Correct.

closed 2021-02-27 by Hypfer

tlpbu, 2021-02-27:
Could you guide me to to what is root problem and how to resolve problem?
Becuse I am not using homebridge.

Hypfer, 2021-02-27:
No.
Only roborock knows what the root problem is.