Hypfer/Valetudo #717: MQTT Map data is updated after cleaning is finished

conrad784, 2021-02-27:
## Describe the bug
I recently deployed monitoring to my robot (why not ;)) and noticed that my WiFi usage is pretty high (5Mb/s) when it should not be doing anything. The usage is almost zero after the nightly reboot, but after I started a cleaning cycle, the map will continuously be updated via MQTT, even when the robot is docked again.
This is not a serious issue but minor inconvenience to know that valuable data is consumed.

## To Reproduce

1. Reboot the robot
2. Start MQTT Explorer, see history of map_data --> data will be updated every ~1min.
3. Start a cleanup
4. Return to dock
5. map_data will be updated every 10s
6. after (nightly) reboot, data will update again every minute

### Screenshots

### Vacuum Model

Roborock v1

### Valetudo Version

2021.02.0

## Expected behavior

If the robot is docked, updates of mqtt data should be less frequent (e.g. minutely)

## Additional context

Btw awesome software, I am very glad I could remove the cloud from my device :)

Hypfer, 2021-02-27:
There is no interval in the MQTT client. Just an event listener
https://github.com/Hypfer/Valetudo/blob/f7601370353659ce8a35e6c2a942c48d4f614226/lib/mqtt/MqttClient.js#L65-L70

with the map repoll rate being determined by the robots state and reduced if it's idle
https://github.com/Hypfer/Valetudo/blob/2d02570f0c85c078941e67020e5e02a89fa62894/lib/robots/roborock/RoborockValetudoRobot.js#L270-L287

I've tested this and it works correctly on my machine.

Is there anything that might be interesting in your logs?

conrad784, 2021-02-27:
I just started a small zoned cleanup and the result is as you described, it goes back to minutely reports.

I usually are doing full cleanups, so I will extract the debug logs next time I am doing a full cleanup. Thank you for your fast response.

conrad784, 2021-02-28:
I am not quite sure how to debug this further.
Currently I am trying to understand the logs. Right not every minute there is a single call for `get_map_v1` and a single call to the "cloud" to `/api/miio/map_upload_handler`. In the logs above you see, that there is a call to upload every minute, but in between there are multiple `get_map_v1` calls.
If I understand correctly this function can only be executed if the last poll is `now.getTime() - 600 > this.lastMapPoll.getTime()` and what I am seeing in the logs that it is called within 3 seconds again, can only happen if `StatusStateAttribute.isActiveState`. To reach this state we will get back true for some value {RETURNING, CLEANING, MANUAL_CONTROL, MOVING} and set the repollSeconds +=2 .

Am I missing a way this could happen? Let me see if I find time this week to log those values.

conrad784, 2021-03-01:
I just ran a full cleanup and the output is still showing this issue
Within MQTT-Explorer I can see that the data is refreshed everytime "map_upload_handler" is called. Currently this is every 1-4s. There is no "isActiveState" but the Repoll Seconds of 60 is not respected.

@Hypfer  what data can I provide to debug this further?
```
[2021-03-01T13:29:21.818Z] [INFO] started sendCloud function
[2021-03-01T13:29:21.819Z] [INFO] sendCloud array: ["map_upload_handler"]
[2021-03-01T13:29:21.820Z] [INFO] Repoll Seconds 60
[2021-03-01T13:29:21.839Z] [DEBUG] map_upload_handler { '29794537682': '' }
[2021-03-01T13:29:24.604Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":4322}
[2021-03-01T13:29:24.609Z] [DEBUG] <<< cloud: {"id":4322,"result":["map_upload_handler"]}
[2021-03-01T13:29:24.610Z] [INFO] started sendCloud function
[2021-03-01T13:29:24.611Z] [INFO] sendCloud array: ["map_upload_handler"]
[2021-03-01T13:29:24.612Z] [INFO] Repoll Seconds 60
[2021-03-01T13:29:24.629Z] [DEBUG] map_upload_handler { '29794833515': '' }
[2021-03-01T13:29:25.426Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":4323}
[2021-03-01T13:29:25.432Z] [DEBUG] <<< cloud: {"id":4323,"result":["map_upload_handler"]}
[2021-03-01T13:29:25.434Z] [INFO] started sendCloud function
[2021-03-01T13:29:25.435Z] [INFO] sendCloud array: ["map_upload_handler"]
[2021-03-01T13:29:25.436Z] [INFO] Repoll Seconds 60
[2021-03-01T13:29:25.454Z] [DEBUG] map_upload_handler { '29794920056': '' }
[2021-03-01T13:29:27.596Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":4324}
[2021-03-01T13:29:27.601Z] [DEBUG] <<< cloud: {"id":4324,"result":["map_upload_handler"]}
[2021-03-01T13:29:27.602Z] [INFO] started sendCloud function
[2021-03-01T13:29:27.604Z] [INFO] sendCloud array: ["map_upload_handler"]
[2021-03-01T13:29:27.605Z] [INFO] Repoll Seconds 60
[2021-03-01T13:29:27.621Z] [DEBUG] map_upload_handler { '29794987140': '' }
[2021-03-01T13:29:28.140Z] [DEBUG] >>> cloud: {"method":"get_consumable","params":[],"id":4325}
[2021-03-01T13:29:28.145Z] [DEBUG] <<< cloud: {"id":4325,"result":[{"main_brush_work_time":698040,"side_brush_work_time":698040,"filter_work_time":153033,"sensor_dirty_time":22223}]}
[2021-03-01T13:29:28.381Z] [DEBUG] <<< cloud: {"method":"event.status","id":204449515,"params":[{"msg_ver":8,"msg_seq":60,"state":8,"battery":75,"clean_time":2890,"clean_area":41570000,"error_code":0,"map_present":1,"in_cleaning":0,"fan_power":75,"dnd_enabled":0}]}
[2021-03-01T13:29:28.385Z] [DEBUG] >>> cloud: {"id":204449515,"result":"ok"}
[2021-03-01T13:29:31.778Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":4326}
[2021-03-01T13:29:31.790Z] [DEBUG] <<< cloud: {"id":4326,"result":["map_upload_handler"]}
[2021-03-01T13:29:31.791Z] [INFO] started sendCloud function
[2021-03-01T13:29:31.792Z] [INFO] sendCloud array: ["map_upload_handler"]
[2021-03-01T13:29:31.793Z] [INFO] Repoll Seconds 60
[2021-03-01T13:29:31.807Z] [DEBUG] map_upload_handler { '29800937682': '' }
[2021-03-01T13:29:32.857Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":4327}
[2021-03-01T13:29:32.862Z] [DEBUG] <<< cloud: {"id":4327,"result":["map_upload_handler"]}
[2021-03-01T13:29:32.864Z] [INFO] started sendCloud function
[2021-03-01T13:29:32.865Z] [INFO] sendCloud array: ["map_upload_handler"]
[2021-03-01T13:29:32.866Z] [INFO] Repoll Seconds 60
[2021-03-01T13:29:32.883Z] [DEBUG] map_upload_handler { '29801091432': '' }
[2021-03-01T13:29:33.642Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":4328}
[2021-03-01T13:29:33.646Z] [DEBUG] <<< cloud: {"id":4328,"result":["map_upload_handler"]}
[2021-03-01T13:29:33.651Z] [INFO] started sendCloud function
[2021-03-01T13:29:33.653Z] [INFO] sendCloud array: ["map_upload_handler"]
[2021-03-01T13:29:33.653Z] [INFO] Repoll Seconds 60
[2021-03-01T13:29:33.658Z] [DEBUG] map_upload_handler { '29801165015': '' }
[2021-03-01T13:29:37.228Z] [DEBUG] <<< cloud* {"stamp":1614605377}
[2021-03-01T13:29:37.230Z] [DEBUG] >>> cloud* {"stamp":1614605377}
[2021-03-01T13:29:37.655Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":4329}
[2021-03-01T13:29:37.660Z] [DEBUG] <<< cloud: {"id":4329,"result":["map_upload_handler"]}
[2021-03-01T13:29:37.662Z] [INFO] started sendCloud function
[2021-03-01T13:29:37.663Z] [INFO] sendCloud array: ["map_upload_handler"]
[2021-03-01T13:29:37.664Z] [INFO] Repoll Seconds 60
[2021-03-01T13:29:37.684Z] [DEBUG] map_upload_handler { '29801229640': '' }
[2021-03-01T13:29:39.197Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":4330}
[2021-03-01T13:29:39.202Z] [DEBUG] <<< cloud: {"id":4330,"result":["map_upload_handler"]}
[2021-03-01T13:29:39.203Z] [INFO] started sendCloud function
[2021-03-01T13:29:39.204Z] [INFO] sendCloud array: ["map_upload_handler"]
[2021-03-01T13:29:39.205Z] [INFO] Repoll Seconds 60
[2021-03-01T13:29:39.224Z] [DEBUG] map_upload_handler { '29806694306': '' }
[2021-03-01T13:29:39.488Z] [DEBUG] <<< cloud: {"id":64167289,"method":"_sync.gen_presigned_url","params":{}}
[2021-03-01T13:29:39.491Z] [DEBUG] >>> cloud: {"id":64167289,"result":["http://127.0.0.1:8079/api/miio/map_upload_handler?38170442266826","http://127.0.0.1:8079/api/miio/map_upload_handler?38170442434368","http://127.0.0.1:8079/api/miio/map_upload_handler?38170442492826","http://127.0.0.1:8079/api/miio/map_upload_handler?38170442529409"]}
[2021-03-01T13:29:44.482Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":4331}
[2021-03-01T13:29:44.487Z] [DEBUG] <<< cloud: {"id":4331,"result":["map_upload_handler"]}
[2021-03-01T13:29:44.489Z] [INFO] started sendCloud function
[... cut ...]
[2021-03-01T13:31:08.617Z] [DEBUG] map_upload_handler { '29818817681': '' }
[2021-03-01T13:31:10.111Z] [DEBUG] >>> cloud: {"method":"get_map_v1","id":4363}
[2021-03-01T13:31:10.115Z] [DEBUG] <<< cloud: {"id":4363,"result":["map_upload_handler"]}
[2021-03-01T13:31:10.117Z] [INFO] started sendCloud function
[2021-03-01T13:31:10.118Z] [INFO] sendCloud array: ["map_upload_handler"]
[2021-03-01T13:31:10.119Z] [INFO] Repoll Seconds 60
[2021-03-01T13:31:10.134Z] [DEBUG] map_upload_handler { '29794537682': '' }
```

Hypfer, 2021-03-01:
You can get a stacktrace with `new Error().stack`. That could be helpful to figure out where these .pollMap calls come from

Flunkyball, 2021-04-07:
I experience the same issue. 
I did a complete factory reset and flashed a freshly built image based on v11_002034 with valetudo 2021.03.0 included using the following parameters 

`./builder_vacuum.sh     --run-custom-script=ALL \
                        --timezone=Europe/Berlin \
                        --public-key=some_key.pub \
                        --enable-greeting \
                        --disable-logs \
                        --replace-adbd \
                        --valetudo-path=../valetudo \
                        --replace-miio \
                        --enable-dns-catcher \
                        --ntpserver=172.17.5.72 \
                        --enable-addon \
                        --enable-ramdisk \
                        -f ../firmware/v11_002034.fullos.pkg`

If MQTT is activated I can see the frequent map updates while docked as well producing the log information @conrad784.
Moreover due to the "huge" network traffic the webinterface as well as the ssh connection becomes almost unresponsive. When hitting a key in the shell it takes 2-3sec for it to appear. 
After disabling MQTT everything runs directly smooth again.

Hypfer, 2021-04-25:
This might be fixed by the MQTT rewrite in 2021.04.0

Please check and open a new issue if this is still an issue

closed 2021-04-25 by Hypfer