Hypfer/Valetudo #721: The Map disappears straight after opening

tempico, 2021-02-27:
The Map disappears straight after opening Map tab on Valetudo 2021.02.0

## To Reproduce
1. Open Valetudo web interface via Safari
2. Switch to Map tab
3. Map loads for less than a second and then disappears.

### Screenshots
https://user-images.githubusercontent.com/6486967/109402962-d7403f80-7962-11eb-93e1-9fe592a5562b.MP4

### Vacuum Model
Roborock S5

### Valetudo Version
2021.02.0

## Expected behavior
Map does not disappear, allowing zoom, zones configuration, etc.

## Additional context
I did tried Safari's cache cleanup, opening via in-app browser — but the issue persists. iOS 14.4.

jonasius, 2021-02-28:
I'm experiencing the same error. 
I also tried it with Chrome, it only displays the No-Go zones and last travelled path.
[image]

Hypfer, 2021-03-01:
Yeah apparently Safari doesn't like the canvas size.

I've reduced it for now. That should work but of course doesn't look very good :/

aa64016c29151499cad8f7fef11d8826e7cd58ad

Hypfer, 2021-03-07:
For reference:
https://github.com/jhildenbiddle/canvas-size#test-results

As of now, mobile safari only supports canvas up to 4096x4096px in size

pkishino, 2021-03-11:
same issue pops up in ios chrome

Hypfer, 2021-03-11:
iirc, Chrome on iOS was actually just Safari + Chrome Sync so that's no suprise.

Anyways, you could try builing the latest master and see if that solves your problem. The path rendering has been redone there

jonasius, 2021-03-14:
I tried it with the latest master, map still disappears.

I saw the parameter you changed doesn't exist anymore. Can you tell me what is the new equivalent parameter?

jonasius, 2021-03-15:
With latest commit 3c05e2d9917e222a777a9d86207712d3b798dbec it is working. 👍

Hypfer, 2021-03-15:
Thanks for testing that

closed 2021-03-15 by Hypfer

Hypfer, 2021-03-15:
Since the path is now an svg, this leaves us with only the map layers being limited to the maximum canvas size which is 4096x4096px on iOS Safari.

With a pixel size of 5cm, that translates to a maximum possible map size of 204.8x204.8m
Thats more than four soccer fields in nonsense-units.

I doubt that we'll reach that limit anytime soon. Even if we add support for lawnmowers