Hypfer/Valetudo #722: Vacuum Error when conventional (i.e: not water-capable) dust bin is used.

PaskoMoto, 2021-02-28:
My vacuum only works when the water tank is used. With the conventional dust deposit it fails with an error.

## To Reproduce

Steps to reproduce the behavior:
1. Fit the conventional dust-only box in the vacuum.
2. The tank is recognized as the voice message says.
3. Click 'Start' in the main web page.
4. The message "Water tank is not installed" is displayed in the main web page. Simultaneously the voice message saying "Make sure the water tank is installed" is reproduced.

### Screenshots

[image]

### Vacuum Model
Viomi V8 

### Valetudo Version
2021.2.0

## Expected behavior

The vacuum shoud start cleaning normally with the dust-only deposit.

## Additional context
I have recently upgraded from version 0.61. I have noticed several new features, but this specific use-case seems broken to me.
This is the message logged whenever I click the 'Start' button in the web page.
`2021-02-28T11:03:19.931Z] [INFO] Unknown cloud message received: {"method":"event.fault_happen","params":[521]}`

Additional info:
```
device_config.ini:software_version=X1.3.17.55.27R
device_config.ini:hardware_version=1.0.3
device_config.ini:has_rtc_hardware=0

```
Best regards and thanks for this amazing software. ;)

Depau, 2021-02-28:
I can't reproduce the issue.

The cleaning-type logic is the following:

- if mop is installed
  - if bin has both water tank and dust bin
    - perform vacuum + mop
  - otherwise just mop
- otherwise just vacuum

So I think you're trying to vacuum with the mop installed. Remove it for vacuuming.

Depau, 2021-02-28:
I tweaked a little the behavior by showing an error message when trying to start cleaning with this accessory configuration.

PaskoMoto, 2021-03-01:
Ah. Ok.
Didn't know that. I removed the mop and it started to work again with the dust-only deposit,
Many thanks and best regards.

closed 2021-03-01 by PaskoMoto