Hypfer/Valetudo #731: Second zone cleanup not starting - Viomi V7

XobrekX, 2021-03-03:
## Describe the bug
After I cleaned some zone(s) and the robot is back at the station, I can't trigger another zone cleaning. I have to trigger a full cleaning. Otherwise the robot won't start the zone cleanup and stays docked in the station.

## To Reproduce
Steps to reproduce the behavior:
1. Go to zones
2. Select some random zones and start cleaning
3. Wait till finished 
4. Try to start another zone cleanup
4. See error - robot not starting

### Screenshots

### Vacuum Model
Viomi V7

### Valetudo Version
2021.02.0

## Expected behavior
The secoing zone cleanup should start as well 

## Additional context

XobrekX, 2021-03-03:
@Depau I created a issue, thank you!

Depau, 2021-03-03:
Do you get a specific error message? Does the vacuum speak an error message?

XobrekX, 2021-03-03:
Nope, doesn't seem so. I just tried it again:
Logs telling me, it starts cleaning, but it stays quiet at its dock

[2021-01-31T15:51:26.205Z] [TRACE] >>> cloud: {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"],"id":955}
[2021-01-31T15:51:26.243Z] [TRACE] <<< cloud: {"result":[5,0,2105,100,1,0,0,0,2,12,1,1,0,0],"id":955}
[2021-01-31T15:51:31.267Z] [TRACE] >>> cloud: {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"],"id":956}
[2021-01-31T15:51:31.305Z] [TRACE] <<< cloud: {"result":[5,0,2105,100,1,0,0,0,2,12,1,1,0,0],"id":956}
[2021-01-31T15:51:35.503Z] [TRACE] areas to clean:  [ '0_0_-0.4400_1.6000_-0.4400_-1.3100_1.8800_-1.3100_1.8800_1.6000' ]
[2021-01-31T15:51:35.506Z] [DEBUG] >>> cloud: {"method":"set_zone","params":[1,"0_0_-0.4400_1.6000_-0.4400_-1.3100_1.8800_-1.3100_1.8800_1.6000"],"id":957}
[2021-01-31T15:51:35.599Z] [DEBUG] <<< cloud: {"result":["ok"],"id":957}
[2021-01-31T15:51:35.601Z] [DEBUG] >>> cloud: {"method":"set_mode","params":[3,1,0],"id":958}
[2021-01-31T15:51:35.650Z] [DEBUG] <<< cloud: {"result":["ok"],"id":958}
[2021-01-31T15:51:36.404Z] [TRACE] >>> cloud: {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"],"id":959}
[2021-01-31T15:51:36.441Z] [TRACE] <<< cloud: {"result":[5,0,2105,100,1,0,0,0,2,12,1,1,0,0],"id":959}
[2021-01-31T15:51:38.761Z] [TRACE] >>> cloud: {"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap"],"id":960}
[2021-01-31T15:51:38.799Z] [TRACE] <<< cloud: {"result":[5,0,2105,100,1,0,0,0,2,12,1,1,0,0],"id":960}

Depau, 2021-03-03:
> Go to zones
> Select some random zones and start cleaning

Wait, GoToCapability isn't implemented yet. Are you using a PR branch such as #705?

XobrekX, 2021-03-03:
Ohh sorry, it was missleading, I meant just click on zones ^^ sorry

Depau, 2021-03-03:
> Ohh sorry, it was missleading, I meant just click on zones on the main page ^^ sorry
> I am using the latest master version
> 
> Ahh and the voice is disabled ^^

The voice might give you additional info on why it's not working. Can you please temporarily enable the voice and let me know what it says?

Depau, 2021-03-03:
I can't reproduce it because, 1) it's too late for me once again at the moment and 2) ...
[image]

XobrekX, 2021-03-04:
I enabled voice and it says the same Chinese what so ever when trying to clean zones for the first time

rumpeltux, 2021-03-06:
There’s only one `set_zone` command in the logs and it only contains one area.
Does Valetudo notice that the cleaning finished and the robot is docked in the main page UI?

Depau, 2021-03-07:
I think I was abla to reproduce this by going into the map,and adding a cleaning area. I noticed that the first time it cleans the area correctly, except it doesn't say "cleaning complete, returning to the dock to charge" at the end.

Then if I run it again it won't start and it won't say anything. It will get "unstuck" only if I start a complete cleaning and cancel it.

I'll do some more investigation but it looks like we're not sending the correct command sequence.

XobrekX, 2021-03-07:
> I think I was able to reproduce this by going into the map and adding a cleaning area. I noticed that the first time it cleans the area correctly, except it doesn't say "cleaning complete, returning to the dock to charge" at the end.
> 
> Then if I run it again it won't start and it won't say anything. It will get "unstuck" only if I start a complete cleaning and cancel it.
> 
> I'll do some more investigation but it looks like we're not sending the correct command sequence.

Yeah, that's exactly what happens for me

rumpeltux, 2021-03-07:
```
[DEBUG] >>> cloud: {"method":"set_zone","params":[1,"0_0_0.4200000000000017_-0.39000000000000057_0.4200000000000017_-1.5899999999999999_1.3099999999999987_-1.5899999999999999_1.3099999999999987_-0.39000000000000057"],"id":30}
[DEBUG] >>> cloud: {"method":"set_mode","params":[3,1],"id":31}
[DEBUG] >>> cloud: {"method":"set_zone","params":[1,"0_0_0.4200000000000017_-0.39000000000000057_0.4200000000000017_-1.5899999999999999_1.3099999999999987_-1.5899999999999999_1.3099999999999987_-0.39000000000000057"],"id":70}
[DEBUG] >>> cloud: {"method":"set_mode","params":[3,1],"id":71}
```

I couldn’t repro with the 0.6 version of Viomi: it worked just fine.

Note that the `set_mode` command is only taking two arguments here and omits the empty roomlist specifier.

tomaszpieczykolan, 2021-03-12:
Also affecting me. I think it could be connected to #747.

Depau, 2021-03-17:
For both issues: I'll look it into it this weekend or next week.

> Also affecting me. I think it could be connected to #747.

It's very likely.

PR #782 (2021-03-28): viomi: Fix basic control functionality
## Type of change

Type A:
- [X] Bug fix (non-breaking change which fixes an issue)
- [x] Refactor/Code Cleanup
- [ ] Docs
- [ ] Capability implementation for existing core capability
- [ ] New robot implementation

# Description (Type A)

This PR attempts to fix a bunch of issues previously reported, explicitly or implicitly:

Closes #731 

The new basic control capability implementation now remembers what cleaning command was last sent, since it is required later for pausing, stopping and resuming. We have to remember it ourselves since there doesn't se [... cut]
closed 2021-03-28 by Hypfer