Hypfer/Valetudo #745: NTP server set in config, but i see drops on firewall for other NTP hosts

darootler, 2021-03-08:
## Describe the bug

I am using the following NTP configuration in valetudo_config.json:

```
  "ntpClient": {
    "enabled": true,
    "server": "myinternal.ntp.local",
    "port": 123,
    "interval": 28800000,
    "timeout": 10000
  },
```

The logs are stating that everything is okay:

`[2021-03-08T07:27:48.003Z] [INFO] Successfully set the robot time via NTP to 2021-03-08T07:27:48.890Z`

But i see drops on my firewall for public NTP servers:
[image]

## To Reproduce

Steps to reproduce the behavior:
1.) Configure the NTP server using "valetudo_config.json"
2.) Restart valetudo
3.) Watch the firewall logs for connections on UDP port 123

### Screenshots
See above

### Vacuum Model
Beijing Roborock Technology Co., Ltd. Xiaomi Mi SDJQR02RR (RoborockV1ValetudoRobot)

### Valetudo Version
2021.02.0

## Expected behavior
The configured NTP server should be used

darootler, 2021-03-08:
Setting the NTP in /opt/rockrobo/watchdog/ntpserver.conf "fixes" the problem. Hmm, is this working as expected? I wonder why there are two places to configure the NTP server.

Regards
Richard

Hypfer, 2021-03-08:
There are other things inside the firmware which may do NTP stuff. This isn't a bug in Valetudo

Please don't open Bugs for things that aren't.

closed 2021-03-08 by Hypfer

darootler, 2021-03-08:
If i know that this isn't a bug i wouldn't open an issue, a simple clarification would be enough.

Regards
Richard

Hypfer, 2021-03-08:
[image]

darootler, 2021-03-08:
I am a user of (for sure) a wonderful software. I configured a NTP server and i see that this server isn't used by the device. Is there anything in documentation that states that there are other "NTP things" done by the firmware? Is that a thing a "user" should know?

Sorry, but you can't assume everyone knows exactly what your software or the firmware is doing.

Regards
Richard

Hypfer, 2021-03-08:
Sure, however I can expect that users stick to the rules layed out and use e.g. the telegram group or discussions to clarify that what they're experiencing is actually a bug before opening a bug report

darootler, 2021-03-08:
Thank you for clarification, and by the way: Great job with valetudo!

In case someone is reading this:

I had to change the NTP server in "/opt/rockrobo/watchdog/ntpserver.conf" to have the "whole" device talk to my internal NTP server.

Regards
Richard