Hypfer/Valetudo #747: Viomi Zone Cleaning doesn't show path/time/area or robot moving

markthehipster, 2021-03-11:
## Describe the bug

When Zone Cleaning is set, the vacuum will show up as being on the charger, no path is recorded, and only zones are marked in a static form. The main page will also say paused and not show Area or Time of cleaning. 

## To Reproduce

Steps to reproduce the behavior:
1. Go to 'Home'
2. Click on 'Zones'
3. Select a specified Zone and start cleaning that area
4. See error

### Screenshots
[image]
[image]
^ the values at area and time are from a separate start/stop.

### Vacuum Model

### Valetudo Version

However I've also seen this since Zoned cleaning is built in.
## Expected behavior
Expected to see the vacuum's path during zoned cleaning, and also expected to see that it's 'cleaning' + area and time since it started.

## Additional context

Kaan88, 2021-03-14:
In one of the builds before the capabilities rewrite, map updated correctly when you started zone cleaning from the map page, but it was still broken when it was done via preset zones. 

I can't remember it exactly but I think it was a build from this branch: https://github.com/fhorinek/Valetudo/commits/master

Depau, 2021-03-17:
For both this and #731 which is very likely related: I'll look it into it this weekend or next week.

The cleaning commands are pretty much just wrong, the only thing that works correctly is all-house cleaning. You haven't noticed but zone mopping won't mop, it just vacuums.

exotsk, 2021-03-23:
[image]
All works! Thank to Jan Loewe.

Kaan88, 2021-03-23:
@exotsk it works for segment cleaning but not for zone, so the issue still persists.

exotsk, 2021-03-28:
[image]
 Fixed now. Thanks!

Depau, 2021-03-28:
Yeah this should all work now. If you notice related issues, please tag me so I see it.

closed 2021-03-28 by Hypfer

forast3r, 2021-03-28:
@Depau I installed a recent commit (two days ago) and found that area is reporting now is showing right square meters, but time seems to be reporting only hours:minutes, for example if the viomi has been vacuuming for 7 minutes 10 seconds it shows 00:00:07
The same data in a rockrobo unit shows 00:07:10

markthehipster, 2021-03-28:
@forast3r Yeah, I can confirm the same, however it's not tied to this bugreport, since I saw that happening for ages now.
Could you please open a separate bugreport?