Hypfer/Valetudo #781: Vacuum status is not properly shared via MQTT

tempico, 2021-03-26:
## Describe the bug
When Valetudo 2021.03.0 is connected to a Home Assistant via MQTT, it shares the incorrect current status. It's set to "Idle" even when robot in reality is cleaning. On the screenshot below you may note the "Idle" status, with the "return to dock" at the same time.

## To Reproduce
1. Setup MQTT integration with Home Assistant auto discovery. Make sure robot in not cleaning (docked)
2. Once robot appears in the entities, check content of `vacuum.rockrobo.attributes.state`
3. Start cleaning
4. Note that status is not updating, whilst the map is re-drawing and other attributes are updating as well.

### Screenshots
[image]

### Vacuum Model
Roborock S5

### Valetudo Version
Valetudo 2021.03.0 

## Expected behavior
Status attribute shows current status and being updated in a timely manner, allowing to create automations with vacuum (e.g. `if status == 'cleaning': turn on <air_purifier> at 100%`)

Hypfer, 2021-03-26:
Considering that this is a core functionality used by many users, I'd say you're a bit too fast with deciding that this is a bug in Valetudo and not a quirk of your setup, since otherwise, there would be a lot more reports.

I can't reproduce that.
[image]

tempico, 2021-03-26:
Try to build an automation, with attribute I mentioned above.
In the "history" I also have proper status. But not on the status field.
[image]

Hypfer, 2021-03-26:
But why would you look at `vacuum.rockrobo.attributes.state` when the state is the state of the vacuum.rockrobo entity?

Also, all of those attributes don't exist anymore since 2021.01.1 so whatever you're running it's not Valetudo 2021.3.0

This is not a bug but a support request.

closed 2021-03-26 by Hypfer

tempico, 2021-03-26:
I was not aware that the attributes got removed. Could be I missed this part in the change logs.
I was able to get the proper status from the `vacuum.rockrobo.state`. Thank you a lot for pointing this. 

However take a look into screenshot below, attributes do update, despite you are saying they don't exist anymore, and the state there is set 3 times: _Template State_ is "docked", _state_ is "idle" and _valetudo_state_ is "idle" as well.
[image]

Valetudo version is latest. I built and re-flashed firmware today via `mirobo`.
[image]