Hypfer/Valetudo #788: Map is very slow on Firefox

Depau, 2021-03-27:
## Describe the bug

Map is currently extremely slow on Firefox. I don't recall it being so slow last time I tried a few weeks ago. It displays correctly but navigating it while it's zoomed in performs poorly at something like 1fps.

On Chromium it seems to work fine.

I read something about map changes in the group so I think it might be related, but I have never touched the front-end side and I'd gladly love to continue staying away from front-ends :)

## To Reproduce

1. Open Firefox
2. Go to the map page
3. Turn off the vacuum so you can hear the computer fans taking off :)

### Screenshots

### Vacuum Model

Viomi v8

### Valetudo Version

At least since e3406502d14a043bf506dc7f118db709e5d9f4e8

## Expected behavior

## Additional context

Hypfer, 2021-03-28:
It works fine on my machine.

Windows 10, Firefox 87, 04f2df4c

Maybe something related to GPU acceleration?

Depau, 2021-03-28:
I indeed verified that this only affects Firefox on Linux, even though `about:support` shows everything hw acceleration-related enabled.

It's smooth on Firefox on Windows, macOS and Android, as well as on Safari on mac and GNOME Web on Linux.

Depau, 2021-03-28:
It seems like this is due to the fact that all platforms ship with Off-The-Main-Thread Compositing enabled while it isn't deemed stable enough for Linux. Whether OMTC is enabled or not is not shown in `about:support`, that's why I'd see everything enabled.

Setting `MOZ_USE_OMTC=1` in the environment does the trick.

https://mozillagfx.wordpress.com/2012/10/06/how-to-help-testing-off-main-thread-compositing/

closed 2021-03-28 by Depau

lorenzhs, 2021-05-05:
For me it's not just slow, it crashes Firefox when I move the map too much. It gets killed by the OOM killer (I have 16 GB of RAM and nothing else running, after the crash 1.3/16 GB were in use, it's not an issue of not having enough memory). Has anyone else experienced this?

I'm using Firefox 88 on Debian unstable, everything up-to-date, Intel graphics, no special flags.