Hypfer/Valetudo #789: Fix Viomi "Unknown water grade not work"

exotsk, 2021-03-28:
Changing "water grade" option make a change on "power mode". WG option still medium after that.

## To Reproduce

1. Go to 'Main menu'
2. Click on 'Water Grade'
3. Change to to any
4. See error

### Screenshots
[image]

### Vacuum Model
Viomi STY**T**J but in Valetudo in info as Viomi V7 

### Valetudo Version
2021.1.3 Dev 28/03/2021

## Expected behavior
Changing water grade and working mop mode, but mop don`t work at all with istalled water tank.

Hypfer, 2021-03-28:
What about https://github.com/Hypfer/Valetudo/pull/784?

exotsk, 2021-03-28:
i check it after fix was added. 
[image]
at last check with mop container, all the same. Changing power, no mop funcion.

Depau, 2021-03-28:
Fixed in #791

closed 2021-03-28 by Hypfer