Hypfer/Valetudo #790: Mapping problems on viomi.v7 (A.k.a. Mop P)

nekromant, 2021-03-28:
A lot of errors show up in the log when cleaning, although web ui displays map with no problem. 
```
Sun Mar 28 18:50:42 2021 daemon.err valetudo[1104]: [2021-03-28T10:50:42.275Z] [WARN] Unable to parse Pose RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 662172. Received 662173
Sun Mar 28 18:50:42 2021 daemon.err valetudo[1104]:     at boundsError (internal/buffer.js:83:9)
Sun Mar 28 18:50:42 2021 daemon.err valetudo[1104]:     at Buffer.readInt16LE (internal/buffer.js:411:5)
Sun Mar 28 18:50:42 2021 daemon.err valetudo[1104]:     at ViomiMapParser.parsePose (/snapshot/Valetudo/lib/ViomiMapParser.js:375:58)
Sun Mar 28 18:50:42 2021 daemon.err valetudo[1104]:     at ViomiMapParser.parse (/snapshot/Valetudo/lib/ViomiMapParser.js:191:22)
Sun Mar 28 18:50:42 2021 daemon.err valetudo[1104]:     at ViomiV7ValetudoRobot.parseMap (/snapshot/Valetudo/lib/robots/viomi/ViomiValetudoRobot.js:414:50)
Sun Mar 28 18:50:42 2021 daemon.err valetudo[1104]:     at /snapshot/Valetudo/lib/robots/MiioValetudoRobot.js:85:48
Sun Mar 28 18:50:42 2021 daemon.err valetudo[1104]:     at runMicrotasks (<anonymous>)
Sun Mar 28 18:50:42 2021 daemon.err valetudo[1104]:     at processTicksAndRejections (internal/process/task_queues.js:93:5) {
Sun Mar 28 18:50:42 2021 daemon.err valetudo[1104]:   code: 'ERR_OUT_OF_RANGE'
Sun Mar 28 18:50:42 2021 daemon.err valetudo[1104]: }
```

Later, this seems to crash icantbelieveitsnotvaletudo with the following error: 

```
Loading configuration file: /srv/ICantBelieveItsNotValetudo/config.json
Connecting to MQTT broker...
Connected to MQTT broker.
/srv/ICantBelieveItsNotValetudo/lib/MapDrawer.js:47
            x1: Math.min(...mapData.layers.flatMap(layer => layer.dimensions.x.min)),
                                           ^

TypeError: mapData.layers.flatMap is not a function
    at MapDrawer.updateMap (/srv/ICantBelieveItsNotValetudo/lib/MapDrawer.js:47:44)
    at Timeout.MqttClient.setInterval [as _onTimeout] (/srv/ICantBelieveItsNotValetudo/lib/MqttClient.js:83:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! i-cant-believe-its-not-valetudo@2021.2.1 start: `node app.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the i-cant-believe-its-not-valetudo@2021.2.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-03-28T11_07_02_763Z-debug.log
```

## To Reproduce

### Vacuum Model
viomi v7 (Viomi Mop P)

### Valetudo Version

2021.03.0

## Expected behavior
No errors in the log ;)

## Additional context

Depau, 2021-04-23:
Fixed in next release.

closed 2021-04-23 by Depau

nekromant, 2021-04-26:
@Depau Thanks a lot!