Hypfer/Valetudo #793: Viomi.v7: segment cleaning is always using the S pattern for mopping

markthehipster, 2021-03-28:
## Description
Viomi v7 is always using S pattern to mop when cleaning in segmentation mode, despite Y mode being selected( and working in other cleaning modes i.e full clean, zone clean.)

## To Reproduce

Steps to reproduce the behavior:
1. Go to Map.
2. Select a room (segment) to clean.
3. Let the robot clean and it'll use S pattern, even if set to Y via miiocli.

### Vacuum Model

Viomi.v7
### Valetudo Version

Latest commit / also experiencing since @jloewe introduced the segmentation cleaning.
## Expected behavior
Vacuum should mop in the pattern which is currently selected. (Y or S)

markthehipster, 2021-03-28:
Funky, I just started my robot with the latest commit, and it doesnt seem to happen now.. 🤔 
Granted, I selected all the available segments, but it's mopping in Y now.

Will keep this open for a bit, but I'll try again later with not all segments selected if it also mops with Y now, then I'll close this one based on the results. 
Could be that your latest fixes @Depau also fixed this one :)

markthehipster, 2021-03-28:
Closing, tried segment cleaning multiple times, and now it mops with Y. 
Thanks.

closed 2021-03-28 by markthehipster

Depau, 2021-03-28:
> Could be that your latest fixes @Depau also fixed this one :)

Yes, this was another side-effect of the issues resolved by my last PRs.