Hypfer/Valetudo #794: viomi: Incorrect cleaning time

Depau, 2021-03-28:
> @Depau I installed a recent commit (two days ago) and found that area is reporting now is showing right square meters, but time seems to be reporting only hours:minutes, for example if the viomi has been vacuuming for 7 minutes 10 seconds it shows 00:00:07
The same data in a rockrobo unit shows 00:07:10

_Originally posted by @forast3r in https://github.com/Hypfer/Valetudo/issues/747#issuecomment-808924362_

---

Will check.

exotsk, 2021-03-29:
[image]
the same

exotsk, 2021-04-10:
[image]
Consumables counting wrong.

Hypfer, 2021-04-10:
Possibly related to #754 ?

Depau, 2021-04-10:
> [image]
> Consumables counting wrong.

What's the problem here?

closed 2021-04-10 by Hypfer

exotsk, 2021-04-11:
What's the problem here?
Wrong time(second equal minute) is a good for consumables  =)

PR #822 (2021-04-10): viomi: Fix last clean time
Closes #794