Hypfer/Valetudo #799: viomi.v7 (Mop Pro) logs spammed with  unhandledRejection MiioTimeoutError: request timed out:{"method":"set_timezone",

forast3r, 2021-03-30:
Latest pr installed, thousands of errors [ERROR] unhandledRejection MiioTimeoutError: request timed out:{"method":"set_timezone","params":[0],"id":xxxx}

## To Reproduce

Robot has valetudo installed and MQTT active. No internet access is given to the robot network.
This error appears every 10 seconds in the logs.

### Screenshots

n.a.
### Vacuum Model

viomi v7 (xiaomi vacuum mop pro)

### Valetudo Version

2021.03.00 + pr 25836e...

## Expected behavior

No errors :)

## Additional context

partial log (log level: trace) attached for a single msg id:2767

> [2021-03-12T20:40:56.445Z] [DEBUG] >>> cloud: {"method":"set_timezone","params":[0],"id":2767}
> [2021-03-12T20:40:57.088Z] [TRACE] <<< cloud: {"result":["ok"],"id":2765}
> [2021-03-12T20:40:57.096Z] [DEBUG] map_upload_handler { '2755629254858': '', suffix: 'urls' }
> [2021-03-12T20:40:57.115Z] [TRACE] fpos:0 take 2@featureFlags <Buffer ff 78>
> [2021-03-12T20:40:57.116Z] [TRACE] fpos:2 take 2@unknown01 <Buffer 00 00>
> [2021-03-12T20:40:57.117Z] [TRACE] fpos:4 take 4@mapId <Buffer 1c d4 4d 60>
> [2021-03-12T20:40:57.117Z] [TRACE] fpos:8 take 40@robot status <Buffer 01 00 00 00 00 00 00 00 c8 00 00 00 01 00 00 00 03 00 00 00 39 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>
> [2021-03-12T20:40:57.118Z] [TRACE] fpos:30 take 40@map head <Buffer 1c d4 4d 60 01 00 00 00 01 00 00 00 20 03 00 00 20 03 00 00 00 00 a0 c1 00 00 a0 c1 00 00 a0 41 00 00 a0 41 cd cc 4c 3d>
> [2021-03-12T20:40:57.154Z] [TRACE] fpos:9c458 take 12@history <Buffer 1c d4 4d 60 00 00 00 00 00 00 00 00>
> [2021-03-12T20:40:57.155Z] [TRACE] fpos:9c464 take 16@charge station <Buffer 1c d4 4d 60 a0 60 2e be e4 d2 e3 3d 0a e8 38 bf>
> [2021-03-12T20:40:57.156Z] [TRACE] fpos:9c474 take 12@virtual wall <Buffer 1c d4 4d 60 00 00 00 00 04 00 00 00>
> [2021-03-12T20:40:57.157Z] [TRACE] fpos:9c480 take 12@virtual wall prefix <Buffer 01 00 00 00 00 00 00 00 04 00 00 00>
> [2021-03-12T20:40:57.157Z] [TRACE] fpos:9c48c take 32@Virtual walls coords <Buffer 9a 99 89 40 33 33 63 c0 9a 99 89 40 33 33 63 c0 00 00 a8 40 33 33 63 c0 00 00 a8 40 33 33 63 c0>
> [2021-03-12T20:40:57.158Z] [TRACE] fpos:9c4ac take 48@unknown48 <Buffer 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>
> [2021-03-12T20:40:57.158Z] [TRACE] fpos:9c4dc take 12@virtual wall prefix <Buffer 02 00 00 00 00 00 00 00 04 00 00 00>
> [2021-03-12T20:40:57.159Z] [TRACE] fpos:9c4e8 take 32@Virtual walls coords <Buffer 00 00 00 40 cd cc 7c c0 33 33 73 40 cd cc 7c c0 33 33 73 40 cd cc 8c c0 00 00 00 40 cd cc 8c c0>
> [2021-03-12T20:40:57.159Z] [TRACE] fpos:9c508 take 48@unknown48 <Buffer 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>
> [2021-03-12T20:40:57.160Z] [TRACE] fpos:9c538 take 12@virtual wall prefix <Buffer 03 00 00 00 00 00 00 00 04 00 00 00>
> [2021-03-12T20:40:57.161Z] [TRACE] fpos:9c544 take 32@Virtual walls coords <Buffer 66 66 e6 3e 00 00 40 3f 66 66 66 40 00 00 40 3f 66 66 66 40 33 33 b3 3e 66 66 e6 3e 33 33 b3 3e>
> [2021-03-12T20:40:57.161Z] [TRACE] fpos:9c564 take 48@unknown48 <Buffer 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>
> [2021-03-12T20:40:57.162Z] [TRACE] fpos:9c594 take 12@virtual wall prefix <Buffer 04 00 00 00 00 00 00 00 04 00 00 00>
> [2021-03-12T20:40:57.162Z] [TRACE] fpos:9c5a0 take 32@Virtual walls coords <Buffer 9a 99 19 be 9a 99 b9 bf 9a 99 59 3f 9a 99 b9 bf 9a 99 59 3f 00 00 20 c0 9a 99 19 be 00 00 20 c0>
> [2021-03-12T20:40:57.163Z] [TRACE] fpos:9c5c0 take 48@unknown48 <Buffer 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>
> [2021-03-12T20:40:57.163Z] [TRACE] fpos:9c5f0 take 12@area head <Buffer 1c d4 4d 60 01 00 00 00 00 00 00 00>
> [2021-03-12T20:40:57.164Z] [TRACE] fpos:9c5fc take 20@navigate <Buffer 1c d4 4d 60 00 00 00 00 00 80 89 44 00 80 89 44 00 00 00 00>
> [2021-03-12T20:40:57.167Z] [TRACE] fpos:9c610 take 21@realtime <Buffer 1c d4 4d 60 00 00 00 00 00 ae cf 90 be 27 75 57 3e 0a e8 38 bf>
> [2021-03-12T20:40:57.167Z] [TRACE] fpos:9c625 take 8@unknown8 <Buffer 1c d4 4d 60 00 00 00 01>
> [2021-03-12T20:40:57.168Z] [TRACE] fpos:9c62d take 4@room header <Buffer 1c d4 4d 60>
> [2021-03-12T20:40:57.169Z] [TRACE] fpos:9c636 take 4@mapArg:Map1 <Buffer 01 00 00 00>
> [2021-03-12T20:40:57.169Z] [TRACE] fpos:9c63a take 4@room length <Buffer 05 00 00 00>
> [2021-03-12T20:40:57.170Z] [TRACE] fpos:9c63e room10: room0 640574
> [2021-03-12T20:40:57.170Z] [TRACE] fpos:9c64e room11: room3 640590
> [2021-03-12T20:40:57.170Z] [TRACE] fpos:9c65e room12: room1 640606
> [2021-03-12T20:40:57.171Z] [TRACE] fpos:9c66e room13: room2 640622
> [2021-03-12T20:40:57.171Z] [TRACE] fpos:9c67e room14: room4 640638
> [2021-03-12T20:40:57.171Z] [TRACE] fpos:9c68e take 6@unknown <Buffer 01 01 00 00 00 00>
> [2021-03-12T20:40:57.172Z] [TRACE] fpos:9c694 take 12@rooms2head <Buffer 1c d4 4d 60 01 00 00 00 04 00 00 00>
> [2021-03-12T20:40:57.172Z] [TRACE] fpos:9c6a0 take 4@rooms2 length <Buffer 01 00 00 00>
> [2021-03-12T20:40:57.173Z] [TRACE] fpos:9c6a4 take 2@room data <Buffer 00 00>
> [2021-03-12T20:40:57.173Z] [TRACE] fpos:9c6a6 take 4@unknown2 <Buffer 00 00 04 00>
> [2021-03-12T20:40:57.174Z] [TRACE] fpos:9c6a7 take 50@unknown50 <Buffer 00 04 00 00 00 9a 99 89 40 33 33 63 c0 9a 99 89 40 33 33 63 c0 00 00 a8 40 33 33 63 c0 00 00 a8 40 33 33 63 c0 00 00 00 00 00 00 00 00 00 00 00 00 00>
> [2021-03-12T20:40:57.174Z] [TRACE] fpos:9c6d9 take 5@unknown5 <Buffer 00 00 00 00 00>
> [2021-03-12T20:40:57.175Z] [TRACE] fpos:9c6de take 4@pose length <Buffer 00 00 00 00>
> [2021-03-12T20:40:57.175Z] [TRACE] fpos:9c6e2 take 5460@trailing <Buffer 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 00 00 00 00 00 04 00 00 00 00 00 00 40 cd cc 7c c0 33 33 73 40 ... 5410 more bytes>
[... 29 lines cut ...]
> [2021-03-12T20:40:58.379Z] [TRACE] fpos:9c62d take 4@room header <Buffer 1c d4 4d 60>
> [2021-03-12T20:40:58.379Z] [TRACE] fpos:9c636 take 4@mapArg:Map1 <Buffer 01 00 00 00>
> [2021-03-12T20:40:58.380Z] [TRACE] fpos:9c63a take 4@room length <Buffer 05 00 00 00>
> [2021-03-12T20:40:58.380Z] [TRACE] fpos:9c63e room10: room0 640574
> [2021-03-12T20:40:58.381Z] [TRACE] fpos:9c64e room11: room3 640590
> [2021-03-12T20:40:58.381Z] [TRACE] fpos:9c65e room12: room1 640606
> [2021-03-12T20:40:58.381Z] [TRACE] fpos:9c66e room13: room2 640622
> [2021-03-12T20:40:58.381Z] [TRACE] fpos:9c67e room14: room4 640638
> [2021-03-12T20:40:58.381Z] [TRACE] fpos:9c68e take 6@unknown <Buffer 01 01 00 00 00 00>
> [2021-03-12T20:40:58.382Z] [TRACE] fpos:9c694 take 12@rooms2head <Buffer 1c d4 4d 60 01 00 00 00 04 00 00 00>
> [2021-03-12T20:40:58.382Z] [TRACE] fpos:9c6a0 take 4@rooms2 length <Buffer 01 00 00 00>
> [2021-03-12T20:40:58.383Z] [TRACE] fpos:9c6a4 take 2@room data <Buffer 00 00>
> [2021-03-12T20:40:58.383Z] [TRACE] fpos:9c6a6 take 4@unknown2 <Buffer 00 00 04 00>
> [2021-03-12T20:40:58.384Z] [TRACE] fpos:9c6a7 take 50@unknown50 <Buffer 00 04 00 00 00 9a 99 89 40 33 33 63 c0 9a 99 89 40 33 33 63 c0 00 00 a8 40 33 33 63 c0 00 00 a8 40 33 33 63 c0 00 00 00 00 00 00 00 00 00 00 00 00 00>
> [2021-03-12T20:40:58.384Z] [TRACE] fpos:9c6d9 take 5@unknown5 <Buffer 00 00 00 00 00>
> [2021-03-12T20:40:58.385Z] [TRACE] fpos:9c6de take 4@pose length <Buffer 00 00 00 00>
> [2021-03-12T20:40:58.385Z] [TRACE] fpos:9c6e2 take 5460@trailing <Buffer 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 00 00 00 00 00 04 00 00 00 00 00 00 40 cd cc 7c c0 33 33 73 40 ... 5410 more bytes>
> [2021-03-12T20:40:58.405Z] [DEBUG] <<< cloud: {"result":["ok"],"id":2768}
> [2021-03-12T20:40:58.406Z] [INFO] << cloud: ignoring response for non-pending request {"result":["ok"],"id":2768}
> [2021-03-12T20:40:58.443Z] [DEBUG] cloud request 2767 set_timezone timed out
> [2021-03-12T20:40:58.446Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
> [2021-03-12T20:40:58.447Z] [ERROR] unhandledRejection MiioTimeoutError: request timed out:{"method":"set_timezone","params":[0],"id":2767}
>     at Timeout.timeout [as _onTimeout] (/snapshot/Valetudo-devel/lib/miio/MiioSocket.js:197:32)
>     at listOnTimeout (internal/timers.js:554:17)
>     at processTimers (internal/timers.js:497:7)

TimP4w, 2021-04-02:
Have you installed Valetudo via the rooting script? [viomi-root.sh](https://github.com/rumpeltux/viomi-rooting/blob/master/viomi-root.sh)

If so, I noticed  a bug there where basically this line is never called [L52](https://github.com/rumpeltux/viomi-rooting/blob/master/viomi-root.sh#L52). This happened to me just now and I managed to fix it this way.

Try to:
1. Uninstall valetudo
2. connecting via SSH to the robot and run `cd /etc/rc.d; ln -s ../init.d/robotManager S90robotManager` 
3. Reinstall valetudo

Possible problem 2:
I had problem with the ntp server, try to change it in your `valetudo_config.json" file.

forast3r, 2021-04-02:
Hi, thanks for answering

> Have you installed Valetudo via the rooting script? [viomi-root.sh](https://github.com/rumpeltux/viomi-rooting/blob/master/viomi-root.sh)
>
> If so, I noticed a bug there where basically this line is [...]
Symlink was ok, I did the rooting manually following rumpeltux's script steps, because some of the commands didn't work as expected. The other functions have been working well, I also installed several PR since last release.

> Possible problem 2:
> I had problem with the ntp server, try to change it in your `valetudo_config.json` file.

That's the actual ntp section, pool.ntp.org is accessible from the robot. What ntp client does valetudo use? An integrated one?
```
  "ntpClient": {                                  
    "enabled": true,                              
    "server": "pool.ntp.org",                     
    "port": 123,                                  
    "interval": 28800000,                         
    "timeout": 10000                              
  }, 
```
At startup it reports a timeout, will check if there is a successful time sync later...
Now I noticed that the viomi is not updating its date... it is at 2021.03.12 (again) will check the ntpd calls.
Will also check openwrt ntpd config (/etc/config/system). Don't know if original firmware will fail if I touch there...

forast3r, 2021-04-03:
Definitively there is a bug with the date setting somewhere: if I enable firewall rules to access WAN (internet), the error remains.
Also if I try to set the date with ntp manually, I have observed that it gets changed but a second later it is "restored" to a previous date:

`root@TinaLinux:~# ntpd -q -p ptbtime1.ptb.de; date; sleep 1; date; sleep 1; date
Sat Apr  3 09:35:08 CET 2021
Fri Mar 12 21:22:05 CET 2021
Fri Mar 12 21:22:06 CET 2021`

Any idea of what's going on?

forast3r, 2021-04-03:
Definetively robotManager is the responsible for setting a wrong date: after stopping it the date is ok.

I have also seen in other issues that without robotManager the vacuum will not work, so no point there.

In the other hand, the error in the logs is still there: 
[2021-04-02T09:56:22.249Z] [ERROR] unhandledRejection MiioTimeoutError: request timed out:{"method":"set_timezone","params":[0],"id":337}
    at Timeout.timeout [as _onTimeout] (/snapshot/Valetudo-devel/lib/miio/MiioSocket.js:197:32)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)

exotsk, 2021-04-03:
[2021-04-03T10:26:02.937Z] [ERROR] unhandledRejection MiioTimeoutError: request timed out:{"method":"set_timezone","params":[0],"id":29844}
    at Timeout.timeout [as _onTimeout] (/snapshot/Valetudo/lib/miio/MiioSocket.js:197:32)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
since 2021 versions

forast3r, 2021-04-03:
> `[2021-04-03T10:26:02.937Z] [ERROR] unhandledRejection MiioTimeoutError: request timed out:{"method":"set_timezone","params":[0],"id":29844} at Timeout.timeout [as _onTimeout] (/snapshot/Valetudo/lib/miio/MiioSocket.js:197:32) at listOnTimeout (internal/timers.js:554:17) at processTimers **(internal/timers.js:497:7)`**
> 
> Every 16 seconds in log since #732 , if I am not mistaken
> ViomiV8

Indeed, every 16 seconds and sometimes every 8 seconds (duplicate number of calls?)

Will study #732 and see what can be done to avoid this error.

Depau, 2021-04-05:
> Hi, is there a fix or some setting that has to be done manually on viomi to get rid of this error message?

Yes there is one, it's configuring the vacuum correctly. If it does that it means it's not properly connecting to the dummy cloud.

Please uninstall Valetudo, backup and remove your current configuration file from the vacuum, then follow the instructions in the Valetudo website to reinstall it: https://valetudo.cloud/pages/installation/viomi.html

forast3r, 2021-04-05:
> Yes there is one, it's configuring the vacuum correctly. If it does that it means it's not properly connecting to the dummy cloud.
>
> Please uninstall Valetudo, backup and remove your current [...]

Thanks for pointing. Will uninstall and double check again the installation steps.

Depau, 2021-04-05:
Test with the changes in #806

forast3r, 2021-04-05:
> Test with the changes in #806

Tested, works like a charm:

```
[2021-03-26T08:10:23.659Z] [WARN] MiioSocket local: Received keep-alive packet with stamp 1616746222 but we're at 1616746224. Discarding.
[2021-04-05T09:34:29.002Z] [INFO] Successfully set the robot time via NTP to 2021-04-05T09:34:29.890Z
[2021-03-26T08:10:25.601Z] [INFO] Connected successfully to mqtt server
```
Thanks for the quick reply!
Applying the suggested changes the error does not appear anymore. 
Note that the robot's date is correct just after valetudo sets it, then it returns to the previous setting -but I guess it's another issue-. Closing this one.

closed 2021-04-05 by forast3r

Depau, 2021-04-05:
> Tested, works like a charm:
>
> ```
> [2021-03-26T08:10:23.659Z] [WARN] MiioSocket local: Received keep-alive packet with stamp 1616746222 but we're at 1616746224. Discarding.
> [2021-04-05T09:34:29.002Z] [...]

Could you please post the entire log snippet around the `set_timezone` part? If you have the log level set to "trace".

Note that it won't show up if you restart it, since it only sets the timezone if it's not UTC.

Depau, 2021-04-05:
>  Note that the robot's date is correct just after valetudo sets it, then it returns to the previous setting -but I guess it's another issue

How are you checking the robot's date? From the command line?

forast3r, 2021-04-05:
> > Note that the robot's date is correct just after valetudo sets it, then it returns to the previous setting -but I guess it's another issue
> 
> How are you checking the robot's date? From the command line?

Both from command line (date command) and with the logs (see the 3 lines in my precious post)

forast3r, 2021-04-05:
> Could you please post the entire log snippet around the `set_timezone` part? If you have the log level set to "trace".
>
> Note that it won't show up if you restart it, since it only sets the timezone [...]

Sorry didn't see this reply. Will post later when I'm at home

forast3r, 2021-04-05:
@Depau here are the significant logs I could see regarding to time setting (not timezone). As you can see, after setting the new time, the robot goes back to the old time some seconds later:
```
[2021-03-26T06:40:49.453Z] [DEBUG] Starting NTP Query for { server: 'pool.ntp.org', port: 123, timeout: 10000 }
[2021-03-26T06:40:49.464Z] [TRACE] Intercepting dns.lookup call for pool.ntp.org
[2021-03-26T06:40:49.483Z] [DEBUG] Got Time from NTP Server: 2021-04-05T11:08:47.890Z
[2021-04-05T11:08:47.002Z] [INFO] Successfully set the robot time via NTP to 2021-04-05T11:08:47.890Z
[2021-04-05T11:08:47.004Z] [DEBUG] Next NTP sync in 28800000 ms
[2021-04-05T11:08:47.253Z] [DEBUG] <<< cloud* {"stamp":1616740850}
[2021-04-05T11:08:47.254Z] [DEBUG] >>> cloud* {"stamp":1616740850}
[2021-04-05T11:08:49.445Z] [DEBUG] >>> cloud: {"method":"get_consumables","params":[],"id":7}
[2021-04-05T11:08:49.527Z] [DEBUG] <<< cloud: {"result":[13,13,13,6],"id":7}
[2021-04-05T11:08:51.442Z] [TRACE] >>> cloud: {"method":"set_uploadmap","params":[2],"id":8}
[2021-03-26T06:40:55.550Z] [TRACE] <<< cloud: {"result":["ok"],"id":8}
```
I had restarted the robot before to restore the old zones to the config file, so I guess it is now in UTC timezone. Will leave it all day in trace mode. If you need further tests don't hesitate to reply with the steps to follow.

Depau, 2021-04-05:
> Will leave it all day in trace mode. If you need further tests don't hesitate to reply with the steps to follow.

No need, this is enough, thank you.

I will have to check why the timezone keeps changing.

forast3r, 2021-04-05:
> > Will leave it all day in trace mode. If you need further tests don't hesitate to reply with the steps to follow.
> 
> No need, this is enough, thank you.
> 
> I will have to check why the timezone keeps changing.

Ok, turning it off.

As stated in a previous comment https://github.com/Hypfer/Valetudo/issues/799#issuecomment-812843735 stoping robotManager the time stays ok.

axcs, 2022-09-01:
@jpf4e Hello I have this error Error while adjusting timezone to UTC. And valetudo or the cloud stops working randomly while vacuuming, losing the Wifi signal, or rather the cloud connection, because over ssh it always continues to work. But the LEDs blink and valetudo no longer has a connection. Have you already solved the problem?

PR #806 (2021-04-05): viomi: Increase timeout for set_timezone
## Type of change

Type A:
- [x] Bug fix (non-breaking change which fixes an issue)
- [ ] Refactor/Code Cleanup
- [ ] Docs
- [ ] Capability implementation for existing core capability
- [ ] New robot implementation
 

# Description (Type A)
It looks like sometimes set_timezone takes ~1s to execute in some cases.
We have time. Increase timeout so it doesn't take the entire connection down.

EDIT:
Actually I looked at the guy's logs better and it seems like `set_timezone` takes 10 seconds. I gave it 12 sec timeout.

Closes #799
commit a128a914 (2021-04-05): fix(vendor.viomi): Increase timeout for set_timezone (#806)

It looks like sometimes set_timezone takes ~10s to execute in some cases.
We have time. Increase timeout so it doesn't take the entire connection down.

Closes #799