Hypfer/Valetudo #810: Unable to parse map data

markthehipster, 2021-04-06:
After upgrading to the latest commit 3c3df9a584cdb7193715b583fbb90c9dcd8591ca, map data cannot be parsed anymore, and `mapdata(n)` gets dumped to `/tmp` folder continuously (probably until the disk is full)
### Screenshots

[image]
[image]

### Vacuum Model

viomi v7
### Valetudo Version

3c3df9a584cdb7193715b583fbb90c9dcd8591ca

## Expected behavior

Mapdata to be parsed :) 

## Additional context

```
ICantBelieveItsNotValetudo also fails to get mqtt data to HA.

Tue Apr  6 16:43:47 2021 daemon.info valetudo[697]: [2021-04-06T14:43:47.287Z] [INFO] Cloud connected
Tue Apr  6 16:43:47 2021 daemon.info valetudo[697]: [2021-04-06T14:43:47.411Z] [INFO] Unknown cloud message received: {"id":256334662,"method":"props","params":{"ota_state":"idle"}}
Tue Apr  6 16:43:51 2021 daemon.info valetudo[697]: [2021-04-06T14:43:51.110Z] [INFO] Unknown cloud message received: {"id":256334662,"method":"props","params":{"ota_state":"idle"}}
Tue Apr  6 16:43:52 2021 daemon.err valetudo[697]: [2021-04-06T14:43:52.409Z] [ERROR] Error parsing map. Dump saved in /tmp/mapdata0
Tue Apr  6 16:43:52 2021 daemon.err valetudo[697]: [2021-04-06T14:43:52.412Z] [ERROR] unhandledRejection Error: Invalid points array
Tue Apr  6 16:43:52 2021 daemon.err valetudo[697]:     at new MapEntity (/snapshot/Valetudo/lib/entities/map/MapEntity.js:19:19)
Tue Apr  6 16:43:52 2021 daemon.err valetudo[697]:     at new PathMapEntity (/snapshot/Valetudo/lib/entities/map/PathMapEntity.js:12:9)
Tue Apr  6 16:43:52 2021 daemon.err valetudo[697]:     at ViomiMapParser.convertToValetudoMap (/snapshot/Valetudo/lib/robots/viomi/ViomiMapParser.js:275:27)
Tue Apr  6 16:43:52 2021 daemon.err valetudo[697]:     at ViomiMapParser.parse (/snapshot/Valetudo/lib/robots/viomi/ViomiMapParser.js:203:21)
Tue Apr  6 16:43:52 2021 daemon.err valetudo[697]:     at ViomiV7ValetudoRobot.parseMap (/snapshot/Valetudo/lib/robots/viomi/ViomiValetudoRobot.js:490:50)
Tue Apr  6 16:43:52 2021 daemon.err valetudo[697]:     at /snapshot/Valetudo/lib/robots/MiioValetudoRobot.js:83:54
Tue Apr  6 16:43:52 2021 daemon.err valetudo[697]:     at processTicksAndRejections (internal/process/task_queues.js:93:5)
Tue Apr  6 16:43:52 2021 daemon.info valetudo[697]: [2021-04-06T14:43:52.418Z] [INFO] << cloud: ignoring response for non-pending request {"result":["ok"],"id":2}
Tue Apr  6 16:43:52 2021 daemon.err valetudo[697]: [2021-04-06T14:43:52.421Z] [WARN] Error while saving unparsable map null
Tue Apr  6 16:43:54 2021 daemon.info valetudo[697]: [2021-04-06T14:43:54.178Z] [INFO] Unknown cloud message received: {"id":256334662,"method":"props","params":{"ota_state":"idle"}}
Tue Apr  6 16:43:57 2021 daemon.info valetudo[697]: [2021-04-06T14:43:57.926Z] [INFO] error response { id: 4, error: { code: -9999, message: 'user ack timeout' } }
Tue Apr  6 16:43:57 2021 daemon.err valetudo[697]: [2021-04-06T14:43:57.928Z] [ERROR] unhandledRejection { code: -9999, message: 'user ack timeout' }
Tue Apr  6 16:44:00 2021 daemon.info valetudo[697]: [2021-04-06T14:44:00.934Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
Tue Apr  6 16:44:03 2021 daemon.info valetudo[697]: [2021-04-06T14:44:03.910Z] [INFO] Cloud connected
Tue Apr  6 16:44:03 2021 daemon.info valetudo[697]: [2021-04-06T14:44:03.928Z] [INFO] << cloud: ignoring response for non-pending request {"id":7,"error":{"code":-9999,"message":"user ack timeout"}}
Tue Apr  6 16:44:09 2021 daemon.err valetudo[697]: [2021-04-06T14:44:09.029Z] [ERROR] Error parsing map. Dump saved in /tmp/mapdata1
Tue Apr  6 16:44:09 2021 daemon.err valetudo[697]: [2021-04-06T14:44:09.031Z] [ERROR] unhandledRejection Error: Invalid points array
Tue Apr  6 16:44:09 2021 daemon.err valetudo[697]:     at new MapEntity (/snapshot/Valetudo/lib/entities/map/MapEntity.js:19:19)
Tue Apr  6 16:44:09 2021 daemon.err valetudo[697]:     at new PathMapEntity (/snapshot/Valetudo/lib/entities/map/PathMapEntity.js:12:9)
Tue Apr  6 16:44:09 2021 daemon.err valetudo[697]:     at ViomiMapParser.convertToValetudoMap (/snapshot/Valetudo/lib/robots/viomi/ViomiMapParser.js:275:27)
Tue Apr  6 16:44:09 2021 daemon.err valetudo[697]:     at ViomiMapParser.parse (/snapshot/Valetudo/lib/robots/viomi/ViomiMapParser.js:203:21)
Tue Apr  6 16:44:09 2021 daemon.err valetudo[697]:     at ViomiV7ValetudoRobot.parseMap (/snapshot/Valetudo/lib/robots/viomi/ViomiValetudoRobot.js:490:50)
Tue Apr  6 16:44:09 2021 daemon.err valetudo[697]:     at /snapshot/Valetudo/lib/robots/MiioValetudoRobot.js:83:54
Tue Apr  6 16:44:09 2021 daemon.err valetudo[697]:     at processTicksAndRejections (internal/process/task_queues.js:93:5)
Tue Apr  6 16:44:09 2021 daemon.info valetudo[697]: [2021-04-06T14:44:09.037Z] [INFO] << cloud: ignoring response for non-pending request {"result":["ok"],"id":9}
Tue Apr  6 16:44:09 2021 daemon.err valetudo[697]: [2021-04-06T14:44:09.039Z] [WARN] Error while saving unparsable map null
Tue Apr  6 16:44:14 2021 daemon.info valetudo[697]: [2021-04-06T14:44:14.912Z] [INFO] error response { id: 11, error: { code: -9999, message: 'user ack timeout' } }
Tue Apr  6 16:44:14 2021 daemon.err valetudo[697]: [2021-04-06T14:44:14.919Z] [ERROR] unhandledRejection { code: -9999, message: 'user ack timeout' }
Tue Apr  6 16:44:52 2021 daemon.err valetudo[697]: [2021-04-06T14:44:52.903Z] [ERROR] Error parsing map. Dump saved in /tmp/mapdata2
```

Could it be that 371ceea9e4e7b591b19b0dc3d4887feca3ab8692 broke it due to a possible hidden dependency?

Hypfer, 2021-04-06:
It was broken by a2e1c21bcc3dbf12a83fdaea84701b98d9387c17 and I've only fixed it for Roborock and Dreame

Hypfer, 2021-04-06:
Should be fixed by 77e71281

markthehipster, 2021-04-06:
Ah awesome, thank you for the quick fix!

closed 2021-04-06 by markthehipster