Hypfer/Valetudo #814: Webinterface stops working after log fills up tmp

johnappletree, 2021-04-07:
valetudo.log fills up my /tmp and causes webinterface to crash.

```
Successfully set the robot time via NTP to 2021-04-07T18:43:47.890Z
```
previous listed NTP set time messages fill up valetudo.log in the /tmp directory. The file quickly outgrows it's maximum allowed size and causes the Valetudo software to crash. Setting the log level from info to warn results in following recurring message:
```
[2021-04-07T10:54:11.060Z] [WARN] Unable to parse Pose TypeError: Cannot set property 'outline' of undefined
    at ViomiMapParser.parsePose (/snapshot/Valetudo/lib/ViomiMapParser.js:381:40)
    at ViomiMapParser.parse (/snapshot/Valetudo/lib/ViomiMapParser.js:191:22)
    at ViomiV7ValetudoRobot.parseMap (/snapshot/Valetudo/lib/robots/viomi/ViomiValetudoRobot.js:414:50)
    at /snapshot/Valetudo/lib/robots/MiioValetudoRobot.js:85:48
    at runMicrotasks (<anonymous>)
```
This in the long run results in the same crash behavior

Is there an way to either prevent NTP messages in the log or specify a max log size?

robot:ViomiV7ValetudoRobot
version: 2021.03.0

Thanks!

JA

johnappletree, 2021-04-07:
For now; it seams that setting the log level to error mitigates the problem.

forast3r, 2021-04-07:
> For now; it seems that I can set the log level to _error_ in order to mitigates the problem.

If the robot reboots, valetudo will restart at info log level, so I guess it is not a solution for you.

Take a look at this issue: https://github.com/Hypfer/Valetudo/issues/799 if the errors are related to timezone then it has a solution. Otherwise I'd add relevant logs so someone can check it.
If the problem is with the integrated ntp client, it can be disabled in the config file.

johnappletree, 2021-04-07:
>  If the robot reboots, valetudo will restart at info log level, so I guess it is not a solution for you.

Good point hadn't considered that. 

> Take a look at this issue: #799 if the errors are related to timezone then it has a solution. Otherwise I'd add relevant logs so someone can check it.
> If the problem is with the integrated ntp client, it can be disabled in the config file.

Nope that looks unrelated, have not seen those messages in my log.

However, it looks like I might have found another solution. i must have swapped interval and timeout in the _valetudo_config.json_ file. I've switched them to:

```
    "interval": 10000,
    "timeout": 100
```

which will hopefully solve my problem.

Thanks again.

JA

johnappletree, 2021-04-07:
Cranked up my _interval_ and _timeout_ setting to mentioned values in #799.
_interval_ = 10000 and _timeout_ = 100 is still to little/often. 

I have set the values to:
```
    "interval": 28800000,
    "timeout": 10000
```

I will close this issue as my issue is solved. Although, a max log size might still be a useful feature ;-)

closed 2021-04-07 by johnappletree