Hypfer/Valetudo #815: Unable to save forbidden markers

tobias1818, 2021-04-08:
## Describe the bug

I'm unable to save the forbidden markers. It says „too many forbidden markers to save“. The forbidden markers are from an previous version, but there it was possible to save that count, but not the same count in this version (8 virtual walls, 9 no-go zones).

## To Reproduce

Add 15 or more virtual zones to the map.

### Vacuum Model

Roborock S6

### Valetudo Version

2021.03.0
Commit ID: 1f427108642e66a2acef225b5d9a76b11f0e0fe2

## Expected behavior

To save the virtual walls and no-go zones successfully (maybe an information, how many you can save).

## Additional context

> [2021-04-08T09:28:49.665Z] [WARN] Error while saving virtual restrictions Error: too many forbidden markers to save!
 at RoborockCombinedVirtualRestrictionsCapability.setVirtualRestrictions (/snapshot/Valetudo/lib/robots/roborock/capabilities/RoborockCombinedVirtualRestrictionsCapability.js:48:19)
    at /snapshot/Valetudo/lib/webserver/capabilityRouters/CombinedVirtualRestrictionsCapabilityRouter.js:29:47
    at Layer.handle [as handle_request] (/snapshot/Valetudo/node_modules/express/lib/router/layer.js:95:5)
    at next (/snapshot/Valetudo/node_modules/express/lib/router/route.js:137:13)
    at Route.dispatch (/snapshot/Valetudo/node_modules/express/lib/router/route.js:112:3)
    at Layer.handle [as handle_request] (/snapshot/Valetudo/node_modules/express/lib/router/layer.js:95:5)
    at /snapshot/Valetudo/node_modules/express/lib/router/index.js:281:22
    at Function.process_params (/snapshot/Valetudo/node_modules/express/lib/router/index.js:335:12)
    at next (/snapshot/Valetudo/node_modules/express/lib/router/index.js:275:10)
    at Function.handle (/snapshot/Valetudo/node_modules/express/lib/router/index.js:174:3)

rimrul, 2021-04-08:
Looking at the code, that restriction should only be reached if you've got over 68 points (virtual walls count as 2 points, no-go zones and no-mop zones as 4) 8 virtual walls and 9 no-go zones should add up to 52 points, so that's still a couple points away from the limit.

tobias1818, 2021-04-08:
Okay good to know how many I can save in points. I don’t know why I can’t save more when I don’t reached the limit, but if I’m the only one with that problem it is maybe connected with the map from an older version. In that version I added and removed more virtual zones, so maybe they weren’t deleted correctly (it was version 0.6.1; I updated from that version directly to 2021.03.0).

closed 2021-04-08 by tobias1818