Hypfer/Valetudo #857: Unable to use MQTT home assistant autodiscovery with homie autodiscovery disabled

flacjacket, 2021-04-26:
As of 2021.4, when I have both homie and home assistant autodiscovery enabled, my vacuum is autodiscovered and shows up in home assistant. When I disable homie autodiscovery, but leave home assistant enabled, the device disappears from home assistant as if the autodiscovered device was deleted (even though I have "Delete autodiscovery on shutdown" disabled for both homie and home assistant). Looking in my logs, I see several messages that look like either:
```
[2021-04-26T01:49:57.528Z] [WARN] MQTT handle vacuum/robot/BatteryStateAttribute failed to configure Error: Handle may only be configured while the MQTT controller is not initialized
    at BatteryStateMqttHandle.configure (/snapshot/Valetudo/lib/mqtt/handles/MqttHandle.js:160:19)
    at BatteryStateMqttHandle.configure (/snapshot/Valetudo/lib/mqtt/handles/RobotStateNodeMqttHandle.js:30:21)
    at RobotMqttHandle.configure (/snapshot/Valetudo/lib/mqtt/handles/MqttHandle.js:167:29)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:93:5)
    at async /snapshot/Valetudo/lib/mqtt/handles/RobotMqttHandle.js:110:13
    at async MqttController.reconfigure (/snapshot/Valetudo/lib/mqtt/MqttController.js:497:13)
    at async RobotMqttHandle.onStatusAttributeEvent (/snapshot/Valetudo/lib/mqtt/handles/RobotMqttHandle.js:97:9)
```
and
```
[2021-04-26T01:50:33.473Z] [ERROR] MQTT publication error: TypeError: Cannot read property 'publish' of null
    at MqttController.publishHass (/snapshot/Valetudo/lib/mqtt/MqttController.js:698:36)
    at HassController.refreshAutoconf (/snapshot/Valetudo/lib/mqtt/homeassistant/HassController.js:180:31)
    at InLineHassComponent.refreshAutoconf (/snapshot/Valetudo/lib/mqtt/homeassistant/components/HassComponent.js:146:25)
    at CallbackHassAnchorSubscriber.callback (/snapshot/Valetudo/lib/mqtt/homeassistant/components/HassComponent.js:23:24)
    at CallbackHassAnchorSubscriber.onAnchorPost (/snapshot/Valetudo/lib/mqtt/homeassistant/CallbackHassAnchorSubscriber.js:18:20)
    at HassAnchor.post (/snapshot/Valetudo/lib/mqtt/homeassistant/HassAnchor.js:62:34)
    at new InLineHassComponent (/snapshot/Valetudo/lib/mqtt/homeassistant/components/InLineHassComponent.js:24:40)
    at /snapshot/Valetudo/lib/mqtt/capabilities/WifiConfigurationCapabilityMqttHandle.js:85:17
    at MqttController.withHass (/snapshot/Valetudo/lib/mqtt/MqttController.js:676:13)
    at new WifiConfigurationCapabilityMqttHandle (/snapshot/Valetudo/lib/mqtt/capabilities/WifiConfigurationCapabilityMqttHandle.js:83:25)
```

## To Reproduce
1. Setup MQTT with homie and home assistant autodiscovery enabled, see device in home assistant, Valetudo logs:
```
[2021-04-26T01:50:33.803Z] [INFO] Connected successfully to MQTT broker
[2021-04-26T01:50:35.648Z] [INFO] MQTT configured
```
2. Disable homie autodiscovery and save, observe Valetudo logs:
```
[2021-04-26T02:01:29.401Z] [ERROR] MQTT publication error: TypeError: Cannot read property 'publish' of null
<repeated for 8 separate capabilities>
[2021-04-26T02:01:30.138Z] [INFO] Connected successfully to MQTT broker
[2021-04-26T02:01:31.981Z] [INFO] MQTT configured
```
In home assistant, the device is deleted.
3. Enable homie autodiscovery and save, observe vacuum reappear in home assistant.

### Vacuum Model
S4 Max

### Valetudo Version
2021.04.0

flacjacket, 2021-05-27:
I noticed that this problem still persists in 2021.05. Now, with Homie disabled and Home assistant autodiscovery enabled I am getting this in my log on boot:
```
[2021-05-27T17:59:41.942Z] [INFO] MQTT configured
[2021-05-27T17:59:42.022Z] [INFO] Cloud connected
[2021-05-27T17:59:42.177Z] [INFO] Unknown cloud message received: {"id":0,"result":"unknown_method"}
[2021-05-27T17:59:42.762Z] [INFO] << cloud: ignoring response for non-pending request {"id":-5295024880113156000,"result":{"code":0,"out":[{"piid":2,"value":"eyJpZCI6ODQ1Njk4OTIsIm1ldGhvZCI6InByb3BlcnRpZXNfY2hhbmd
lZCIsInBhcmFtcyI6W3siZGlkIjoiMzkxMjE3NDA0Iiwic2lpZCI6OSwicGlpZCI6MiwidmFsdWUiOjkyfV19"}]}}
[2021-05-27T17:59:52.764Z] [WARN] MQTT handle vacuum/robot/ConsumableMonitoringCapability/brush-side_right failed to configure Error: Handle may only be configured while the MQTT controller is not initialized
    at PropertyMqttHandle.configure (/snapshot/Valetudo/backend/lib/mqtt/handles/MqttHandle.js:160:19)
    at ConsumableMonitoringCapabilityMqttHandle.configure (/snapshot/Valetudo/backend/lib/mqtt/handles/MqttHandle.js:167:29)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async ConsumableMonitoringCapabilityMqttHandle.configure (/snapshot/Valetudo/backend/lib/mqtt/handles/RobotStateNodeMqttHandle.js:30:9)
    at async /snapshot/Valetudo/backend/lib/mqtt/capabilities/ConsumableMonitoringCapabilityMqttHandle.js:162:17
    at async MqttController.reconfigure (/snapshot/Valetudo/backend/lib/mqtt/MqttController.js:452:13)
    at async ConsumableMonitoringCapabilityMqttHandle.findNewConsumables (/snapshot/Valetudo/backend/lib/mqtt/capabilities/ConsumableMonitoringCapabilityMqttHandle.js:153:13)
    at async ConsumableMonitoringCapabilityMqttHandle.refresh (/snapshot/Valetudo/backend/lib/mqtt/capabilities/ConsumableMonitoringCapabilityMqttHandle.js:168:9)
```
Full log file attached.
[valetudo.log](https://github.com/Hypfer/Valetudo/files/6555916/valetudo.log)

flacjacket, 2021-05-27:
It looks like I can't re-open this issue, happy to make a new issue if that is better.

Hypfer, 2021-05-28:
Unrelated to this issue but related to your log:

`[2021-05-27T17:59:42.762Z] [INFO] << cloud: ignoring response for non-pending request {"id":-5295024880113156000,"result":{"code":0,"out":[{"piid":2,"value":"eyJpZCI6ODQ1Njk4OTIsIm1ldGhvZCI6InByb3BlcnRpZXNfY2hhbmd
lZCIsInBhcmFtcyI6W3siZGlkIjoiMzkxMjE3NDA0Iiwic2lpZCI6OSwicGlpZCI6MiwidmFsdWUiOjkyfV19"}]}}`

That base64 encodes this:

`{"id":84569892,"method":"properties_changed","params":[{"did":"391217404","siid":9,"piid":2,"value":92}]}`

SIID 9 is the Main Brush and PIID 2 its percentage left.

Why is it base64 encoded and why this insanely high ID. What is roborock doing there
We may or may not need to handle that

flacjacket, 2021-06-03:
This is covered by the fix in #947

PR #860 (2021-04-26): Publish state for both hass and homie, as the hass device availabilit…
## Type of change

- [x] Bug fix (non-breaking change which fixes an issue)
- [ ] Refactor/Code Cleanup
- [ ] Docs
- [ ] Capability implementation for existing core capability
- [ ] New robot implementation
  
# Description

Enable state topic when either hass or homie are enabled, because hass uses it as well

Fixes #857
closed 2021-06-03 by flacjacket