Hypfer/Valetudo #885: Default timezone cannot be changed

dmlemos, 2021-05-02:
## Describe the bug
I couldn't find a way to check the timezone on valetudo, neither could I change it.
Default timezone is set to `Europe/Berlin`.

Since only the DND timer is configured, I believe this only matters for this case as well as accurate logs.

## To Reproduce
1. `date`, noticed the time is different than my localtime
2. `printf 'Europe/London' > /mnt/data/rockrobo/timezone`
3. Changed `country_code` in `/mnt/data/rockrobo/country_info` to `GB`
3. Restart
4. `ntpdate -d pool.ntp.org` still shows the incorrect time
5. `hwclock -r` shows the same time as above
6. `/mnt/data/rockrobo/timezone` -> survived restart
7.  `/mnt/data/rockrobo/country_info`, `country_code=DE` -> gets reset back to default

### Vacuum Model
Roborock S5

### Valetudo Version
2021.04.0
Built using https://builder.dontvacuum.me/_s5.html
```
[root@rockrobo rockrobo]# cat /etc/os-release
NAME="Ubuntu"
VERSION="14.04.3 LTS, Trusty Tahr"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 14.04.3 LTS"
VERSION_ID="14.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
ROBOROCK_VERSION=3.5.8_002034
```

## Expected behavior
Timezone should have changed to the correct value.
Also think it is important to show the timezone on the valuetudo page.

## Additional context
No `apt` available. I couldn't use `dpkg-reconfigure tzdata`.
I have not tried linking `/usr/share/zoneinfo/Europe/London`, as I was afraid it won't do anything or worse.

I've found this issue #379 had a similar trouble

Depau, 2021-05-02:
This is an issue but not for the reason you reported it: the timezone should be set to UTC.

It will accidentally make you happy since it happens to be your timezone, but the idea is that the front-end should be in charge of handling timezone conversions from UTC. This should lower the overall code complexity.

Hypfer, 2021-05-02:
IMO it is a bad user experience for the device to have it's own timezone, since it requires you to do the conversion between your local time and the appliance time in your head. Since the robot does need to have some timezone, UTC was choosen.

The discussion about that can be found here: https://github.com/Hypfer/Valetudo/discussions/728

Roborock unfortunately needs some additional thought due to its 4am localtime automatic reboot.

You can ignore the timesync warning. That happens, because Valetudo starts before the network is available.

Anyways, not a bug

closed 2021-05-02 by Hypfer