Hypfer/Valetudo #894: No status in HomeAssistant

darootler, 2021-05-04:
## Describe the bug
After upgrading to version 2021.04.0 there is no status available for the vacuum.

## To Reproduce
Remove the vacuum from MQTT and republish all topics as described in the release notes.

### Screenshots
[image]

### Vacuum Model
Xiaomi Mi Robot Vacuum V1

### Valetudo Version
2021.04.0

## Expected behavior
Status of the vacuum is reported correctly

Hypfer, 2021-05-04:
Is homie disabled? If yes please enable that.

darootler, 2021-05-04:
Yes, it is. Here is my config:

{code}
{
  "embedded": true,
  "robot": {
    "implementation": "auto",
    "implementationSpecificConfig": {
      "ip": "127.0.0.1"
    }
  },
  "webserver": {
    "port": 80,
    "basicAuth": {
      "enabled": true,
      "username": "XXX",
      "password": "XXX"
    }
  },
  "zonePresets": {},
  "goToLocationPresets": {},
  "mqtt": {
    "enabled": true,
    "server": "XXX",
    "port": "8883",
    "clientId": null,
    "clean": false,
    "cleanTopicsOnShutdown": false,
    "username": "XXX",
    "password": "XXX",
    "usetls": true,
    "ca": "XXX",
    "clientCert": "",
    "clientKey": "",
    "qos": 0,
    "identifier": "abstellraum-staubsauger",
    "friendlyName": "abstellraum-staubsauger",
    "topicPrefix": "valetudo",
    "refreshInterval": "30",
    "homie": {
      "enabled": true,
      "addICBINVMapProperty": false,
      "cleanAttributesOnShutdown": false
    },
    "homeassistant": {
      "enabled": true,
      "autoconfPrefix": "homeassistant",
      "cleanAutoconfOnShutdown": false
    },
    "provideMapData": true
  },
  "ntpClient": {
    "enabled": true,
    "server": "XXX",
    "port": 123,
    "interval": 28800000,
    "timeout": 10000
  },
  "allowSSHKeyUpload": true,
  "logLevel": "info",
  "debug": {
    "memoryStatInterval": false,
    "enableRawCommandCapability": false,
    "debugHassAnchors": false
  }
}
{code}

Regards
Richard

Depau, 2021-05-05:
Can you post the logs?

---

Can we make it mandatory to attach the logs?

darootler, 2021-05-05:
I just rebooted the vacuum to get fresh logs and the state is now reported correctly :confused:

Regards
Richard

closed 2021-05-05 by darootler